x-station/xstation-issues #1: Castlevania SOTN: Richter intro skip on fast load

NotMoreZombies, 2020-10-16:
I've been able to reproduce this issue fairly consistently after starting SOTN with fast load.

Start a new game as normal then skip the first cutscene to get to the intro where you play as Richter. As soon as you have control, double-tap left on the d-pad to dash, and the screen should go white. The Richter intro is skipped and the next cutscene plays. I have video of all this below, but it's timecoded to start during the Richter intro.

https://www.youtube.com/watch?v=2VzuNF0DngU&t=28

I did keep on playing a little bit after this, and as far as I can tell the game was playing normally. It also doesn't occur at all if you use full load, or if you just walk up the stairs - you seem to be able to dash just fine after that.

ramapcsx2, 2020-10-17:
Yep, it's a loader issue that we're working on. A fix might be out shortly.

ramapcsx2, 2020-10-20:
Issue has been fixed as of release 1.0.2.

closed 2020-10-20 by ramapcsx2

NotMoreZombies, 2020-10-20:
For what it's worth, can confirm it is fixed on my end. Cheers rama!