x-station/xstation-issues #2: WipEout 3 (USA) crossing finish line hang

ramapcsx2, 2020-10-17:
Game hangs after crossing the finish line. It does so right at the start of a round, so checking is quick.

ramapcsx2, 2020-10-17:
The game uses some form of region lockout.
The NTSC-U version can not be played on an PAL machine (or any machine that uses a PAL mechacon chip).
It will work fine on an NTSC-J or (of course) an NTSC-U machine.

msimplay, 2021-03-17:
Getting the same issue on the Need for Speed NTSC game on a PAL Console

badass-boss, 2021-08-01:
I can confirm the original Wipeout 3 USA disk was hanging on the starting line on PS1 slim PAL modchipped. I dumped the disk and ran it on PS1 Japanese with Xstation and it works perfectly fine. Hence this is not Xstation issue but a regional hardware issue. The solution is to patch the game or develop some sort of "Xstation patch on the fly" solution