x-station/xstation-issues #5: Cannot load multi-bin images.

damontgomery, 2020-11-03:
If you have a set of bin files and one cue file, one per audio track (for example Dead or Alive), various bin files will show up in the list (but not always all of them) and you can only load one. Loading the first causes issues such as missing music.

A work around is to recreate the image without multiple bin files.

This is probably a minor issue since there is a simple work-around.

Thanks for the work on this!

ramapcsx2, 2020-11-03:
Hey,
this sounds like a file rename formatting violation.
Check out the redump entry for this game and download the .cue to see how the files have to be named:
http://redump.org/disc/73760/

damontgomery, 2020-11-03:
Thanks for the info. You mentioned in #6 that the system searches for `.bin` files. Does it read the `.cue` file when it's time to load the game and does it have to be a certain name match to the `.bin` or to one of the redump files?

My files did not match redump, but the names in the `.cue` file did match the `.bin` file names for the tracks.

The redump cue sheet has `Dead or Alive (USA) (Track 01).bin` and I had `Dead or Alive Track 1.bin` if it helps. Maybe it's confused since `Track 1` isn't `(Track 1)` with the `(`s.

Thanks again!

ramapcsx2, 2020-11-04:
That's right, it reads the .cue when mounting a disk.
The "Track 1" name might confuse the list display, but at worst, it would fail to remove these chars. Functionality would not be affected.

ramapcsx2, 2020-11-06:
I'll merge this with https://github.com/x-station/xstation-issues/issues/6. It is related and I can try to improve that.

closed 2020-11-06 by ramapcsx2