x-station/xstation-issues #6: Games list can show deleted games after refresh.

damontgomery, 2020-11-03:
I had some games with various file names and wanted them to be consistent. After renaming the games and running the list refresh, some of the old names remained.

Easy enough to fix by deleting the text files that store the list of games / history and then rebuilding the list.

Thanks for all the work on this. It's working great so far.

ramapcsx2, 2020-11-03:
Hey,
a list refresh always completely removes the old lists and rebuilds them.
If you see any old names, then these files still exist somewhere on the SD card.
If it helps: The names are taken from the .bin files, then shortened (removing "Track", etc) for display.

damontgomery, 2020-11-03:
Thanks for the info.

I was renaming the files on MacOS in which likes to create all of those `.` files. Not sure if that confused the system and left the duplicates. I thought I'd removed them and only deleted those game list text files between list build attempts, but not sure.

Easy enough to fix by checking for the hidden files and / or deleting those lists.

Thanks again.

ramapcsx2, 2020-11-04:
Oh yeah, please try to avoid hidden folders like that. They would cause performance issues as well.
I think there is a way to disable the unasked for hidden file creation in MacOS.

damontgomery, 2020-11-04:
I saw that you can disable `.DS_Store`, but not the `._` files. I think these were created by Finder when I copied files with my Mac to the FAT32 MicroSD card. Some of these are created because the drive isn't HFS+ and some seem to be created on all drives. See https://apple.stackexchange.com/a/208495 for some examples. :(

It's hard to remove these on MacOS since the files are usually hidden. It can be done with VSCode or the terminal, or on a Windows / Linux machine.

It might be a nice feature to ignore `.*` files and folders in your scans if possible.

Curious. Are the performance problems just in building the list or in running games once the list is built?

Thanks again and please don't feel any pressure to do anything about this. :) I've been having fun with the device and none of these things are blocking issues for me.

ramapcsx2, 2020-11-05:
Yeah, I can look into ignoring folders named "." :)
The performance problems would be in scanning only.

ramapcsx2, 2020-11-15:
Filters are updated and each such system folder will be ignored from now on :)

closed 2020-12-29 by ramapcsx2