x-station/xstation-issues #7: No FMV audio

defknot, 2020-11-05:
Installed the X-Station and PS1 Digital a couple of days ago at the same time. I have a SCPH_5501 PS1 and had verified everything working prior to the modifications. After I had completed both installs everything was working as expected. It wasn't until I launched Resident Evil, where I noticed that the Capcom Logo and the FMV right before the title screen of Resident Evil, where I noticed I had no audio. I launched the game and during the opening FMV sequence, there was no sound at all. Once in the game, I can hear the music and the Sound Effects. There is, however, no in game voices. I don't hear any of the dialogue in the game.

When I tried launching Tony Hawk Pro Skater 1, The opening Logos at the front of the game has no sound. While I'm actually skateboarding, there is no music at all. I have tested these games on other PS1's and they all work fine.

I tested the analog port and I still get the same thing. Today, I went with my multimeter and checked all my points to make sure they were all making contact.The are. I tested the flex cable and verified they all make contact from one end to the other.

I verified that the pins lifted from IC701 are the correct ones and do not have any damaged or bridges.

Not sure what I can do at this point. Any ideas on what I should check on the PU-18 board?

ramapcsx2, 2020-11-05:
Oh, I'm sorry to say, but this sounds like a hardware failure that I saw on an unmodified PU-8 console once.
No matter what, XA audio (such as used in FMV, but also many ingame music) would not play.
I was never able to find out what caused this ><

If you want, you could try to undo the xstation mod and retry with a normal drive.
It should still be broken.

defknot, 2020-11-11:
I tried swapping various chips, the SPU, the CD Controller, and even IC701. All no go. I’m getting a replacement board soon and will test it thoroughly before reinstalling the X-Station. The only chip I could think of swapping is the CD Driver on the B side of the board. I will just use the new board when it comes in. Thank you so much for getting back to me!

ramapcsx2, 2020-11-11:
Yeah, I could never figure out what was wrong with mine either.
Best of luck with the new board though, and if it still runs CDs, best try it out first! :)

closed 2020-11-11 by ramapcsx2