x-station/xstation-issues #10: Guide suggestion: re-use disc drive shock / spacer

damontgomery, 2020-11-06:
Thanks for the really clear guide.

I have a 5501 with an original (broken) disc drive and was able to pretty easily remove the rubber shock / spacer thing from the disc drive and use it as a spacer for the xstation. I put two of them between the board and the top case and it wasn't too difficult to align the pieces while shutting the case. Now the drive stays in place much better.

[image]

I thought other's might find this useful.

The rubber piece is a bit dusty after 20 years, but still seems to work well.

Thanks again!

ramapcsx2, 2020-11-08:
Yea, this is a good tip :)
I'll see how this fits into the next guide updates!

birdybro, 2020-11-15:
@ramapcsx2 Side note, potentially add a link to the SD Association's official Formatting Tool as well since we might be seeing a few issues with that on much larger drives. The SD Association's format tool formats differently than windows or other OS's format tools.

https://www.sdcard.org/downloads/formatter/

[image]

I don't think I have the latest .doc file else I would just make the change and submit the PR :P

ramapcsx2, 2020-11-15:
Updated the guide :)

closed 2020-12-07 by ramapcsx2