x-station/xstation-issues #9: Streaming Audio Glitching

samson7point1, 2020-11-06:
Background audio became distorted for several seconds (3-5) during gameplay.  Sound effects were unaffected.  This happened multiple times during play throughs of R4 and Bust a Move 4. The glitching was essentially identical in both games.  The audio would distort, make glitchy noise, go silent then pick back up normally. 

This was on a PU-18 motherboard with a PS1 digital mod. The image used was a verified redump image.  SD card is a 512 GB Samsung EVO micro SD connected via Laser Bear extension mount.

Prior to installation of Xstation these issues were not observed in these games on this PlayStation.

nicolasnoble, 2020-11-06:
Any chance for a video+audio capture?

samson7point1, 2020-11-06:
I was planning on trying to grab one this weekend. Will upload if I can manage it.

ramapcsx2, 2020-11-06:
Almost surely the SD card. I can make this occur on one of my Samsung EVO+ (a 128GB one).
Please try using any other card and see whether the problem changes / goes away.

samson7point1, 2020-11-06:
Here's a capture of the glitch (I wanted to reproduce the issue first).  Going to switch to a SanDisk card and see if I can reproduce it.

[https://drive.google.com/file/d/1ty9FDbM9B-8OKz_7-fapoluH5KZE9ByP](url)

ramapcsx2, 2020-11-08:
Okay, I managed 4 rounds and it played fine. :)

ThanosSiopoudis, 2020-11-11:
@samson7point1 try a couple of things please:
- remove the ps1digital hdmi cover from the back. It's been reported that it may be a little bit too big, and doesn't allow the lid switch to fully depress, causing glitches.
- remove the sd card extension, to make sure it's not causing any issues and try again

samson7point1, 2020-11-11:
@ThanosSiopoudis Neither of those is an issue on my console - I initially tested without the SD card extension and it still happened, and my HDMI port cover does not prevent the case from closing flush.  I'm fairly certain @ramapcsx2 is correct that there's a compatibility bug with the Xstation reading the Samsung EVO SD cards.  I haven't played anywhere near as long from the SanDisk card as I have the Samsung, but so far I have not experienced the issue with the SanDisk card.  Hopefully I will have some time today to do some additional testing.  About two more Gran Prix runs and I'll  be satisfied that the card is the problem.  Hopefully the compatibility issue is easy to fix and I'll be able to keep using the Samsung card I bought specifically for this.

ThanosSiopoudis, 2020-11-11:
Can you post a photo of the card here please, so we can track down the exact model? I have tested a few of the Samsung cards here and they worked without issues, so it might be a specific one I don't have.

samson7point1, 2020-11-11:
[image]

samson7point1, 2020-11-11:
I've played through several R4 Gran Prix runs with the SanDisk card without triggering the glitch.  It's really difficult to actually prove a negative condition (possibly impossible), but this amount of play time has consistently been enough to trigger the glitch when using the Samsung card so that's definitely the prime suspect.

FWIW, I tried to eliminate as many extraneous factors as possible.  MD5 sums for the R4 disc image are identical on both cards and both cards are formatted exFAT.  They're both large capacity cards (the SanDisk is only 400GB but that's decently close).

ThanosSiopoudis, 2020-11-11:
I just tried the same Samsung EVO Select, but 32GB capacity, played 6 levels of Bust-a-Move 4, and didn't have any sound issues. I'll see if I can get my hands on a 512GB one, in case there's a difference. This was using Collingal's SD card extension as well.

Can you try formatting your Samsung card using the SD Association's SD Card formatter, and try with just just Bust-a-Move on it?
https://www.sdcard.org/downloads/formatter/

birdybro, 2020-11-15:
I will try the same 512GB card you pictured today and with Xenogears as that is notoriously buggy when CD read issues are happening. I will also use the ad association formatter.

birdybro, 2020-11-18:
I finally filled mine up about 400+GB now. I do not notice any streaming audio issues. I do notice some random games feeling like they have occasional slow load, but this very well could be that specific game since I'm not testing side by side. I did test games I have on disc with another psx side by side, and everything seems normal from what I can tell, even with Xenogears which is finnicky for cd loads.

This is with the same MicroSD from Samsung in the picture posted above, so works for me. I did use the SD Association's formatter for this test. I haven't filled it up completely, and I could try if you want, just got tired since I was making a "best of" MicroSD for hours when I did this :P

ramapcsx2, 2020-11-22:
I will account this issue as similar to the other SD card related ones.

closed 2020-11-22 by ramapcsx2

Leugrim, 2021-12-16:
> [image]

I had the same problems, I was crazy looking for the problem in the instalation, i have the same sd card in 256gb samsung, with sandisk 4gb, there is not problem, thanks.

The game with problems is the Marvel Super Heroes

pirsir76, 2024-03-15:
Sorry for the necro of this issue, but I'm having audio glitching on all games.
My micro SD = Sandisk Ultra 128GB.

Very strange because the audio wasn't glitching out at first, it just appeared in the last few days.
I suspect the reason might be due to filling up my SD card to the maximum capacity as the problem first appeared when I did that.

ramapcsx2, 2024-03-15:
Not too sure @pirsir76. Things working at first, then suddenly not anymore, that could be caused by various things.
I would try to further narrow it down. If you have different SD cards, give those a try, if only to see whether the problem goes away.
It might also be caused by solder joints coming loose, as well as the SD card becoming fragmented.

To format it cleanly, get the SD Association formatter, and format the card to exFAT.
Then make sure to copy your games on in one go, as not to fragment them due to multiple copy streams.

If the issue persists, a video with audio will help clarify.

pirsir76, 2024-03-15:
@ramapcsx2 doesn't the Micro SD Card need a folder called "00xstation" what about that?
Should I just copy the existing "00xstation" back onto the Micro SD Card after it is formatted?
(I am a fairly new user to xstation.)

ramapcsx2, 2024-03-16:
Yes, the layout of the file on the card is affected by how it got copied on, the file system parameters of course, and sometimes it seems to be phase of the moon related, too :p 
You can follow those best practices and it should always be fine.
In addition to one-go copying all games, also try to not delete or move those files later.
It's okay to add more games, but deleting and then adding others will definitely cause fragmentation.

ramapcsx2, 2024-03-16:
The 00xstation folder is just the storage place for some system files, none of which are essential. It will try to fix it if it's missing.