x-station/xstation-issues #11: 1.04 boots to CD player screen

mcole24, 2020-11-07:
Xstation doesn't boot when firmware is 1.04. Issue occurs with multiple microSD cards tested, both exFAT and FAT32, and more than one Xstation main board (QSB was the same). Downgrading to 1.02 resolve the issue for me. Playstation model is 5501.

ramapcsx2, 2020-11-07:
Hey,
how did you downgrade if the xStation didn't boot?
The 1.0.4 update fixes a few bugs that could have prevented booting.
I suspect a different problem.

ramapcsx2, 2020-11-07:
Try these steps:
Attach your real drive's 4 pin cable to its connector and power up the console. With the lid switch closed, the spindle motor should come on, and stay on. If you open the lid, the motor should stop.

Make sure that the PSU reset button is working. When you touch it slightly, the console should not reset. When you press and hold reset, the console should not start up (show any image).

ramapcsx2, 2020-11-15:
Stale? Comment again and I'll reopen.

closed 2020-11-15 by ramapcsx2