x-station/xstation-issues #12: XStation not loading or hanging intermittently

darcagn, 2020-11-11:
Hi,

I have installed XStation units successfully into five SCPH-5501 consoles so far with 0 problems, but I'm having issues on one unit in particular, my personal PS1 SCPH-5501 console with PS1Digital installed. 
So far I am seeing the following:
- Sometimes it boots up and works just fine!
- Sometimes stays at the white Sony Computer Entertainment screen for a long time, then goes to CD player
- Sometimes hangs at the white Sony Computer Entertainment screen for a long time, then finally boots
- It may hang at the PlayStation License screen
- It may hang at the XStation loading screen
- It may hang in-game

Things I have tried:
- Different XStation main board
- Different ribbon cable
- Different SD card
- Tried both exFAT and FAT32 filesystems
- Reflowed the pins on IC701
- I have disassembled the console and reflowed the QSB connections countless times, at least 20+ times
- Tested 1.0.2, 1.0.3, and 1.0.4 firmwares

I am now sort of at a loss as to what I should do next.
However, I am starting to wonder if the problem is not my installation. When I installed the PS1Digital, I removed a Mayumi chip so that I could use the PSNee functionality built into the PS1Digital.
The functionality of the PSNee was very very unreliable, and I had similar symptoms to what I am experiencing with the XStation, and I found out from chriz2600 that the implementation was buggy.
This was fixed in PS1Digital 1.1 update on 2020-09-28 and loading games worked great after that. 
Is it possible that the XStation's implementation of PSNee is causing similar problems to what was happening on PS1Digital before the 1.1 update? 

Thanks for your time, and thanks for the XStation -- in the other consoles, it works great and is a fantastic product.

ramapcsx2, 2020-11-11:
Hey,
first thing to verify is that you're *not* using the PSNee functionality of the PS1Digital.
The xStation does this job, and it has to do it alone, or else there will be conflicts.

Next, are you using an SD card extender / 3D printed mount? If so, please try without it and see whether that helps.

Also, the lid switch has very little range of contact. You could try to force it down (lid opened, a small tool wedged in so that the lid switch is always pressed).

Please tell me when any of these suggestions helped :)

ThanosSiopoudis, 2020-11-11:
@darcagn try the following:
- remove the 3d printed HDMI port cover from the back
- remove the SD card extension if you are using one
- re-assemble and make sure the screws at the back are screwed in tightly

When you talk about the PSDigital PSNee implementation, did you wire the 4 wires for it on the PSDigital? If you did, you need to remove this, the XStation takes over this task as @ramapcsx2 mentioned

darcagn, 2020-11-11:
Hi guys, thanks for your responses.
- I had removed the PS1Digital's four PSNee connections prior to installing the XStation. 
- I am not using a 3D printed mount at the moment, the one I want is still in the mail.
- I do not believe the lid switch is at fault because the green light on the XStation is sometimes lit during failure, and turns off when I lift the lid switch. To be sure that this is not at fault, I traced the lid switch signal to a via on the top of the board and wired a toggle switch to it so that I can manipulate it at will. With the switch in the ON position, the fault still occurs. 
- The 3D-printed HDMI panel has been removed and all screws are very secure. The problem still persists in this configuration. 
- The motherboard I am having problems with is a PU-18 1-664-537-52 board. The five boards I installed successfully were PU-18 1-664-537-62 boards. I don't know if this could be a clue.
- I recapped this motherboard and PSU in August. I have played PSX a lot since getting the PS1Digital and it has been working fine. I reflowed the caps near IC701 just in case that had anything to do with it. That changed nothing. 
- The console gets quite hot. I suspected thermals, but I tried running the console with the top removed and a large fan blowing on it. The shield is cool to the touch with the fan. That changed nothing. 

I made some videos demonstrating the issue while trying to play Final Fantasy VIII. Maybe the pattern of the access light will give some clue. 
XStation loads up fine, but hangs in game: https://youtu.be/DkWulkFtMbA
XStation struggles to load, then works, but hangs in game: https://youtu.be/e9JqgQJKxE4
XStation struggles to load, then works, but hangs after loading FF8 save file: https://youtu.be/saPYKERE1uo

ThanosSiopoudis, 2020-11-11:
Ok, the hot PSU might be a clue.
Can you try swapping in a PSU from one of the consoles that don't have this issue?

darcagn, 2020-11-11:
I swapped out the PSU with one of my working consoles. The working console still works, the problematic console failed on the first attempt to boot. :(

ramapcsx2, 2020-11-11:
In FF8, does the long intro video play flawlessly?
The game hates even the slightest delay in the intro. It would lead to a noticeable stutter.

darcagn, 2020-11-11:
Booted the console, it went to BIOS menu, then after a moment of waiting it started XStation menu. I loaded FF8 and started a new game, it played flawlessly with no stuttering at all, but then hung when Rinoa releases the feathers. The access light was still blinking fast as if it's still attempting to transfer data, but the screen was hung.
After it sat there for about 2-3 minutes hung, then it began playing the rest. No stutters, but then it hung again right at the climax with Seifer slashing Squall's face. Same with the access light blinking. After another 2-3 minutes hung, it played a few more frames and hung again. Then I turned it off.

ThanosSiopoudis, 2020-11-11:
Can you post a photo of the QSB, the APLL wire and the lifted pins on the DSP?

ramapcsx2, 2020-11-12:
Hm this sounds 99% like the SD card stall I get with the Lexar card. It works really well for a while, then starts to have "tea time", where every sector is 100+ms delayed.
Please try yet another card with the same FF8 intro.
I know the intro plays perfectly fine, if the data flow doesn't stall.

ThanosSiopoudis, 2020-11-12:
@ramapcsx2 I was also getting this behaviour when I forgot to remove the APLL resistor

ramapcsx2, 2020-11-12:
True, any installation mistake could look similar.
I've decided to add an SD card read test to the firmware that can run over an image file and report any issues when done.
It should help narrow down where the problem is.
(Installation pictures would still help, @darcagn :) )

birdybro, 2020-11-12:
@darcagn 
If you want a really good rom to diagnose read issues on the mainboard or with the mod having been done, please use Xenogears as well. Xenogears will very quickly show any problems with detecting the CD-ROM more so than most. It was hard for PSIO and multiple emulators to handle correctly for quite some time. Make sure to use redump validated roms only when testing this out, as anything else can introduce issues, especially patched roms. At least for testing purposes.

I will add that SD card quality matters a great deal for sensitive stuff like this (from my career anyways). I typically advise either Samsung or SanDisk SD cards for anything that is sensitive to read inconsistencies. I have had major issues in the past in all sorts of cases with Kingston, MicroCenter, etc... Also, lots of people end up being tricked into buying counterfeit MicroSD of these main brands, even from Amazon, but especially from eBay. So keep that in mind. 

Also as your MicroSD's gets larger in capacity, it is theoretically more prone to error, but only if it's on the bleeding edge. A 512GB MicroSD for instance today, might be a lot more prone to error than a 256GB MicroSD, however there is a negligible difference in reliability from everything on down because continuous improvement in manufacturing process has a longer history for 256GB or less than for 512GB.

Just some food for thought.

My IT experience + longtime emulation&retro user experience guesses are: A) faulty CD-ROM tray switch B) MicroSD read inconsistency C) bad ROM dump D) user installation error/bad solder joints/etc...

mastaksf, 2020-11-12:
Hi there. I installed xstation into my 5501 and worked fine with testing games and playing for about an hour then lost visual. System lights were still on but system was pretty hot. Any clue where to start?

darcagn, 2020-11-16:
@birdybro Appreciate the advice, but the tray switch has already been eliminated as a variable by permanently grounding the lid detect line (aside from access light behavior), the microSD has been eliminated as it works fine in my other (now 7) XStation installs that I've stress tested for an hour each (and I've tried other cards as well that all work on the other units), and it's not bad dumps for the same reason (not to mention the fault can occur before dumps are accessed). 

I have pretty much isolated it down to a faulty install, but wanted to make sure there was no issue with the code as I pulled my hair out trying to get the PSNee functionality working properly on the PS1Digital and in the end it turned out to have shipped with faulty PSNee code causing problems with this revision motherboard that needed a patch and it wasn't disclosed at shipping time, and these projects share the PSNee ESP32 code base. I have since confirmed with another modder this weekend that the XStation code is working properly with this rev motherboard, so that pretty much leaves the installation as the final variable. PSX motherboards are a dime a dozen and I have spares, so I'm just going to restart this install fresh on a new board. I'll go ahead and close out this issue, thanks everyone.

closed 2020-11-16 by darcagn

ramapcsx2, 2020-11-17:
Okay, thanks for the detailed report!
If it helps any, the PSNee code in xStation is robust, since the xStation itself determines the virtual read head position.

I agree that this all sounds like an installation issue.
However, it could also be things like an aging PSU reset / power switch that constantly fires short reset pulses, or an issue with the lid switch and similar things.
Debugging this can be quite the adventure.

Vlad-00, 2021-01-17:
> Hi,
> 
> I have installed XStation units successfully into five SCPH-5501 consoles so far with 0 problems, but I'm having issues on one unit in particular, my personal PS1 SCPH-5501 console with PS1Digital installed.
> So far I am seeing the following:
> 
> * Sometimes it boots up and works just fine!
> * Sometimes stays at the white Sony Computer Entertainment screen for a long time, then goes to CD player
> * Sometimes hangs at the white Sony Computer Entertainment screen for a long time, then finally boots
> * It may hang at the PlayStation License screen
> * It may hang at the XStation loading screen
> * It may hang in-game
>   (When it goes to CD player, or hangs in game, if I just let it sit there for a couple minutes, sometimes eventually the XStation does load and sometimes games do recover)
> 
> Things I have tried:
> 
> * Different XStation main board
> * Different ribbon cable
> * Different SD card
> * Tried both exFAT and FAT32 filesystems
> * Reflowed the pins on IC701
> * I have disassembled the console and reflowed the QSB connections countless times, at least 20+ times
> * Tested 1.0.2, 1.0.3, and 1.0.4 firmwares
> 
> I am now sort of at a loss as to what I should do next.
> However, I am starting to wonder if the problem is not my installation. When I installed the PS1Digital, I removed a Mayumi chip so that I could use the PSNee functionality built into the PS1Digital.
> The functionality of the PSNee was very very unreliable, and I had similar symptoms to what I am experiencing with the XStation, and I found out from chriz2600 that the implementation was buggy.
> This was fixed in PS1Digital 1.1 update on 2020-09-28 and loading games worked great after that.
> Is it possible that the XStation's implementation of PSNee is causing similar problems to what was happening on PS1Digital before the 1.1 update?
> 
> Thanks for your time, and thanks for the XStation -- in the other consoles, it works great and is a fantastic product.

Hello! How are you? I'm facing the issue with the same exact ps1 board. Did you solve the issue? Any clue of why this behaviour?

darcagn, 2021-01-17:
Unfortunately I never did determine why this problem was occurring. In the end I used hot air to remove the PS1Digital flexes and I moved everything over to a PU-18 1-664-537-62 motherboard, using a different QSB. I've done nine XStation installs on -62 boards without a problem, but I haven't gotten another -52 board to install on. 
The only suggestion I can give you is to use a multimeter to make sure that none of the solder connections are touching each other or are shorting to any other signal on the board, including under the QSB. And make sure you have continuity between the legs of the ribbon connector and all alternative points listed in the miniFAQ of this github repo. 
https://github.com/x-station/xstation-issues#test-points

Vlad-00, 2021-01-17:
> Unfortunately I never did determine why this problem was occurring. In the end I used hot air to remove the PS1Digital flexes and I moved everything over to a PU-18 1-664-537-62 motherboard, using a [...]

I tried all the test points and everything looks fine. Can this be the exception to the SCPH-550X compatibility? The thing is, how do I know (I mean, buying another 550X) that is not another 52 revision? :( I feel sad, I tried everything, but nothing works...

ramapcsx2, 2021-01-17:
@Vlad-00 Every PU-18 revision is fully compatible. So the problem is elsewhere for sure.

Vlad-00, 2021-01-17:
> @Vlad-00 Every PU-18 revision is fully compatible. So the problem is elsewhere for sure.

The problem is, where? I'm testing everything, and I read all the tests that @darcagn did and the issue remains. But with the 62 revisio  everything is ok.

miguelk5, 2021-06-23:
@darcagn

Hello I had this same issue after try to repair one Xstation QSB which were bad installed apparently after that the xstation cache or firmware were damaged, after upgrade the firmware the problem is solved.

gtsamour, 2021-09-01:
The games **All Star Racing 1 and 2** (USA latest redump images) seem to have an issue with dual shock controllers (I have a PAL console).
There seems to be intermittent Start button presses without me doing anything, in menu and during game, which causes the game to pause.

I tried a Dual shock PSOne controller and a dual shock PS2 controller and with both its the same thing. All of my controllers pass the Analog Controler testing program so they work fine, and with all the other games.

But when I use a non dual shock (old) PS1 controller both games behave right. I searched online and there are no issues reported with these games so it can't be a game bug.

Matt90116, 2022-02-04:
I was lifting pins on IC701 and when I got to 50's I managed to lift on IC308 not noticing. I went ahead and pushed them back and reflowed not installing the xstation yet and I tested it to make sure all was correct. If the xstation isnt installed after this point will you still get sound? The boot menu takes 10 seconds now and I dont get sound anymore. I hope once I install the xstation that will change because I reflowed those 3 pins back nicely, but no difference from before separating all the pins when I had sound