x-station/xstation-issues #13: XStation not booting into menu; Getting 'Please insert PlayStation CD-ROM.' screen.

ranfoxconn, 2020-11-11:
I have tried two different SD cards and I have resoldered the QSB and checked with a multimeter to make sure everything is making contact. I also tried all the available loader releases but nothing changes this. I tried letting the system boot to the BIOS and closing the lid but that doesn't help. I haven't been able to try with another XStation or PS1 either to see if this persists.

ThanosSiopoudis, 2020-11-11:
Hey there,

It'd be better if we could see photos of the install to be able to help better.
Also, make sure the card is properly set up. You need to format it either as Fat32 or Exfat, and put the loader.bin file in a folder called `00xstation`.
The lid switch needs to be held down in order to boot to the menu.

Can you also please let me know what the LEDs are doing on the xstation?

ranfoxconn, 2020-11-12:
Hello,
Holding down the lid switch the power LED is on and the access LED blinks twice once the system powers on but that's it.
I did try formatting my SD cards in both FAT32 and exFAT though this didn't have any effect, and the loader file was placed in the 00xstation folder.
Here's some images of the install. I had to run some wires and some component legs because I had messed up when I was resoldering the QSB back onto the system and some pads came off. I was able to connect them to where they needed to go and the connection appears fine after checking with a multimeter. The areas I circled are also connected to ground from what I can tell. Not sure if they're all supposed to be but it's something that I've been wondering. 

[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2020-11-12:
Oh gee xD
Are you located in Europe, or otherwise willing/able to send me the whole board + xStation?
I'll fix it ;p

skykiller30, 2020-11-16:
I was going to post my own issue but I wanted to ask two questions first. I botched my first install and when trying to re-do the QSB i lifted traces on the original Mobo. I have since gotten another PU-18 mobo and after installing the QSB a second time i still have an issue. I want to rule out if I damaged the QSB.

Question #1: same as OP... I also have continuity across the points in red circled in OP picture. Is that correct?

Question #2: Is there supposed to be continuity from the solder point closest to 104 (see picture below) to the back of the ribbon cable header? If so, which pin on the header? 
[image]

I was able to find continuity from the ps1 mobo to the back of the ribbon cable header for every solder point except the point by 104. I can find continuity from 104 to the ps1 mobo, but not from 104 to the ribbon cable header.  Im hoping i can run a bodge wire on the QSB to fix a QSB trace i may have killed

Thanks for your help!

ramapcsx2, 2020-11-17:
The circled in red point should go to the flat flex connector pin 7 from the right.
For the points that op circled, those should all be ground. That is correct.

ryanm101, 2020-11-22:
I get this issue with this memory card [SanDisk HIGH ENDURANCE 128 GB Class 10, U3, V30](https://www.amazon.co.uk/gp/product/B07NY23WBG/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) currently going to reformat and try again. 

however my [Toshiba M203 32Gb U1 I](https://www.picstop.co.uk/microsd-sdhc/toshiba-exceria-m203-micro-sdhc-100mbs-class-10-card-32gb.html) card works fine.

ramapcsx2, 2020-11-22:
Okay, it would be weird for that Sandisk card to not even boot the loader.
@ranfoxconn any update?

ryanm101, 2020-11-22:
1st format was Win10 format exFat with 32K Blocksize -> didnt work
2nd format:
-> Diskpart -> clean 
-> sd card formatter -> full format 

2nd format works now boots to XStation. 

@ramapcsx2 I would assume it couldnt read the loader from the sd card hence going to please insert disk. 
Perhaps an update which puts a small part of the 'menu' in the ESP would tell us if the issue is with Xstation or the SD Card.

ramapcsx2, 2020-11-23:
I have some ideas, yea :) 
Next firmware update will increase the SD card drive strength, which may improve the situation here as well.

ramapcsx2, 2020-12-02:
@ryanm101 was this the issue you fixed? :)

ryanm101, 2020-12-02:
This is working for me fine now.

ramapcsx2, 2020-12-03:
Great :)

closed 2020-12-03 by ramapcsx2