x-station/xstation-issues #14: Reset to x-station menu doesnt seem to work

orcutt989, 2020-11-12:
I've seen mention of holding the reset button for 3 seconds and even 1 second, but no matter what it always results in a full console boot. How do we get to the x-station menu without a full console boot?

ramapcsx2, 2020-11-13:
A short reset press simply restarts the console with the last played game still active.
A long reset press switches the mounted image to the loader, but still resets the console as well.
Short reset brings up the last played game while long reset brings up the loader. That is the only difference :)

orcutt989, 2020-11-13:
Ah thanks. So there is no way to get to the menu while in game?

nicolasnoble, 2020-11-13:
It's not physically possible.

nicolasnoble, 2020-11-13:
Well, never say never, but this may be really dodgy to do.

ramapcsx2, 2020-11-13:
No good ideas for now though, sorry :p

closed 2020-11-13 by ramapcsx2