x-station/xstation-issues #16: Overwrite game name list-entry

ArcadeTV, 2020-11-18:
I'd love to see an optional game-name overwrite e.g. with a textfile. I am re-using my sd card that I used with another PS1 ODE and my xstation menu lists all games as "DISC" with no way to distinguish the games. If the list-entry was generated off parent-folder or  maybe a textfile this would resolve this issue.

birdybro, 2020-11-18:
Which ODE that you are using requires the filename structure to be like this for example:

`\Crash Bandicoot (USA)\DISC.bin`

I'm only aware of PSIO as the other and they don't have this requirement.  It might be possible for someone who is clever with a scripting language to write a script to just fix filenames based upon folder name I'm sure. You also could use a PSX Redump DAT file combined with a utility like romcenter to automatically rename your collection to the standard naming scheme.

ArcadeTV, 2020-11-18:
Yes, I wasn't sure if it was okay to write its name, also I didn't meant to be comparative. I'm pretty sure when I've set up the psio the requirement was exactly this scheme, they might have dropped it in the meantime. Yes, a script would help renaming all files and restore names in the x-station gamelist, but it wouldn't help in having a single sd card which works on both.
If you're correct about the non-necessary naming convention then the issue would be resolved.

ramapcsx2, 2020-11-19:
Would be great if you could test keeping the regular names on PSIO.

birdybro, 2020-11-19:
> Yes, I wasn't sure if it was okay to write its name, also I didn't meant to be comparative. I'm pretty sure when I've set up the psio the requirement was exactly this scheme, they might have dropped [...]

Given the documentation on PSIO that I was reading I didn't see any naming convention differences. 
[image]

ArcadeTV, 2020-11-19:
Then I'm sorry. It's been long since I filled the card for psio and must have mistaken my filenames as a convention.
I will rename them and test if they still work on both systems then.

tjanas, 2020-11-21:
On my PSIO, for multidisc games, I typically name them “Disc 1.bin”, “Disc 2.bin”, etc or something like “Bonus Disc.bin”, which are referenced in the multidisc.lst file, so that when I select the parent directory and it prompts me which disc to choose, it can be a simple display name.

ArcadeTV, 2020-11-22:
I revisited both filesystems today and have to admit that using a single card doesn't make sense due to the prorietary cu2 format that the other ODE expects. I re-organited my x-station card to have genre folders, I'd love to see the loader make use of such a structure some day. Thanks everybody involved in this great project.

closed 2020-11-22 by ArcadeTV

ramapcsx2, 2020-11-22:
Folder based lists will come in an update :)