x-station/xstation-issues #17: Issue renaming BIN and CUE and menu display

whitezombie2000, 2020-11-18:
I personally don't like really long file name like this:

Resident Evil 2 - Dual Shock Ver. (USA) (Disc 1) (Leon).cue
Resident Evil 2 - Dual Shock Ver. (USA) (Disc 1) (Leon) (Track 1).bin
Resident Evil 2 - Dual Shock Ver. (USA) (Disc 1) (Leon) (Track 2).bin

If I go and and rename the files and rename them within the cue file they don't display properly in the menu after refreshing.

Resident Evil 2.cue
Resident Evil 2 Track 1.bin
Resident Evil 2 Track 2.bin

FILE "Resident Evil 2 Track 1.bin" BINARY
  TRACK 01 MODE2/2352
    INDEX 01 00:00:00
FILE "Resident Evil 2 Track 2.bin" BINARY
  TRACK 02 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00

After refreshing the menu they show up like this in the X-Station menu:

Resident Evil 2 Track 1
Resident Evil 2 Track 2

This isn't big deal or anything but I don't have this issue with PSIO or Mednafen. Can this be fixed?

birdybro, 2020-11-18:
What does it do when you put parenthesis around Track _x_?

`Resident Evil 2 (Track 2).bin`

For example.

whitezombie2000, 2020-11-19:
Yes that works by adding the parenthesis around Track x. Will it remain this way or will you change it? It doesn't matter to me but thought I would ask. Thanks for helping me out!

ramapcsx2, 2020-11-19:
Yes, the parenthesis are important.
As a programmer, I need to watch out to not filter out valid items.
Consider the example of "International Track & Field" :)

closed 2020-11-19 by whitezombie2000