x-station/xstation-issues #18: XStation resets to game - not to game list even if I set it to reset to menu

valippo, 2020-11-20:
I installed my XStation yesterday, updated to the latest firmware 1.04, set it to reset to menu and saved settings.
I tried a few times and it worked.
Now, after 24 hours, I'm playing games but it always resets to the game, no matter how long I keep pressing the reset button.
The only way to go back to the game list is to turn the console off and on again.

ramapcsx2, 2020-11-21:
The boot to menu option is meant for cold boots, so that is working for you, I believe.
What doesn't seem to work is the reset signal detection. It may be that the reset button is just dirty (try toggling it several times, use a drop of contact cleaner on it, if you have some).
The reset signal solder connection on the QSB could also have no contact. It is the one signal below the APLL point: https://i.imgur.com/rTjoyKg.png

valippo, 2020-11-21:
Hi Robert, thank you for your quick reply.
I reflowed the solder on that point, it looked like it fixed it at first but then it went back to the previous behaviour.
I then speayed some isopropilic alcohol on the reset button, pressed it a few times and let it dry and... it is working now 100% of the times.

For those of you with the same issue: when I was holding the reset button before I had a black screen with a random line with dots while I was holding the button.
Non while I hold the button the screen stays black until I release the button. That’s where I see the dotted line on the screen, I believe the console stops the video signal for a moment, then sends the signal again to the tv.
So checking for weird stuff on the screen while holding the reset button might help people identify wheter or not their reset button is working properly.
I forgot to say that I am using a CRT tv with my RGB cable, I don’t know if this can be noticed on an LCD or using different cables.

ramapcsx2, 2020-11-22:
Cheers :)

closed 2020-12-02 by ramapcsx2