x-station/xstation-issues #19: Pausing for a while can cause hanging

ProfgLX, 2020-11-21:
Exemple of this would be Castlevania SOTN, in the intro stage, when pausing for a while (few minutes), unpausing won't restart the music. When new sounds should be streamed (richter and dracula talking), no speach can be heard. When the next cutscene is supposed to start, the game hangs.

It seems like the "CD drive" isn't "spinning" back up, so no new data is read.

Will try to recreate on a console with a CD drive to see the difference in behavior.

birdybro, 2020-11-21:
What microSD card are you using? There have been possible issues with certain brands.

ProfgLX, 2020-11-21:
Sandisk Ultra 256gb U1/A1

jonasongithub, 2020-11-21:
Does it happen every time and how long is a few minutes? I paused SOTN during the prologue for 15 minutes and when I continued the music started back up, dialogue was also played as usual.

ProfgLX, 2020-11-21:
Tried 25 minutes yesterday, will retry with less time to figure out at what point it stops!

jonasongithub, 2020-11-21:
I cannot replicate the Xenogears bug either. So we're on the same page: I queued up disc2 and disc1, booted up disc2, selected new game, opened and closed the lid to switch the disc and let the entire intro play without it locking up.

ProfgLX, 2020-11-21:
Just retried it, 15 minutes was OK, 20 minutes was problematic.

I highly doubt it has anything to do with it, but I can try to rehit some of the solder points...

birdybro, 2020-11-21:
Can you take a high def picture of both sides of your motherboard for us while you are in there?

ProfgLX, 2020-11-21:
[image]
[image]

ProfgLX, 2020-11-21:
Pins on top of the picture look weird on the CXD2545Q, but it's due to the angle, they're straight and don't touch.

birdybro, 2020-11-21:
Can you include a picture of the wire on the right, where it leads? Thank you for the pictures.

ProfgLX, 2020-11-21:
[image]

I didn't open it up yet, these pictures were taken when I installed it.

I'll open it up a little later, I have some work to do around the house first, Thank you for your time :)

ProfgLX, 2020-11-21:
Oh, update on the Xenogears problem, retried it with a 32 gb sd card and wasn't able to recreate. That might've been the SD card. The one with castlevania is repeatable though. I'm gonna try it without pausing to see if it's problematic this way.

birdybro, 2020-11-21:
Thanks for providing those. From the picture, it maybe looks like you may have some cold solder joints here and there. Make sure to use a good amount of flux when you touch up the spots, and maintain some light pressure to ensure the board is kinda low. 

The pictures are a bit blurry so it's hard to tell if anything is wrong with the soldering for sure. 

A couple other things. Did you use the SD Association Formatting Tool? How full is your MicroSD? Do these roms match the redump hashes? Etc... But given it happens over those two games I doubt it's the rom, but never know.

ProfgLX, 2020-11-21:
Will do, but I doubt (and I'd be embarrased) to have cold solder joints 😂 It's sitting flat, the light source in the picture isn't directly over the board, the black lines are shadows.

Yeah, it's the SD association tool. The 256 gb was around 60% full. The 32 gb one only has SOTN and Xenogears on it. Iso source is Redump on archive.org.

I'll try everything and report back later in the day, and retake pictures of the board!

birdybro, 2020-11-21:
Yup, sounds like everything on that front is right. My next guesses would be bad capacitors, maybe pots on PSU and board got nudged, etc... Hard to tell. Definitely outside my level of expertise at this point so hopefully others can see what is going on. Good luck! :)

ProfgLX, 2020-11-21:
Update: Tried just leaving the game on without pausing, eventually the music stops, speach doesn't trigger, cutscene after beating drac doesn't trigger. Pausing the game doesn't have anything to do with it. It's really like the drive goes to sleep. 

Still gonna try hitting the solder points again later in the day.

ProfgLX, 2020-11-21:
Rehit all solder points, retried everything. It took approximately 23 minutes before the music cut abruptly, it's not even when the music loops. I'm at a loss.

jonasongithub, 2020-11-21:
I have now let SOTN run idle during the prologue for a good 45 minutes and the music was still playing when I returned. Speach was also correctly triggered.

ProfgLX, 2020-11-22:
Just retried with a 3rd SD card, problem occured at 30min 24 seconds mark. This time it was a Gigastone 64gb A1 U1. Did the long formatting with the SD association tool, only put the 00xstation folder and game on the card. Surprised no one else has this bug, assuming of course that it's a bug.

I'll reopen the PSX and recheck that every point has continuity... I'll also make a test without the card holder to see if pressure is applied and could cause it.

ProfgLX, 2020-11-22:
Some more results, with good news!

I tried it once again, same test as before, but this time without the 3d printed SD card mount. No problem whatsoever after 70 minutes (of course this happens when I'm recording...). I'm gonna retry it with the SD card mount on the outside of the console. My guess is that there can be 3 causes to this problem : 

A - The SD card extender itself is causing the problem.
B - The mount puts pressure on the board, when there's enough heat built up, this causes the glitch (connexion makes a bad contact or something due to thermal expansion/bending and physical pressure).
C - The mount itself causes too much heat to build up.

My money at this point is on cause B. I will run some more tests to confirm, then I'll close the thread (and rename it) for future reference.

tjanas, 2020-11-22:
Who did you buy your 3D printed mount from?

ProfgLX, 2020-11-22:
Should be my last update. It's heat. Simply heat.

Test #1 : I tested the XStation directly without SD card mount. Result: 1h10 without any problem.
Test #2 : I tested for cause A by taking the mount out of the PSX, closing the lid over the cable. Result : 1h08m without any problem.
Test #3 : I then tested for cause C by keeping the SD card mount out of the PSX, but placing a cardboard to close the drive hole. Result : Problem at 47m10s mark. (That ESP chip was really hot to the touch)
Test #4 : Same conditions as test #2, but tried it as soon as I could after test #3 as I was curious if it would fail. Result : 1h25m test without failure. ESP32 is warm/hot to the touch, but nowhere as near as test #3. 

I wonder what RamaPCSX2 thinks about this?

If it doesn't happen to anyone else, maybe it's my unit, or we're gonna hear about people having problems with long game sessions if they install a mount. I'm just happy to have found my problem.

3d printed mount was from Greg at Laser Bear. The guy makes beautiful products. Just too bad I won't be able to leave mine in.

ProfgLX, 2020-11-22:
I have to note that I also have a PS1digital installed, which might also contribute some additional heat, but it's doubtful as it's under the metal shield of the PSX.

ProfgLX, 2020-11-22:
Just did an 8.5 hour test without sd mount to see if it would fail  after more than 1.5 hour and recorded it. No problem, no cut in audio, could beat dracula and have next  cutscene start without problem. Esp32 is warm/hot, still nowhere near as hot as when I had the drive hole covered.

ramapcsx2, 2020-11-22:
Okay, so I think it's a tie between the SD card extender causing signal issues, and heat buildup.
All that hardware will be heating up the system, as ventilation isn't good to begin with.

Could you test placing the console in a colder place and with better ventilation?

ProfgLX, 2020-11-22:
I don't think the extender is problematic, unless you mean the extender itself is the one causing problems with the heat, which I haven't tested. Ambient temperature is a constant 69 degrees here.

I'll do two extra tests which should help figure out which one is the cause:

1 - No extender, card directly in Xstation, covered drive hole.
2 - If problem with test 1, retest but this time with a fan pointed directly at the PSX.
3 - If no problem with test 1, retest with another card extender.

If you have any other condition you want me to test, please tell me!

ramapcsx2, 2020-11-22:
These tests sound good. I would start with the fan test though. 
If the fan fixes is, there will be little speculation on what exactly caused it.

ProfgLX, 2020-11-22:
I already started the first test before reading your message. Still working after 1h10m. I'll let it run another hour, if it still works, It'll confirm it's the extender that causes the problem!

ramapcsx2, 2020-11-22:
Okay, I can soon (in 1-2 days) test a couple extenders with the oscilloscope. I'll be able to say more then :)

ProfgLX, 2020-11-22:
I let it work for 2.5 hours and it worked without failing.

Currently testing with fan, blowing on PSX, I'm half an hour in. After that, I'll retry without fan.

It might be the extender I got that's failing due to heat, maybe it won't happen to anyone else. I'm really glad the Xstation is out of the possible causes :) 

I have an oscilloscope if you want me to test what happens when I recreate the problem. I'd need the test points and trigger type.

ramapcsx2, 2020-11-22:
If you want to help, please check the signal quality of the various fast signals on the SD card interface.
The signals should not degrade too much when going through the extender cables.

This can be quite tricky to properly capture.

ProfgLX, 2020-11-23:
I'll try and check it out. I recreated the problem, but only when the extender is present, it might simply have a connection that breaks due to heat.
The Xstation itself is fully functionnal, thanks for the help and suggestions everyone!

ramapcsx2, 2020-11-23:
I've now scoped the signals with an extender: The clock is a bit weak and might easily be not sufficient in some setups.
Next firmware update will have an increased drive strength. It fixes the issue :) 

I can't guarantee that this is the problem we see, but there is a good chance that this easy fix solves it :)

darcagn, 2020-11-23:
I am also using PS1Digital + XStation + LaserBear 3D mount as well. I hope I'm not raising a false alarm here, but it appears this combo could be deadly to the console. 
I wanted to test the same scenario as @ProfgLX so I started Castlevania SOTN and let the intro play, and once gameplay started I left the console just sitting there for 90 mins with no problem. I was satisfied that I wasn't having the same problem as @ProfgLX, but I had to take a phone call, and upon returning to the PlayStation, the screen was garbled (in a manner that appeared to be from the console faulting, not the PS1Digital).
The console will no longer boot: I get no video over digital or analog, but I hear the intro begin over the speakers, but it hangs on one of the notes and just drones on. I disassembled the console and let it cool down, but the same thing happens. I reflowed the GPU flex and the XStation points just in case there was some issue there, but no change.
I removed the PS1Digital flexes with hot air and reinstalled in another PlayStation with XStation and that is working fine.
Given that I observed the console getting very hot prior to receiving the 3D printed mount, the mount may just suffocate things enough to cause failures. 
I don't think I'll feel secure with these mods installed together, even if I removed the 3D printed mount, unless I add some additional cooling, so I am going to see if there's enough clearance in there to rig up a small quiet Noctua fan.

ramapcsx2, 2020-11-26:
@darcagn I can only imagine that the heat may have disconnected one of the fine GPU pins from the motherboard.
This wouldn't be from melting any solder, but rather from slight warping + some other factors..
It's all speculation though, so it's hard to give out recommendations now.

I've thoroughly tested the xStation installed alone, and this is fine. The hottest spot is the LDO that gets up to 58C if the lid is closed. Ambient temperature around it drops to 45C, all normal.

As an aside, the new firmware with tweaked SD card drive strength has been released.
It would be nice if everyone could retest their setups :)

birdybro, 2020-11-27:
@darcagn what's the ambient temperature where you are using your playstation? Also have you seen the 5v fan mods before? Seems relatively straight forward to pull 5v from the PSU and put a fan or two in that blows across the side. I've also heard of people getting heatsinks before. You are right that adding these two mods will change the thermals beyond what was intended. Each mod individually might not pose an issue. Just my two cents. Here's an example of someone doing the mod. 

https://youtu.be/HSzCs60izbM

ramapcsx2, 2020-12-02:
I'm assuming the original issue is fixed with the SD card drive strength adjustment in 1.0.5.
If heat remains an problem, it warrants a separate issue.

closed 2020-12-02 by ramapcsx2

ProfgLX, 2020-12-03:
Sorry for not replying, Greg sent me a new mount, I've been waiting for it to be delivered. Will compare after, but the culprit was obvious at this point. As for heat, will try another PSU to see the difference. Thanks again!