x-station/xstation-issues #22: No

DarrenRaptis, 2020-11-24:
After installing xstation, pressing power button flashes a blue screen stating loading games followed to a new blue screen stating 
(xStation no games found. press (triangle) to refresh list!)

I am unable to get games recognized or get to main loader screen.  I resoldered the the board, checked connections, read tutorial on how to add games to sd card to no avail.

jonasongithub, 2020-11-24:
Just to get the obvious out of the way: did you go to the settings and refresh the list? XStation doesn't do that automatically, it just loads a prebuilt list on startup which you don't have after setting up your sdcard.

DarrenRaptis, 2020-11-25:
Thank you for the help. This will sound ridiculous but after many hours of troubleshooting the problem was a faulty controller with a non functioning triangle button :(    On a positive note though, I love the product and thank the creators and community for keeping the history of gaming and consoles healthy and fun. Thanks again.

ramapcsx2, 2020-11-25:
Murphy's law strikes again! :p

closed 2020-11-25 by ramapcsx2