x-station/xstation-issues #23: Feature Requests

ryanm101, 2020-11-24:
This is just a dump of 'like to haves' if you let me know which are likely to be implemented i can open tickets for those:
* MDS/MDF support would be nice.
* ecm support would be nice.
* Multi disks -> it would be nice to be able to create a '.game' file that holds the name (in order) of CDs so as to make multi disk games cleaner..
e.g.
``` syphon filter 2.game
Syphon Filter 2 D1.bin
Syphon FilRead more about review stating Amazing product a few improvements could make it even better thoughter 2 D2.bin
```
This would remove the need for selecting disks in order and cluttering the screen with Disk 1 and Disk 2
* Enabling Wifi while idling on the Xstation screen would be useful for OTA firmware updates and/or a mini FTP server to transfer Images. (yes it would be slow but i can leave it running and go out for a walk or make a brew. :)
* Support for 7502 :P would also be nice, here's hoping it happens.
* Action Replay like support.
* Maybe include instructions to short the lid open switch or add to the ODE a physical switch to replace it, so that it can be enabled / disabled while lid is open.

darcagn, 2020-11-24:
- AR/GS support can be done now by selecting an AR/GS disc for multidisc and then selecting your game image. But it would be nice to have support for codes directly in the loader so one wouldn't have to physically get up to switch the game image.
- If you want to install your own button/switch for the lid control, you can install one yourself without needing to add to the ODE.
See image here: https://i.imgur.com/TS7hx3W.png
Solder a wire to the via labelled "lid switch" to ground with a small wire in order to permanently have the lid closed.
Install a SPST normally closed momentary push button switch if you want to have the lid signal always closed, and a button press would cycle the lid open/closed in order to swap discs. This might make it a pain to get to the BIOS menu though (would have to keep it held down at boot time). You could use a maintained button instead of a momentary button if you want to set it to stay open without having to hold the button down.

ryanm101, 2020-11-24:
@darcagn 

* AR/GS -> Cool, will have to look at that, i have a XPlorer so might give it a whirl too. 

* WRG to the lid closed I meant:
  * if there was a rocker switch (or transistor) added to future revisions of the board, it would allow people to 'open/close' at will regardless of the lids actual position. Seems silly when written down but thinking more if you wanted to leave the lid open so you could switch memory cards easily.
  *  Yes I could perma do it but that then means i can never get to the memory card / music management screen, (though this could be added to the install instructions)
The solution was really designed at removing the 'problem' of people not closing the lid properly by taking the choice (mostly) out of their hands.

ramapcsx2, 2020-11-24:
I'm reading this and taking notes by the way :)

ryanm101, 2020-11-24:
@ramapcsx2 Feel free to close this and just open tickets for the ones that you are going to work on.. The big one i'd love is the wifi/ftp option.. 

Really Silly ask... But what about MQTT integration (like tasmota).. I could tie this to homeassistant to show what is being played..

darcagn, 2020-11-24:
If we're doing 'like to haves'...
- Speaking of WiFi/FTP, a feature I'd love to see is an SMB or FTP client, so images can be read from network storage. Perhaps the host/port, path, username, password could be stored on a config file on the SD card. When a game is chosen on the PlayStation, it could copy it to the local SD card and play from there. This may be a niche use case, but I have a home server with network storage storing game libraries, so using WiFi instead of large SD cards is something I'd like to see ODE makers explore more as WiFi is coming built in now on many devices. 
- A memory card manager application allowing backup of saves from the memory card to SD card and back again
- Cheat engine, for example: Allow storing cheats in a .txt file with the same filename as the .cue file, press select while highlighting a game title and then have ability to toggle on/off individual codes for that title
- Dither on/off patching
- More colors options for the menu besides blue
- Allow setting the background from an image file placed in the 00xstation folder
- Folder navigation

ramapcsx2, 2020-11-25:
For everything WiFi, keep in mind that the SD card bulk transfer speed will be very limited.
We could be looking at 1MB/s or so.

valippo, 2020-11-26:
I'll add my requests:
- add 60hz/NTSC mode for menu on pal systems
- push menu resolution to 640*480 or the best resolution to improve readability
- add cover art support
- toggle to switch fast boot between square and X buttons

ryanm101, 2020-11-26:
@ramapcsx2 Yea but once you have the card configured on a PC if you only need to add 1 or 2 games every so often it's an easy fire and forget. I see the FTP use case as 99% uploading updates you release, and 1% adding a game or doing some level of file management (deleting games that i dont play etc).
 
Also an idea might be (for future revisions) to expose unused pins on the ESP. So you can expand with a wire (e.g. i could run a wire from Pin3 to the lid switch and pull it high (depending the voltage) to trick teh system into thinking the lid is open. or i could make. my own rely board and use it to trigger reset over wifi or wire it to the controller to detect key combos for a reset.

https://github.com/keirf/FlashFloppy/wiki/Hardware-Mods is a good example (yes it would be a 'do it at your own risk) would improve your modability (i know you are closed source but people could submit requests for enhancements and if you wanted them you could release binaries with the extra capability)

ramapcsx2, 2020-11-26:
We've got 4 somewhat free to use GPIO for this purpose, yes :)

ramapcsx2, 2020-11-30:
Ah, I was just going to paste the feature requests I read in the product review over at Castlemania, but they're all here :D 
- Many of the improvements here relate to a revamp of the file walker / game list display. This is high on the priority list.   
- Lid switch changes are difficult, as it requires extra hardware. Using addons is doable though, as we've got 4 free GPIO.
- WiFi: I really need to get some tests started ;p

ryanm101, 2020-11-30:
Yea they were mine if it was the big list. :)

I think the hardware ones if there is the capability in the firmware and someone is mad enough they can hack them on manually until you update the HW revision.
While the ESP pins are the largest they are not exactly tiny either.

If you are looking for a tester I'm happy to assist.

CountCreep, 2020-12-17:
I would love to see read speed limiting options for different models of PS1 and PS2 FDS.

birdybro, 2020-12-31:
>     * Multi disks -> it would be nice to be able to create a '.game' file that holds the name (in order) of CDs so as to make multi disk games cleaner..
>       e.g.
> 
> ```
> Syphon Filter 2 D1.bin
> Syphon Filter 2 D2.bin
> ```
> 
> This would remove the need for selecting disks in order and cluttering the screen with Disk 1 and Disk 2

The .m3u new "standard" being used by multiple emulators and frontends (mednafen, retroarch, etc...) seems to be the best way tot o do this by the way. Given that there is already an automated .m3u generator out there (https://github.com/tralph3/Cue-Maker), it would probably be simplest to do this instead of creating a new filename extension.

mtmccurdy, 2021-01-30:
Hey Rama, happy user and semi-neurotic update-follower here.  This project is incredible and one of a few that I constantly check up on because it brings me joy.  Every few days I check the latest release number and pop over to Issues to see if there’s been any comments or action that hints at what will be in the next release (card access timing changes, etc).

Any plans for a public roadmap/release plan? I’d love to see where the project is headed next, but also understand if you prefer to surprise people.

ramapcsx2, 2021-01-31:
Hey,
the roadmap is to finalize the next firmware for now, which focuses on SD card compatibility and edge case timings, then after that is all the features :)

I'm occasionally checking the social medias and Github issues, but otherwise I'm on Discord, where I get a lot of feedback directly from the users.

mtmccurdy, 2021-01-31:
@ramapcsx2 awesome! Thanks for the info.

InvaderZimm, 2021-02-24:
I would love a disc swap option for the people like me who glued down the lid switch years ago

ramapcsx2, 2021-02-25:
This can't be done. With a fixed lid switch, there is no way to tell the console that a new disc should be checked for.
You need to undo your old mod :p

willemrosenthal, 2021-03-11:
Virtual Memory Cards that saves to the SD card would be incredible.

msimplay, 2021-03-18:
It would be nice if the console could switch between **NTSC / PAL in software** as there are boot discs that can already do this for games

ramapcsx2, 2021-03-18:
There will be a memory card manager of sorts soon'ish, but we'll have to see what features there'll be :)

A PAL/NTSC switcher is an invasive patch to games. Many titles will break from it.
As such, unless a reliable patch gets developed, I would recommend to use a diskqueue with something like Import Player Light as the first disk.

arkfan, 2021-03-20:
I have an idea for better handling multidisc games : 

-User tries to chain discs together with the multidisk button.
-XStation tries to see if those discs are placed in the same folder.
-If yes XStation offers to save that multidisk "chaining".
-If user accepts a text file is created in the folder containing the discs. This text file would be like PSIO's multidisc.lst files, it would contain the name of the discs and the order of the chaining. XStation may ask to overwrite if a text file already exists.
-Multidisk chaining for that multidisc set becomes no longer necessary.
-The entry for each disc in the menu is replaced by one entry : the folder name.

Users would have to place the discs of a multidisc set in the same folder for this to work though

ramapcsx2, 2021-03-24:
"Kuo" requested a button or hotkey to start a random game.

FreddoUK, 2021-03-27:
**Designated Cheat Disc:**

If we could designate a cheat disc somehow (perhaps by naming it x00_Cheats.bin or similar) I think this could be useful for two reasons:

1. A button in the launcher similar to boot/fastboot could be "boot with cheats"
2. When boot with cheats is use, the correct game title ID (and not the cheat discs Id) could be sent to PS1Digital/MemcardPro.

mtmccurdy, 2021-04-03:
I’d love to see a really rich image-based option for the loader, maybe even with game genre and descriptions and/or sorting by attributes (like genre). If it had the right hooks, the user could provide the images and/or some sort of text or load file with genre and descriptions. I’m sure the community would create packs.

msimplay, 2021-04-03:
> I’d love to see a really rich image-based option for the loader, maybe even with game genre and descriptions and/or sorting by attributes (like genre). If it had the right hooks, the user could [...]

Renaming of games also would be nice I'm not used to the name of some games in their long format

ramapcsx2, 2021-04-03:
@mtmccurdy 
A rich UI with lots of graphics would be a poor experience on the PSX with a microcontroller driving the features.
I'm sure it's possible to have graphics and music, but it would definitely have to be tailored to the console.
You would need some 3D features with animation, 480i flickering interlace to get the detail.. It would be a lot of work.
Uploading data, even simple text (such as the game list), is already quite the challenge. Rich UI features would not make it easier.

In short, I don't think this fits the xStation.
The ODE concept is to be accurate, fast and efficient. It has one job, play games! :)

ramapcsx2, 2021-04-03:
@msimplay 
This is possible. I'll look into file operations like that once folder based browsing is implemented (looking good on that!)

arkfan, 2021-04-03:
ramapcsx2, do you have any plan for better managing multidisk games?

PsychoFox11, 2021-04-03:
As a light gun fan, I'd like to see a few things and figured I'd mention them here as well.

Konami Justifier support - Currently you can't select a game using the Justifier. If playing a game using 2 of them, you need to start with a controller in one of the ports then hot swap, sometimes needing a reset. You can only start the default selected game on startup with the gun itself. The trigger and both buttons just start the highlighted game. It does have a trigger and two buttons though, which each do different things in a game (not duplicate buttons), so it seems it could likely work like the GunCon does at least.

Easier navigation with Guncon - with a really large card, navigating one title at a time with the GunCon A/B buttons takes a while. Again, only an issue when using 2 guns at once, otherwise you can use controller 2 to pick the game. Pressing A+B together on the gun to jump a page would be a big help.

Ultimately, having an on-screen cursor for both types of guns would be ideal, but I'm sure a lot of work. The PSIO supports this for both gun types though so it would be cool to see here some day as I still prefer the hassle and using the X-Station . For now, I'd just like quicker navigation when using 2 guns at once.

I haven't run into an issue so far with not being able to choose Full Load with the GunCon, and I don't think any gun games were multidisc, so while those options can't be accessed, I don't think they are necessary, though I haven't tested every gun game.

lamprey13, 2021-04-22:
I'd like to second a browsable "folder structure" to help organize things instead of one big list.

For example I like to organize my games by Alphabetical and by Genre.

For Genre the folders go something like:

- Action
- Adventure
- Shump
- LightGun
- Sport
-  - Football
-  -  Soccer
-  Strategy

Nested folders (like Sport/Football) would be Ideal. But, even having one level of folders to organize things would be great.

Thanks!

badass-boss, 2021-05-11:
I will add my request:

- make Xstation available to Europe

ramapcsx2, 2021-05-11:
As mentioned in the other issue, this is in the works :)
Ordering from Castlemania is as hassle-free as it gets though, as we take care of import duties.
This means the price you see is what you actually pay in total, as well as fast forwarding the package in customs.

badass-boss, 2021-05-11:
Well at this point this is very interesting indeed. I will consider that then. Great support right here

riggles1, 2021-05-13:
I would love to have the boot sequence (Sony Playstation logo screens) to be 60Hz on PAL region machines too, since even if the menu is set to 60Hz and the game loaded is then played in 60Hz, it'll always play that boot sequence in a squished flickery 50Hz no matter what. Miss having it 60Hz.

Before the xstation when I was just burning discs with a modchip I would get a proper sized 60Hz boot sequence on PAL region machines, so it's not a system limitation.

ramapcsx2, 2021-05-14:
The xStation doesn't do BIOS patches, so no video mode change is possible.
You may have had a modchip with BIOS patch before. The way it works has the side effect of running the startup sequence in 60Hz mode.

If you really can't stand the 50Hz startup, you can replace the BIOS chip.
https://willsconsolemodifications.co.uk/ps1_replacement_bios.php

msimplay, 2021-06-05:
Last game played feature.

Every time I reset the console the list is always at the top but I am in the process of testing my library I would like it if the list started at the last game played the way you can set it on OPL.

It's good for testing in case the game didn't work also

ramapcsx2, 2021-06-05:
Try Start :)

msimplay, 2021-08-11:
> 
> 
> Try Start :)

Lol I think you are not understanding what I am saying basically I tested 200 games and I had to scroll back to the last position on each reset it would be nice just to start where the last game played was.

Also a new request a favourites list? Add to favourites?

Jeffersonian, 2021-08-12:
- Recent game list
- Favorite games list

FXPAK Pro implements these well in the menu.

badass-boss, 2021-08-12:
> > Try Start :)
> 
> Lol I think you are not understanding what I am saying basically I tested 200 games and I had to scroll back to the last position on each reset it would be nice just to start where the last game played was.
> 
> Start will play the last game played but I want to test the next game in the list
> 
> Also a new request a favourites list? Add to favourites?

What you're saying is you want to have the cursor pointing at the latest position after reset. This might be interesting but I'm not in the position to say whether it's possible or not. I wonder if that would work in folder mode as the cursor would have to navigate automatically to the latest folder and position

msimplay, 2021-08-12:
> What you're saying is you want to have the cursor pointing at the latest position after reset. This might be interesting but I'm not in the position to say whether it's possible or not. I wonder if [...]

Yes I didn't know I needed this until I was testing my library of games.
I installed around 2TB worth of games on my ps2 and OPL has this functionality so it was very convenient to have the cursor start at the last game played

ramapcsx2, 2021-08-12:
A recent or favourites list is planned. That list could offer a selection to "jump to location", if it's reasonable to do.
We have much more information available now (at the cost of a little performance), so it might work.

In any case, the new beta is currently getting tested. It has the foundations to make new features work :)

LOLJustCarted, 2021-11-29:
I would love to see three things implemented 

1) a way to organize by genre/tags. Could either be added tags to each game/disk/folder or something like multiple pointers to the same data (like Spyro being in action/platforming/S folders but only one instance of the disk is on the SD card.

2) cover art/descriptions

3) on the fly dithering patching, maybe implemented before starting the disc so switching would need to be done in the loader

BUT regardless xstation is an absolutely amazing product. Thank you for all that you do!

ramapcsx2, 2021-11-29:
Hey,
regarding organization, it's planned to have a full recently played list. Maybe this goes well with a "tag" system that can mark individual games. This could then be extended further.

Cover art wouldn't be very enjoyable, due to the low resolution display and the performance hit, which would be severe.
It could probably be done, but at quite the cost.

De-dither, yes. It's in the firmware but only works with a few titles currently. I really need to make it available :p

msimplay, 2021-11-30:
I think if you could do it then leave it as a user choice if they wish to take the performance hit or not.
Personally I prefer the fast menu as it currently is but sometimes I use the PSIO and I rather enjoy the menu since it supports thumbnails

badass-boss, 2021-12-07:
I noticed new update has a simple memory card manager. Will there be an option to copy memory card data to SD card and vice versa?? This would be huge. I also noticed xstation memory card manager does not encode Japanese characters which, in rare cases, is actually very important for people like me. Will this be implemented?? I can offer some help language-wise or programming-wise... if only I knew how to code Xstation

Betonar, 2023-04-02:
> **Designated Cheat Disc:**
> 
> If we could designate a cheat disc somehow (perhaps by naming it x00_Cheats.bin or similar) I think this could be useful for two reasons:
> 
> 1. A button in the launcher similar to boot/fastboot could be "boot with cheats"
> **2. When boot with cheats is used, the correct game title ID (and not the cheat discs Id) could be sent to PS1Digital/MemcardPro.**

This seems like really good idea and should not be hard to implement. Currently its frankly annoing to use MemCard Pro and PD1Digital combined with Cheat CD on Xstation modded PSX. PD1Digital game specific setting are unusable and automatic vitual memcard selection on MemCard Pro requires launching desired game in advance then removing MemCard Pro and then launching cheat CD and desired game and then insertion of MemCard Pro. Alternatively it requires messing with web user interface which is breaking easy to use seamless integration between Xstation/MemCard Pro/PS1Digital. It seems that only detection of multidisk games should be required as exception and otherwise it should be easier feature for implementation which doesnt even require changes in user interface.

Betonar/TomB
Very satisfier buyer of 4 xstations ;-)

jkuch77, 2023-09-21:
I wasn't sure where to post this, but I was wondering if in a future release you could make the the selected game name scroll for long games with long names.  Atm it seems to just cut off long game names like there is some sort of character limit for the names.  My everdrives scroll long game names and it looks much better.  Loving the xstation btw!  Thx!

ramapcsx2, 2023-09-24:
@jkuch77 probably, something like that. I'll write it down :)

Peredonov, 2023-09-27:
Since a new release is in the works, I wonder if this humble (and hopefully simple to implement) feature request can be squeezed in for the next release: option to swap O and X button functions on the Loader, for the sake of us Japan/NTSC-J system users. Thank you for considering this request, @ramapcsx2 :)

riggles1, 2023-09-27:
Still hoping for guncon menu navigation support, essentially proper 2 player support as you have to use a controller in slot 2 atm. PS1 hardware isn't really meant for hotswapping controllers (fueses tend to die on them) so being able to select a lightgun game using the guncon alone is essential. That would be huge news!

JuanWNeira, 2025-02-13:
Good evening I hope you are all doing well, I would like to make a request, allow me to explain the situation, before I used to use memory cards with the xstation and it works wonderfully, however, recently with the rise of the sd2psxtd projects, I purchased a psxmemcard gen 2 looking for better save options, together with the xstation it works wonderfully and loads the title ID without problems when used (Image A), however, when I use/apply cheats (engine compilation) on a specific game and load it, the xstation does not send or recognize the title id anymore to the sd2psxtd card and although the game starts without  a problem it asks to create a new save / memory card in order to save.

Image A [image]

After looking for some info i found thta this problem comes from before a like @Betonar mentioned with the MemCard Pro his idea of a Designated Cheat Disc or a way to recognize / separate the title id in the multidisk queue selection will allow to do the job and avoid that all sd2psxtd cards require a new vmc after applying cheats.

ramapcsx2, 2025-02-15:
Eventually, at some point. I know what has to be done for this, but I dislike the amount of hacks on top oh hacks needed ><

msimplay, 2025-02-15:
I would recommend this functionality be turn off as standard and on for those that wish to use it.
Due to the hacky nature I'm guessing it can cause issues

JuanWNeira, 2025-02-18:
I agree with what msimplay mentions, I had even seen that someone mentioned the option of being able to give a special name to the .bin of the hacks, for example, 00cheats.bin, when applying the queued cheats, disk 1: cheat (00cheats.bin) - disk 2 Game (Show id). In the end it is just an idea and in this case only the developer knows what is best for this case.