x-station/xstation-issues #24: ISOs not found

ryanm101, 2020-11-24:
I converted to ISO from MDF/MDS but when doing a refresh list the ISOs don't appear.

ramapcsx2, 2020-11-24:
The file scanner looks at files with a .img, .iso and .bin extension.
It looks for PSX data disk patterns in a location where they have to be.

So if the converted files don't show up, I imagine the conversion was not correct.

ryanm101, 2020-11-24:
Ahh, I'll try a different converter

ramapcsx2, 2020-11-24:
Yes, and please don't use .iso for this. Iso doesn't allow for any CD audio.
You'll want to convert to .bin/.cue.

birdybro, 2020-11-24:
I've used AnyBurn with success in the past.

http://www.anyburn.com/tutorials/convert-image-file-to-bin-cue.htm

When you finish that, check the MD5 with redump:

http://redump.org/discs/system/psx/

If it doesn't match, something might be off. They also have CUE sheets for every game in their database for download as one zip.

http://redump.org/cues/psx/

That way if Anyburn messed up the CUE in conversion, you can just download this. Be sure to rename the BIN to be the same as it's referenced in the CUE sheet.

ryanm101, 2020-11-26:
I'll close this as clearly it's a 'me' issue and not an application one.

closed 2020-11-26 by ryanm101