x-station/xstation-issues #25: Booting to CD player - PU-8

8-bitBishop, 2020-12-02:
I've installed 2 xstations in PU-8 Late revisions, and both boot directly to cd player with lids closed. 
You'll notice that one solder point on the QSB, hasn't been soldered. I caught that on my final inspection, before finishing the install. 

I used the suggested formatting tool. I'm using a high quality SDXC,  Samsung SD card. I've tried all of the loader.bin and update.bin files, from every update. No luck. 

Any suggestions? 

[image]
[image]
[image]
[image]
[image]

8-bitBishop, 2020-12-02:
Also, the Power LED remains steady. The access LED comes on and off.

ryanm101, 2020-12-02:
I put this together for testpoints to ensure everything was connected (I think there is one point missing)
Also check your lifted pins are all correct, its easy to be off by one or miss one

[image]

8-bitBishop, 2020-12-02:
> I had a strange issue with SD card that caused this issue, using 'diskpart' to clean and then the formatting tool solved it.

I'll try that, thanks. I did try a few different cards that I formatted with the tool. No luck. 

If I can't get these working in these PU-8 boards, I'll have to try a different revision. Just a bummer, since I ordered SD mounts from Laser Bear, specifically for this revision. 

Hope I can figure this out. Was a super easy install. I was so excited. After 3 hours of troubleshooting, I'm pretty discouraged.

Also, I made sure every point on the QSB, was connected, using continuity testing.

ryanm101, 2020-12-02:
You could also try shorting the lid closed switch, to ensure it's not faulty.

8-bitBishop, 2020-12-02:
> You could also try shorting the lid closed switch, to ensure it's not faulty.

I thought about bypassing the lid switch, but both are working just fine. I may do that though, for convenience.

ramapcsx2, 2020-12-02:
So you say it boots to the CD player, not the "BIOS main menu"?
If so, then the lid was closed and a CD of some sort was detected.
Is it showing you one (or more) tracks to play as music CD? If so, do the seconds advance if you play it?

ramapcsx2, 2020-12-11:
I'll reopen this if there are updates :)

Timsaysso790, 2020-12-11:
I'm having this same issue on the pu8boards, I would just get a black screen when I turn on the console. I noticed when I touched the xstation board it would light up. To me it seems like a strange grounding issue on the cd posts it rests on. I havent had this problem on the pu18s. If I position it properly on the posts it works fine, but when I move the console around I have to fiddle with it again. I'm considering just soldering the board to the post to make it more reliable. Curious if this may be your issue

8-bitBishop, 2020-12-12:
> I'll reopen this if there are updates :)

Sorry I haven't responded sooner. I got Covid and have been out of it. 

Nothing has changed. I have 2 xstations that have been properly and professionally installed, that don't work. 

Both PU-8 boards. One gives me a black screen, one boots to the cd player. 

No idea what to do, other than install these in different consoles and board revisions.

8-bitBishop, 2020-12-12:
> I'm having this same issue on the pu8boards, I would just get a black screen when I turn on the console. I noticed when I touched the xstation board it would light up. To me it seems like a strange [...]

I knew it would just be a matter of time, before someone made the same complaint, as I am confident I have installed these correctly. 

I don't think it has anything to do with the posts, but lemme know if you figure it out. Very disappointing situation, as I have 2 customers waiting to get their consoles back. Bummer.

Timsaysso790, 2020-12-12:
I think I did figure it out, I used a brass grinder bit to scratch up the cd drive posts to clean them up. Then I used the rubber bumpers from the old drive and cut and shaped them to be used as a stand. See photos now it seems to make good contact on ground. With the pu 18 install I could have the board out of the console and it would still work. But seems with the pu 8 install it requires solid ground contact. 
[image]
[image]

Timsaysso790, 2020-12-12:
[image]

ramapcsx2, 2020-12-12:
There is no hard requirement for an extra ground on either PU-8 or PU-18.
If you get a solid black screen, something is very wrong. The same is true for getting different results if you wiggle the boards.
I suspect a QSB install problem.

closed 2020-12-19 by ramapcsx2