x-station/xstation-issues #26: Document Testpoints

ryanm101, 2020-12-02:
I put this together for my 5502 (it's missing 1 point i think)

It would be useful to have as part of the install guide a list of continuity test points to ensure all solder connections are solid (esp those ones that are holes in the middle of the board. 

[image]

ramapcsx2, 2020-12-02:
Thank you VERY much!
I know the work that goes into creating such diagrams, this is very much appreciated!

I'll check and try to get this into the guides and the mini FAQ :)

matjolic, 2020-12-02:
I'm currently trying to diagnose my X-Station not booting up even though it gets voltage. This is really helpful! It is missing a test point for "7", which I think is the problematic testpoint in my case. On the QSB it is linked to ground, but not on the motherboard. The testpoint I found for "7" is 2 vias very close to testpoint "8" near the CXD2545Q IC, which in turn lead to pin 87 of IC CXD1815Q above.

I would be super helpful if someone could tell me which flex cable pin "7" is wired to on the QSB, as I said, on mine it is GND (same a 10/11) but I don't think it is supposed to be grounded.

ryanm101, 2020-12-02:
@ramapcsx2 Thanks, glad it helps, feel free to edit and use as you like, I was testing my own install and figured it may help others.

ryanm101, 2020-12-02:
@matjolic I *think* 7 comes up near 8 on the topside of the board, maybe goes to one of the pins on IC701, @ramapcsx2 might be able to confirm.

matjolic, 2020-12-02:
@ryanm101 Yes, that is what I found, it comes up near "8", there are 2 vias on the same trace near that point that lead up to pin 87 of IC CXD1815Q. What I am not understanding is why this point is tied to 10/11 (GND) on the QSB, with the QSB desoldered from the mainboard. Is it supposed to be like that? Is it supposed to be wired to one of the flex connector pins?
[image]

ramapcsx2, 2020-12-02:
7: On the QSB it is linked to ground << That is correct. Point 7 should tie the PSX signal down to GND. It is the signal on the (lifted) DSP pin 56.

matjolic, 2020-12-02:
@ramapcsx2 Thanks for the reply, good to know! I that case, Is the pin of the flexcable connector that I point to in yellow supposed to be connected to anything?

ramapcsx2, 2020-12-02:
I just gave away my last loose QSB, so I can't quite tell you now :p
But generally, we just care that the PSX point is grounded for this signal.

ramapcsx2, 2020-12-02:
Surprisingly, another test point image got created today. I've added it to the landing page as well :) 
https://github.com/x-station/xstation-issues

ryanm101, 2020-12-02:
Brill. You should add @matjolic's one as well as it shows the connector.
My image needs Pin 7 added to it as well.

matjolic, 2020-12-02:
Great work by everyone :)
I just noticed something though, pins 10/11 from @ryanm101 ’s version are not connected together in the new image. Both are still tied to ground, right?

ryanm101, 2020-12-03:
my image was done after i'd already installed it. So the test points where purely found from tracing (using pictures of the motherboard for hidden traces) and continuity testing. (so for example GND may be testable from a single point but i'll have listed where i've traced the runs to.

ramapcsx2, 2020-12-05:
10/11 are indeed grounded, right next to them actually. Once the QSB is installed, point 7 is also grounded.
@ryanm101 I'm not sure if it's worth adding to the picture, but if you do, could you also resize the image to be a little more bandwidth friendly? Cheers :)

closed 2020-12-13 by ramapcsx2