x-station/xstation-issues #27: PCB step file

littlebalup, 2020-12-03:
Hi,

First, congratulation for your amazing work.  Your ODE is excelent.

I'm actualy working to design a 3D printable cover that will mimic the lazer bloc for a better look. That's what it should looks like at the end:
[image]

In order to design that I reproduce your board, playing with my caliper, but it's not 100% accurate.
[image]

If you could provide a 3D model in STEP format it will help me a lot and save my time.
Of course I'll share my work here when finished.

Thanks

ramapcsx2, 2020-12-05:
Hey, this is a cool looking render!
I didn't do any of the PCB work, so I don't have such files.
The only tip I can think of is that you can prototype quickly by doing rough 3D prints. Hope this helps?

birdybro, 2020-12-07:
I'd only add a minor suggestion, some small openings for venting, as the original design of the PS1 with the laser assembly had little gaps here and there which vented heat upwards and out (since hot air rises), whereas the Laser Bear mount for instance does not allow as much heat to escape. Maybe make the xStation logo itself the vent?

littlebalup, 2020-12-07:
> I'd only add a minor suggestion, some small openings for venting, as the original design of the PS1 with the laser assembly had little gaps here and there which vented heat upwards and out (since [...]

Hum, why not. But I don't think it's hot enough to really justify an extra ventilation. Anyway, kept some small gaps.

littlebalup, 2020-12-07:
Job done : https://www.thingiverse.com/thing:4678180

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2020-12-09:
Good work!
I hope you can sell a couple of these, and maybe you want to get in contact with the other makers of custom shells?

littlebalup, 2020-12-09:
> 
> 
> Good work!
> I hope you can sell a couple of these, and maybe you want to get in contact with the other makers of custom shells?

Thank you.

Thingiverse site seems to have some troubles... once again. So I uploaded the files to myminifactory too. Link added to my previous post.

I sincerelly won't have the time to make and send extra units. It's shared under [by-nc-sa](https://creativecommons.org/licenses/by-nc-sa/4.0/) license. But if another maker is interested to make and distribute it we can discuss.

birdybro, 2020-12-09:
We have a 3d printing learning station being setup at our work, i think it will be with a prusa mini, so I'll fire up the slicer and see if it runs smooth in there. Very cool!

closed 2020-12-19 by ramapcsx2