x-station/xstation-issues #28: De-Dither patching GUI

JayBee-git, 2020-12-04:
I coded a simple GUI for easier De-Dithering of games.
The app will scan a folder or drive to find all cue/bin game files, and then you can select which ones you want to patch.
It also has various save/backup options.
Here is the link to it;
https://github.com/JayBee-git/PS1_De-Dither_GUI

Kito1984, 2020-12-06:
Love the idea, but for some reason it freezes for me after any image patching. 

[image]

Tried on two PCs with Windows 10 and Windows 7. Also it can't detect either 2Xtreme or 3Xtreme bins (No valid bin file found).

JayBee-git, 2020-12-13:
> Love the idea, but for some reason it freezes for me after any image patching.
> 
> [image]
> 
> Tried on two PCs with Windows 10 and Windows 7. Also it can't detect either 2Xtreme or 3Xtreme bins (No valid bin file found).

For the "no valid bin file found" it means your .cue file doesn't contain the right file names.

For the error you get, can you copy paste the full error message here please so I can take a look?

Kito1984, 2020-12-13:
> For the "no valid bin file found" it means your .cue file doesn't contain the right file names.
 
> For the error you get, can you copy paste the full error message here please so I can take a look?

Aha, so that's the problem. I thought you only need bin file, now I'll know better.

Yes, of course. I'll attach it below. Thank you for looking into it!

"Подробная информация об использовании оперативной 
(JIT) отладки вместо данного диалогового 
окна содержится в конце этого сообщения.

************** Текст исключения **************
System.IO.IOException: Папка не пуста.

   в System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   в System.IO.Directory.DeleteHelper(String fullPath, String userPath, Boolean recursive, Boolean throwOnTopLevelDirectoryNotFound, WIN32_FIND_DATA& data)
   в System.IO.Directory.Delete(String fullPath, String userPath, Boolean recursive, Boolean checkHost)
   в PS1_De_Dither_GUI.PS1DDGUI.patchBinFile(String fileName, String inPath, String outPath, Int32 mode, progressBar myBar)
   в PS1_De_Dither_GUI.PS1DDGUI.button1_Click(Object sender, EventArgs e)
   в System.Windows.Forms.Control.OnClick(EventArgs e)
   в System.Windows.Forms.Button.OnClick(EventArgs e)
   в System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   в System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   в System.Windows.Forms.Control.WndProc(Message& m)
   в System.Windows.Forms.ButtonBase.WndProc(Message& m)
   в System.Windows.Forms.Button.WndProc(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Загруженные сборки **************
mscorlib
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4300.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
PS1 De-Dither GUI
    Версия сборки: 1.0.0.0
    Версия Win32: 1.0.0.0
    CodeBase: file:///K:/Rom%20Sets/Sony%20PlayStation%20(NTSC-U)/Tools/PS1%20De-Dither%20GUI/PS1%20De-Dither%20GUI.exe
----------------------------------------
System.Windows.Forms
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4270.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4300.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4300.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
mscorlib.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
System.Windows.Forms.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Оперативная отладка (JIT) **************
Для подключения оперативной (JIT) отладки файл .config данного
приложения или компьютера (machine.config) должен иметь
значение jitDebugging, установленное в секции system.windows.forms.
Приложение также должно быть скомпилировано с включенной
отладкой.

Например:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

При включенной отладке JIT любое необрабатываемое исключение
пересылается отладчику JIT, зарегистрированному на данном компьютере,
вместо того чтобы обрабатываться данным диалоговым окном.

JayBee-git, 2020-12-13:
Thanks for the copy paste!
It seems to have trouble to overwrite or delete files.
You might want to run the app in with administrator right to be sure it can edit the files correctly.
I will check on my side if I can reproduce the issue.

JayBee-git, 2020-12-13:
Oh and for the .cue, yes my app requires it to only patch the first .bin file of each game 😉

JayBee-git, 2020-12-13:
There seems to be an issue here too, I will fix it and keep you updated.

Kito1984, 2020-12-13:
> There seems to be an issue here too, I will fix it and keep you updated.

Thanks, looking forward to it! Unfortunately, adninistrator rights didn't fix the problem for me.