x-station/xstation-issues #29: Possible addition to the installation instructions (Suggestion)

calaerial, 2020-12-05:
Apologies if this is the wrong place to be putting this but its just a minor suggestion of something that the devs may wish to add to their installation instructions (which are very good on the whole)

I had an issue it took me 2 days to track down an intermittent fault whereby the X-Station would not boot into its loader.

The installation had proceeded without a hitch and the instructions had been absolutely clear. No issues were identified and at first power on after installation it booted the loader without issue. 

Upon powering it down to add some games to the SD for testing it failed to boot (memory card/music screen)

Removing the config file from the SD card at that time seemed to resolve the issue. Was able to test a game (Resident Evil) and it worked perfectly.

The pattern continued whereby it would only boot the loader the first time the card was inserted, once a config file was generated it stopped working. At that time i tried updating the firmware. This appeared to correct the issue and it booted the loader every time for a period of around an hour. Multiple games tested, all performed perfectly.

It was then left for an hour and booted again - Failed to boot to loader
Removed config - No effect

I tried reseating the ribbon, ensuring the SD was prepared with the correct SD association tool, tried a second SD card (though both were official Sandisk Ultras) nothing worked. 

In desperation i tried reflowing the joints of the QSB and after this it worked - Loader booted.

It then worked for the rest of the evening but on the following morning - Loader failed

Reflowed again - Working

Next morning - Loader failed

I then tried putting the thing only partially back together, leaving the chassis incomplete - The loader worked and continued to work the next day. 

I then tried insulating the QSB against the RF shield with electrical tape and fully reassemble - This appears to have resolved the problem.

I'm not a hardware engineer (just a relatively experienced bodger) but i suspect some sort of ground leak may be the issue and it may be wise to consider adding instructions to insulate either the QSB or the RF shield itself to prevent shorts/ground leaks when fully reassembled. Alternatively providing a small self-adhesive vinyl panel to place over the unit would maybe be a nice little addition to the SKU. 

Have to say though, these small issues aside im really impressed with the unit itself, its performance is fantastic. I wish it had included some rubber grommits to stop the X-Station rattling on its mounting pins but i recognise that most users will use those from the laser (My unit didnt have a laser to begin with as it had gone faulty years ago and thats why this unit was selected for the conversion)

birdybro, 2020-12-05:
This is an interesting problem. Is your RF shield bent in some places perhaps? Or maybe some screws were overtightened? Mostly interested since was the root cause the RF shield bridging or shorting a circuit actually determined to be the root cause?

calaerial, 2020-12-05:
To be totally honest Birdybro im not 100% certain. 

In terms of bends to the RF shield - None are visible, however it was noted that the top shield (The thicker piece with the laser mounting prongs) didnt sit very cleanly once the unit was reassembled. It seemed to want to pivot up at the rear right of the unit for unknown reasons. This may be some damage that pre-dates my having the unit as it came in a batch of bad PS1s years ago.

Unclear if something is overtightened, they werent torqued to extreme but they were fully tightened down.

I think it would require someone with more electronics knowledge than me to fully diagnose the specific root cause, but whatever it is it seemed to be resolved by insulating the QSB, which could be useful for people to have in their back pocket during troubleshooting. Perhaps if not in the instructions then in the FAQs

ramapcsx2, 2020-12-06:
Hey, so the QSB has at its highest elevation the flat flex connector. This is made of plastic.
If this connector makes contact with the shield (which it should not, there should still be some space), then nothing will happen.
Any part of the QSB that could shorten out is lower than this connector, and the layout of the QSB is such that it will not touch any elevations on the shield.

I imagine what is going on is that the added insulation now presses the QSB onto the board, fixing a cold solder joint.

calaerial, 2020-12-06:
Thats a fair point. 

I cant imagine why my solder joints would be dry at this stage, they have been reflowed within an inch of their lives. But I'm not a soldering expert so there may be something im missing. My usual approach to a dry joint would be to redo it with more flux but since this is a flat QSB going directly onto the mainboard im not sure how i would approach that. Not without removing the whole thing and im more likely to damage it doing so. And to my knowledge i cant purchase a new QSB separately?

Perhaps some of the more liquid type of flux would help, something that can seep under. At the moment the flux ive been using is a very dense waxy sort of consistency. 

Either way, thanks for getting back to me man :)

closed 2020-12-13 by ramapcsx2

kranibal, 2020-12-25:
I have THE EXACT same problem. Almost word for word. I am getting tired of refluxing every time.

ramapcsx2, 2020-12-27:
Hey, chances are that something is creating pressure then, that may break the contact after a while (or after it heated up a little).
To debug this, you could use the test point images and verify continuity while softly applying some pressure on the QSB.

kranibal, 2020-12-28:
Hi,

I believe I might have found my problem...

I refluxed all soldering points and this time I applied pressure on the board where I was refluxing.

This seems to have resolved my intermittent problems.

Thanks for getting back to me.

Sincerely,

Martin