x-station/xstation-issues #30: Booting to PS1 Menu / CD Player PU-18

jennoa, 2020-12-05:
I'm trying to install the xstation to a PU-18 psx and when I turn it the console on it goes through it's normal startup and once it gets to the PS1 logo it then goes to the playstation menu like if the cd lid is open. I've confirmed that the lid switch works too.  The xstation power led turns on blue and stays solid through out. The green access light will flash like it is reading memory at the boot screen and then stop once it goes to the psx menu.

I don't have a great camera, but these are some shots of my install. Please let me know if you need better images, I'll try again.

[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2020-12-06:
Hey, the solder joints look very dry. I recommend reflowing every connection and using flux.
You should also shorten the exposed parts of the APLL wire, so it can't short with anything.

See the Mini FAQ for good continuity check points: 
https://github.com/x-station/xstation-issues/blob/master/README.md

jennoa, 2020-12-06:
Ok thank you. I’ll keep working on it.

closed 2020-12-06 by jennoa