x-station/xstation-issues #31: Super strange issue - Wrong disc is loaded

uberyoji, 2020-12-06:
For some strange reason, some disc are not loading. Instead some other game is loaded instead. It is always the same though.

For example, I try to launch "Castlevania - Symphony of the Night (USA) [Improvement v1.2]" but instead "Tobal 2 (Japan) [T-En v1.00b]" is launched.

I am trying to find some pattern but can't seem to find any for far. Maybe it is related to file length. Or maybe to much folder nesting.

I attached my file structure to the post.
[tree.txt](https://github.com/x-station/xstation-issues/files/5648226/tree.txt)

uberyoji, 2020-12-06:
I made sure no more than two folders deep are used and it seemed to fix the issue.  (ex: root:\folder A\folder B\file.bin)
I will try some more games to see if it is really the case.

ramapcsx2, 2020-12-06:
You may have missed refreshing the games list when you changed some files on the card.
An automatism to remind the user of this is in the works :)

uberyoji, 2020-12-06:
Seems fixed now. I believe you can close the issue :D

closed 2020-12-06 by ramapcsx2