x-station/xstation-issues #33: Cannot Launch PAL games(?)

aaaaafuck, 2020-12-09:
Hello! I recently installed this and I noticed that when I try and launch a PAL game my video signal cuts out and I have to restart the console. I don't know if this is an issue with my install or something that isn't supported and I didn't read up on it (if so then oops). Worth noting is when using the 240PTestSuite when I change from NTSC it does the same exact thing. Japanese NTSCJ games launch fine it is only PAL games. My motherboard is a Late PU-8 revision if it matters. I tried multiple TVs and multiple games all resulting in the same issue. I am able to launch the same games on my PS2 at the same resolution and they work fine on all the TVs I tried. I have no other issues with the mod everything else works with intended functionality. Sorry if this is me being an idiot happens quite a bit more than I'd like to admit.

ryanm101, 2020-12-09:
I'd check all the test points per the TS guide 1st just to be sure all connections are good. 
You might need a DFO, it is *possible* that the freq from your PS1 is too far out for the tvs to handle assuming you PS1 is not PAL. 

I can confirm mine works fine with PAL games (5502, PAL, PU-18) and NTSC(J) games though I did have an issue with an LCD being unable to play any video (CRT worked fine), suspect ps1 couldn't drive the LCD).

ramapcsx2, 2020-12-09:
Ah, I think you may have one of the consoles where the mainboard grounds the PAL GPU clock input.
Whenever such a console switches to PAL mode, the GPU just stops.

http://www.psxdev.net/forum/viewtopic.php?t=628#p5587

ryanm101, 2020-12-09:
So.... Remove mod, and sell as ultra rare console £10000,😂😂

aaaaafuck, 2020-12-09:
Thanks for your help Rama that 100% is my problem after looking at the link you posted. Your product is really awesome I've been enjoying it even without being able to play my favorite game Vib-Ribbon. :( I'll be looking to get a hold of a 550x model and I'll just put the extra QSB in that one and use that as my main system. Thanks for your hard work on this. You should maybe document somewhere that the late PU-8 boards can't run out of region games without extra modification just so nobody gets frustrated after buying a Late PU-8 thinking it'll work.

ramapcsx2, 2020-12-10:
Oh no no, this isn't meant to tell you that you can't run PAL titles.
You simply have to remove a resistor and place it somewhere else, iirc :p

aaaaafuck, 2020-12-10:
All working now! Thanks for being patient with me. A lot was talked about in the thread and I kinda got lost in the mess of it, but after looking again I figured it out. Now just to get a SCART cable so I'm not seeing in B&W.

closed 2020-12-10 by aaaaafuck

Barawer, 2021-01-01:
> 
> 
> All working now! Thanks for being patient with me. A lot was talked about in the thread and I kinda got lost in the mess of it, but after looking again I figured it out. Now just to get a SCART cable so I'm not seeing in B&W.
Sorry to bring an issue from the dead, but can you point me in what to do? i'm having trouble understanding

aaaaafuck, 2021-01-01:
> Sorry to bring an issue from the dead, but can you point me in what to do? i'm having trouble understanding
https://i.imgur.com/w52uJMt.png
The resistor that is being pointed at by the yellow arrow in this picture needs to be moved to the points that are being pointed at by the red arrow. The resistor is running the PAL video signal to ground. If you move it to the other points it no longer turn off the GPU whenever you launch a PAL game. The only problem is that you will need to use a RGB cable to get PAL games to display in color for reasons I don't understand well enough to give you the proper reasoning, so I'm not going to attempt explaining it.

https://i.imgur.com/3PcLtEa.jpg
Just in case you are unable to find the part of the board from the first picture I made this for you. Important to note tough is this was made using an early PU-8 board because I couldn't find a picture of a LATE PU-8 board's bottom side. Good luck, I can try to help as much as I can, but I am far from an expert on this stuff.

Barawer, 2021-01-01:
> 
> 
> > Sorry to bring an issue from the dead, but can you point me in what to do? i'm having trouble understanding
> 
> https://i.imgur.com/w52uJMt.png
> The resistor that is being pointed at by the yellow arrow in this picture needs to be moved to the points that are being pointed at by the red arrow. The resistor is running the PAL video signal to ground. If you move it to the other points it no longer turns off the GPU whenever you launch a PAL game. The only problem is that you will need to use a RGB cable to get PAL games to display in color. This is for reasons I don't understand well enough to give you the proper reasoning, so I'm not going to attempt explaining it.
> 
> https://i.imgur.com/3PcLtEa.jpg
> Just in case you are unable to find the part of the board from the first picture I made this for you. Important to note though is this was made using an early PU-8 board because I couldn't find a picture of a LATE PU-8 board's bottom side. Good luck! I can try to help as much as I can, but I am far from an expert on this stuff.

Thank you! It worked. I was using RGB before so I'm OK with that :) 
@ramapcsx2 Just a suggestion, but since I'm guessing this might come up again, maybe add something about it in the FAQ?