x-station/xstation-issues #34: Pin mapping for PU-18 QSB board

naggie, 2020-12-11:
Hi,

Unfortunately I received my QSB with part of the PCB snapped off. Rather than RMA and wait for shipment to the UK I'd like to make the connections manually from the FPC connector to the pads with enamelled wire. It seems there's a handy pinout for the PU-8 QSB in the readme, but not for the PU-8.

Here is an image that shows the missing leg of the PCB and the pads marked A and B.

[image]

I'd appreciate it if someone could tell me what FPC pin the 2 missing connections go to. I think one is ground.

Thanks for making a cool piece of hardware!

ramapcsx2, 2020-12-11:
Ow ><
You need to link your A and B points so both are grounded.
Then route a stronger wire from the A ground point to the flat flex connector mounting metal, so the QSB is getting ground.
Check and confirm with the images on the bottom of the page here:
https://github.com/x-station/xstation-issues

ramapcsx2, 2020-12-11:
As a follow up, the place where the QSB broke carries a signal that you soldered. This signal should not be connecting to ground, once you've made your repairs. If it does, the solder is bridging to the QSB ground plane and you need to clear that connection.

naggie, 2020-12-11:
Thanks @ramapcsx2 I appreciate the quick response -- I got it working! I had checked the broken part for the exposed copper and checked it wasn't soldered or touching.

[image]

By the way, I had an idea; I drilled some holes using the main PCB as a template, held in place with the other stand-offs and a piece of foam on a drill press.

[image]

[image]

I used M3 bolts and a 3mm drill bit. The result is the PCB is much more sturdily attached 👍

closed 2020-12-11 by naggie