x-station/xstation-issues #35: Pu-8 Grounding Issue

Timsaysso790, 2020-12-11:
This is my 3rd install, but my first on a pu 8. I noticed after going crazy for about 2 hours and constantly getting black screens and intermittent status light on the xstation board. Resoldered the qsb, checked all the test point and they all were good. Finally I started fiddling with the xstation as it sat on the cd post. Bam! It worked! So my point is that does this version of the install require the post for it to be grounded? If so I'm just considering soldering the xstation pcb to the posts, as moving it round my house it become pretty unreliable. And help or info would be appreciated

ramapcsx2, 2020-12-12:
No, it doesn't require extra grounding. You have some sort of install issue.
Did you isolate the CPU oscillator (remove the resistor)?

Timsaysso790, 2020-12-12:
Of course, I have it working, its just what I'm saying is does the pcb require and or is that the only grounding source from the mounting holes that the post insert into? Of thats the case, moving and or shipping the console I forsee issues if the contact isn't the greatest. I noticed on the pu 18 installs I've done, I could have the board out of the console and it would work fine. Seems that the pu 8 install requires that for it to work properly

ramapcsx2, 2020-12-12:
The PU-8 also does not require any extra grounding. You still have an install issue ;p

ramapcsx2, 2020-12-12:
Things you can try:
- permanently attach a multimeter probe to an xStation ground point, the other to the PU-8 board ground. Wiggle cables and the QSB and see whether contact breaks
- check the flat flex for damage
- touch parts on the xStation and see whether there are any cold solder joints

ramapcsx2, 2020-12-28:
Is this still a problem?

closed 2020-12-29 by ramapcsx2

NuVanDibe, 2021-05-04:
I have this same issue on my late PU-8. The thing only works when the brass cd rom leg makes good contact with the grounding plane of the xstation.