x-station/xstation-issues #36: NTSC games on PAL console.

lukesk, 2020-12-12:
Hi, i have a problem with playing NTSC games on my PAL console. When i run NTSC (USA or Japan) game i have "rainbow" colours.
I have SCPH-5552 PAL (PU-18) console. Problem is on original Sony composite cable and on original Sony S-Video cable. Has i done something wrong in process on installation of XStation?. PAL games works great and games with PAL/NTSC selector.
[image]

ramapcsx2, 2020-12-12:
That's an interesting look, not bad! :)
The problem is the out of spec GPU frequency for running Composite or S-Video.
You have several options to get around it, the easiest probably being an RGB cable.

lukesk, 2020-12-12:
Thank You for fast reply and sorry for my bad english.
I dont had this issue with my two other playstations 7502 and 9002 (moded with chip) using same cables.
Can You tell me what are other possibilities? I'm asking because i prefer to use my Retrotink Pro instead of RGB(scart) cable.

ramapcsx2, 2020-12-12:
Ah, very interesting that the Retrotink can (almost) produce color :) 

The later PSX revisions (SCPH-7000+) produce a PAL 60 format if you play an NTSC game on them, but the earlier ones produce pretty much nonsense.

Your other solution is to make your PSX multi format by installing a DFO:
https://videogameperfection.com/products/dual-frequency-oscillator/

lukesk, 2020-12-13:
I will try to get one. They are out of stock for now. I must wait for restock.
Thanks for help and your time.
Best regards
Lucas

ramapcsx2, 2020-12-13:
Good luck :)

closed 2020-12-13 by ramapcsx2