x-station/xstation-issues #37: Gex - Enter the Gecko FMV stuttering

mario-64, 2020-12-13:
Tested on two different freshly-formatted brand new SanDisk SD cards, an Ultra Plus and an Extreme Plus.

The intro FMV for this game stutters and hitches at random places. Compared against the actual game disc and a PSIO (yes, I know). Neither exhibit this problem. It is specific to Xstation. Using the Redump image.

Happy to help troubleshoot in any way I can.

ramapcsx2, 2020-12-14:
Thanks for the report, I'll check it out!

ramapcsx2, 2020-12-14:
Hm, is your console NTSC?
So far I didn't spot an issue.
Are your dropouts skipping audio?

mario-64, 2020-12-14:
1) Yes. USA 5501 console NTSC
2) No. Only the video stutters. Audio seems unaffected. It is 100% reproducible. Does it every time the game is played but the hitches are random. I'm using Laserbear's SD card mount if it matters.

ramapcsx2, 2020-12-14:
Okay, I can kind of see what you may mean. Could you do me a favour and record a short sample video?
It'd be great to have it in 60fps, but whatever works for you :)

mario-64, 2020-12-14:
Sure. Give me just a few minutes.

mario-64, 2020-12-14:
These files are around 30MB each and Git won't let me attach them. Any other way I can get them to you?

ramapcsx2, 2020-12-14:
Okay, this is similar to what I see.
I've started looking into it, but so far all seems normal.

ramapcsx2, 2020-12-15:
Hm, in your video it is pretty noticeable. 
I barely get a hitch every couple of runs, so that makes it hard to test :/

I'll try a few things to affect it.

mario-64, 2020-12-15:
Thanks for investigating. Not sure what is causing it.

ramapcsx2, 2020-12-17:
Update:
I found what makes this FMV special, and I think I can fix it :)

mario-64, 2020-12-17:
Awesome! Thank you so much! Thrilled to hear that :)

ramapcsx2, 2021-01-14:
As an update:
A full fix for this game's FMV requires microsecond precision from the SD card sector data fetch to the I2S (data) send completion.
I improved it somewhat on the upcoming firmware, but it's not fully there yet.
Specifically, it depends a lot on the SD card now.
On average though, it runs smoother.

mario-64, 2021-01-14:
Thanks for the effort and update. Is this fully fixable or will it always be an issue with xstation?

ramapcsx2, 2021-01-14:
It's not so much the speed of the card, but how evenly the sectors can be fetched. This isn't something users can influence much.

I think that I may be able to get it better than it's now, by using a fully different sync method.
But maybe it's already close enough with the next firmware. We'll see then.

mario-64, 2021-01-14:
Ok cool. Can't wait to try out the update :-)

Peredonov, 2021-01-15:
Was very glad to read an update on this issue. Thank you rama for your hard work on it. As to it being close enough, my opinion would be to compare it to original hardware in good shape, perhaps a PS1 with a new old stock OEM laser assembly. 

If it's not 100% the same but this is the only game affected by SD card sector even fetching, the next FW may be all that's needed, but it if's a more global factor that may impact cutscenes in other titles then exploring that new sync method would be worth it. Based on what you learned about it, do you suspect that it's only due to idiosyncratic coding by Crystal Dynamics?

ramapcsx2, 2021-01-16:
The issue is down to that game engine, and I don't think anyone else does something like it.
It appears (to me at least) that they're syncing each drawn frame to the timecode on the CD. 
It's some horrible overhead on the CD control commands alone, and it's pretty useless for an FMV.

ramapcsx2, 2021-02-13:
Okay, here is an update:
The game can be made to run better on xStation, it is very hard and doesn't seem quite correct however.
A tester has checked the FMV on his real drive, and the stutter is there as well.
He also tested with an older firmware PSIO, where the same problem was observed.

I'm waiting for a new drive, so I can check it here as well.
For now though, it looks like the game engine is just as bad as it appears to be :)

mario-64, 2021-02-13:
Thanks for the update. FWIW, I don't get any stutters on my PSIO as observed in the attached video. However, PSIO firmware updates have repeatedly broken then fixed this game. Several times. Clearly it is unique in some way. I do have a PSX I can test the game disc with. I'll try that then report back.

ramapcsx2, 2021-02-21:
I've now tested Gex 3 extensively (same engine and issues). The game stutters extremely if the media is CD-R or the drive isn't perfect.
In my best drive and best burn, I got it down to a few juddery moments with just a hint of audio dropout occasionally.
The next xStation firmware improves the ODE performance to better than this drive, but I'm afraid it can never be perfect.
Here is a small rundown on what I think is going on:

- game wants to know the current sector location constantly (getLoc() spam)
- DSP streams FMV sectors to the 1815Q chip
- DSP notifies mechacon of new subcode ready for each CD sector
- mechacon picks up the SUBQ whenever it gets to it
- game sometimes misses the getLoc() response in time it wants it, because mechacon is currently fetching SUBQ
- (this is the Gex special part) game blocks video updates until a new frame comes in

It's very weird to code an FMV routine like that, and it's sure to cause issues on even the best drives.
I assume they just didn't quite notice it when testing.

mario-64, 2021-02-21:
Thank you so much for the extensive deep dive into this issue. Looking forward to the upcoming firmware :)

closed 2021-02-21 by ramapcsx2