x-station/xstation-issues #38: PU-18 boots straight to the bios screen

Test045, 2020-12-16:
Just received and installed my Xstation myself on a PU-18, japanese console. This console was tested before doing the mod.
When I power on the console, the led on the xstation board lights up blue.
Unfortunately, it boots straight into the BIOS after the sony logo (with white background), where I can use the memory card manager or play a CD (which plays nothing).
Here are screens of the install. Continuity has been check for the IC on which legs were lifted, to make sure there are no shorts or that any leg came out of the board while lifting the nearby ones.
My solder points look good, I have no idea what is wrong. I installed the concurrent device multiple times and it always worked on first try, so I'm either being dumb here or the QCB or xstation board itself is faulty.

Also maybe worth noting, the console just stays on a black screen (no BIOS or anything) without the xstation board plugged to the QCB with the ribbon cable.

https://i.ibb.co/7GsQM8D/20201216-215444.jpg
https://i.ibb.co/mhqfXSz/20201216-215439.jpg
https://i.ibb.co/mSGcYTk/20201216-215559.jpg
https://i.ibb.co/pLFDx1m/20201216-215449.jpg
https://i.ibb.co/VJcrLMp/20201216-215435.jpg
https://i.ibb.co/JkYCrkC/20201216-214600.jpg
https://i.ibb.co/QkYn0xV/20201216-215531.jpg
https://i.ibb.co/DDswGNm/20201216-215501.jpg

Test045, 2020-12-16:
nevermind, please close the issue...
I was testing the hardware without putting the top shell back, so the button which tells that the CD cover is "closed" wasn't being pressed obviously...
works wonder thank you for the product !

birdybro, 2020-12-17:
Add it to the guide like this?
[image]

ramapcsx2, 2020-12-17:
I'll try to improve this somehow :)

closed 2020-12-17 by ramapcsx2