x-station/xstation-issues #39: Boot to Bios PU-18 (yes the lid button is down and working)

brilliancenp, 2020-12-17:
I just read the previous closed ticket that was related to this.  I was hoping it would be my problem but it is not.  I AM pressing the disk close button on the board down.  I thought maybe that the button was not working but I did a continuity test on the button open and closed and it seems to be working.

I have continuity tested all points on the board and everything seems to be correct.  I have set my SD card up according to the guide and even saw a ticket in here suggesting to use diskpart to clear it and then used the official formatter to format the SD card. My Xstation main board is getting a solid blue light, the access light will flicker a couple of times when you turn on the console.  I also noticed that a file called 'config.txt' gets written to the 00xstation folder.  This file is 1KB in size and just contains 'SOH' as the text inside.  Also the board knows if there is no SD card inserted as it will flash the access light slowly if one is not inserted.

I also am running PS1Digital and it works properly.  I just cannot figure out what might be wrong.  I apologize for coming here but I have scoured your tickets, the internet and your faq and I cannot find anything that might help.

chrisrock1984, 2020-12-17:
I had the same problem. I also checked all the points and they were good.

Solution was to try it on a second console and then it was working.
Don't ask me mhy ;-/

ramapcsx2, 2020-12-17:
It sounds like you thoroughly checked your work, so I assume you have a hidden QSB install problem.
One example I just had: One install checked out on each QSB to PSX mainboard test point, but the drive limit switch signal did not reach the CD controller chip anymore. The copper in the via must have evaporated from too much heat.

https://i.imgur.com/fLLLpr6.png
The leftmost red trace signal should reach a pin on the square IC in the upper left (where the + sign is).

dubesinhower, 2020-12-17:
I have this same issue. PU 18. Verified all QSB test points, and the FFC cable has continuity on both ends. I get the access light lighting up, but then it turns off and dumps to the Memory Card screen

brilliancenp, 2020-12-17:
Thanks for the responses guys.  After work today I am going to try some more testing and may just undo the QSB board and start over.  Ill let you know if I find anything and let me know if you think of anything else that might help.  Also @ramapcsx2, are you saying that even if the continuity test for each point check out to be ok, there can still be problems such as the one in your picture?  What was the solution for the instance there with the CD controller chip?

ramapcsx2, 2020-12-17:
Yes, there can still be issues, even if the test points connect. This test just verifies that the QSB solder point makes contact with the PSX solder point.
The case I just had was that a PSX mainboard via evaporated.
So the QSB connected, you could still verify it, yet the signal on that line could not reach the destination.

The config file doesn't say anything about the operation. What you get is normal.

Assume the xStation PCB is fine and would work, but there is a problem with the QSB / PSX / switches etc.
The xStations are tested in production and are then very unlikely to fail, and I didn't hear of a bad xStation yet.

brilliancenp, 2020-12-17:
@ramapcsx2 thank you for your response.  I will keep looking and report back, hopefully with a resolution and success.

dubesinhower, 2020-12-17:
Keep us updated, I don't have time to diagnose problems on my current console so I ordered another

brilliancenp, 2020-12-17:
@dubesinhower I actually just figured it out.  Issue number 13 that had been closed had one post with a picture, posted by @ramapcsx2 that questioned if some of the pads where he had pulled pins may be crossed.  After seeing this I put some liquid flux on the pulled pins and the pads and just slightly touched them with my iron to ensure there was no bridges on the pulled pins or the pads from which they came and this seemed to resolve the problem.  

Sorry I missed that closed ticket.  This ticket can be closed.

closed 2020-12-17 by brilliancenp

ramapcsx2, 2020-12-18:
Glad to hear :)