x-station/xstation-issues #40: Spyro 2 PAL version in french has sound cut in intro on japanese console

Test045, 2020-12-17:
Steps to reproduce : 
1) Launch a new game selecting "Français" (french) on the PAL version of spyro 2
2) Watch the intro until the Avalar inhabitants show up (roughly 2 minutes later)
3) Pay attention to Enora and Hunter dialog
4) As soon as Hunter starts to describe how Dragons look like, the sound cuts completely for the rest of the cutscene, and resumes just after that when the gameplay actually starts.

ramapcsx2, 2020-12-17:
This is a general incompatibility between the PAL game on an NTSC machine (and vice-versa).
A few games have this problem, and it usually means problems with sound.

You could fix this in hardware by installing a DFO mod, or in software, by using the game region that fits your console.

ramapcsx2, 2020-12-19:
Just to clarify some more:
The problem would happen from a CD or a PC emulator as well. 
It is a general problem that is caused by the wrong GPU oscillator for the region of the game.
If the game times XA audio scenes on the Vsync rate, this issue will show up.

Tossler, 2020-12-21:
Does it matter which 3.3V DFO I get? Could you suggest any place I could get one?

ramapcsx2, 2020-12-21:
Hm, these are made for exactly this reason, so everywhere you can buy them, they should advertise PSX compatibility. But generally, you are looking for a 3.3V model, correct.

Test045, 2020-12-25:
I ended up buying a PU-18 PAL ps1 (all my PAL consoles were old PU-8s revisions or PU-7s and there is no QSB for that as far as I know...).
Just finished my xstation install on it, and indeed, the Spyro 2 (PAL version) cutscene issue initially described in this issue doesn't appear. Everything work as intended

ryanm101, 2020-12-25:
@ramapcsx2 is there a guide for the early PU-8s for which connections go where (in case i want to try to make the connections myself)? I'm keen to get one I have here with a Xstation on it.

ramapcsx2, 2020-12-27:
[image]
This goes with:
[image]

You should be able to trace most signals with these.

ryanm101, 2020-12-30:
@ramapcsx2 Just saw this thanks, Ill maybe give this a whirl at the weekend, any particular thickness of wire you recommend?

ramapcsx2, 2020-12-30:
I like working with enameled wire for this kind of multiple signals job.
It's not pretty, but it works :)
https://i.imgur.com/aSOwpxR.jpg

closed 2021-02-21 by ramapcsx2