x-station/xstation-issues #41: PU-18

leachicm, 2020-12-19:
I could use some feedback my issue. The xstation has a solid power light and the access light has a steady blink. The bootup times out and goes to the memory screen. I checked all the solder connections, tested the probe points, got continuity. I had a clean pin lift on the chip. I am pressing the lid button as well.

ramapcsx2, 2020-12-19:
I can recommend you read the other install issues, particularly these ones might apply:
https://github.com/x-station/xstation-issues/issues/39#issuecomment-747551682
https://github.com/x-station/xstation-issues/issues/39#issuecomment-747722155

If the xStation could create files on the SD card, it is almost certainly a QSB issue.
If you know the PSX worked before the mod (it could read a disc), then it is 99% surely a QSB issue.

leachicm, 2020-12-19:
Thanks, I missed the closed section. no additional files on the SD card, the unit was tested before installing, the drive was on its last leg but it could read discs. This is my second install attempt with a different PCB. I had the exact same issue with the first PCB but that unit had disc reading issues even with a replacement drive.

ramapcsx2, 2020-12-19:
Check whether you have this particular issue:
https://github.com/x-station/xstation-issues/issues/39#issuecomment-747400309

leachicm, 2020-12-29:
Just attempted an install on a late PU8 with a fresh QSB. I'm still getting the same behavior. The soldiering the pads on this board was easier than the PU18. I've also tried reformatting the sd card and grab a second one to test. I also tested the ribbon, everything checks out.

ramapcsx2, 2020-12-29:
Is the blinking regular, and the same if you remove the SD card?
In that case, you can narrow down your debugging to the SD card and the xStation cart slot.

leachicm, 2020-12-29:
Yes, I get a steady blink of about 2 flashes per second regardless of the SD card being inserted.

ramapcsx2, 2020-12-30:
We just had a case of bad SD card slot here: https://github.com/x-station/xstation-issues/issues/44#issuecomment-752281446

If the card is fine, and formatted with FAT32 or exFAT, then chances are your slot is also bad.
If you have an SD card extender/adapter to try, give it a go.
I would recommend you RMA the xStation, if none of the SD cards you have around will work.

ramapcsx2, 2020-12-31:
Repeating this here as well, so you get notified:
Please try repeated insertions / removals of a preferably not very valuable SD card.
This will exercise all the slot mechanics, scrub off any oxide from the pins, etc.
It might just work after that.

ryanm101, 2021-01-02:
This was just posted to a FB group I am in where someone installed MODE on a PS1. 
The QSB looks (as one would expect) very similar to the one for xstation.

```
It's actually not that bad of a install. The DSP chip pins are quite easy to lift (i used a x-acto knife)
However, the QSB can be tricky imo, due to the two small vias that need their solder mask removed, since i had to reflow it twice, as it kept going to the MC bios screen, even though they appeared fine with continuity.
It's caught a few people out.
```

leachicm, 2021-01-06:
Thanks. Don't know if I'll be trying an install on a PU18 again, the QSB is damaged from multiple reflows and removals. So I'm stuck with the PU8. I'm attempting to swap out the ribbon and the mainboard but it ill be awhile before they arrive.

leachicm, 2021-01-07:
Attempted another reflow with PU8. I have a new behavior finally but now the PS1 is unstable. The system is having a hard time starting when the access light is solid. When it does boot it goes into a boot loop with long periods of black screen or the system is incredibly slow. Don't know if its a QSB issue or a system issue.

ramapcsx2, 2021-01-07:
Have you removed the resistor for the original CPU oscillator?

But more importantly, was that regular blinking LED problem resolved?
Without the SD card, there is no point in installing the xStation in other systems.

birdybro, 2021-01-07:
> 
> 
> This was just posted to a FB group I am in where someone installed MODE on a PS1.
> The QSB looks (as one would expect) very similar to the one for xstation.
> 
> ```
> It's actually not that bad of a install. The DSP chip pins are quite easy to lift (i used a x-acto knife)
> However, the QSB can be tricky imo, due to the two small vias that need their solder mask removed, since i had to reflow it twice, as it kept going to the MC bios screen, even though they appeared fine with continuity.
> It's caught a few people out.
> ```
> 
> I was having issues with a reinstall and this fixed it for me, and all mine where passing the continuity tests

for the PU-18 install that I did successfully, I used voultar's guide, and he clearly indicated to scratch a few spots. I believe you might be onto something regarding an important note in the instructions to be updated if this is not present in both.

carlblackwell, 2021-01-23:
The suggestion made by @bahamutfan64 on the post referenced below helped me to resolve a similar issue. The SD card extender seems to cause issues with the retention clip that did not allow regular microSD cards to work for me after initially inserting the extender into the microSD card slot. https://github.com/x-station/xstation-issues/issues/54

closed 2021-02-01 by ramapcsx2