x-station/xstation-issues #42: Issue with PU-18 Mobo

DrDorito, 2020-12-20:
My playstation continues to boot to memcard/player menu. There is also a PS1 digital installed as well that works perfectly fine. The main board has a steady power light and steady flashing access light.

I have made sure all the joints are super clean and the pulled pins are not making contact with anything
I have tested all the points for continuity, both to the qsb and through the qsb (to the ribbon connector on the main board).
I cleaned and fully formatted the SD and tried another one just to be sure.

It gets hung up a little longer on the startup screen when the lid is closed (or button is pressed) so I'm pretty sure it's registering that the lid closed. It will go to the memcard/player screen noticeably faster with the lid open.

But the most frustrating thing is it worked once. I got it to boot into the Xstation 1 time when I initially put it together, then I shut it down and closed the console up and it has not worked since.

I believe I've looked through all the troubleshooting documentation and the 'closed' tickets but nothing seemed to help. Is there anything else I can try to get it working again? Is it possible something onboard became corrupt in the process?- because everything seems physically fine.

ramapcsx2, 2020-12-20:
As with the other reports like yours, I think that a QSB solder point became loose, or another such defect developed.
If you have any expansion rom cart (xplorer, gameshark, etc), you could install nocash's rom on it. When booting this, it will tell you the disc state on powering up, making it much easier to debug.

I've uploaded the installer disc here:
https://www.sendspace.com/file/dpivxj

DrDorito, 2020-12-22:
I went through and tested all the test points again. It seems like there is a connection between ground and the blue point (I drew an arrow on it). Is this supposed to be the case? I'm going to desolder the QSB and see if the connection is shorted within the QSB, but I want to be sure before I do so.
[image]

blzla, 2020-12-22:
That's one of the three points on the QSB that's supposed to be grounded.
mmmonkey's PU-18 diagram: https://mmmonkey.co.uk/wp-content/uploads/2020/12/xStation-550X-QSB-map.png

DrDorito, 2020-12-22:
Thank you, after testing everything again. It is definitely not the connections. I will look into getting a gameshark so I can see the disc state while it is powering on.

DrDorito, 2020-12-23:
*sigh*
After looking into expansion ROM carts. The only thing I could find were carts with Parallel ports, which I guess is to be expected since it plugs into the PS1 parallel port, but I don't have the ability to connect to these.

@ramapcsx2, is there a ROM cart solution that doesn't use the parallel port of a computer. i.e. SD, USB. I know it's a longshot but my next step is to buy another PU-18 PS1 and try that.

ryanm101, 2020-12-23:
Why not get a usb to parallel cable they aren't dear

DrDorito, 2020-12-23:
The only reason I am hesitant is because the ROM carts + Cable will end up costing about as much as a new PS1 5501, which is what I might discover is necessary once I purchase those.

ramapcsx2, 2020-12-23:
What? No, you don't need to connect to those parallel breakouts.
The cheat cart simply plugs into the expansion port and provides a new boot rom.
Usually, these are cheating softwares. You can flash anything you want to them though, and one such item is the nocash kernel replacement rom.
Note that you need another PSX and a way for it to play backups, so you can run the installer CD-R.

Getting another 5501 sounds a bit like a waste, but if you think that has better chances of success, then go with it :)

DrDorito, 2020-12-23:
Yea, I think that might be the best solution, But I won't purchase it until I verify the QSB completely (the only thing I haven't checked is underneath the board). The only other thing that comes to mind is if the lid button is messed up. I know it's effecting the load times but maybe it's not reading properly? (grasping at straws here). I'm going to see if I can locate alternate test points for the lid closure button to see if that's part of it.

DrDorito, 2020-12-23:
This one is pretty cheap.
https://www.ebay.ca/itm/Gameshark-Video-Game-Enhancer-v2-2-For-Playstation-PS1-Cheat-Codes/293911322491?hash=item446e7adf7b:g:~7gAAOSw~DBf1YJP

Would this work? and would I be able to install NOCASH's ROM via a parallel to USB cable? I have one other PS but it isn't modded for backups and I don't actually have a computer with a disk drive anymore haha.

ramapcsx2, 2020-12-23:
The cart should work. Any cart with a reprogrammable / flashable ROM chip in it will do it.
I think you'll have a hard time programming the chip via an old parallel cable though. You'll have to find guides for that.

devsfan1830, 2020-12-29:
In that picture, check to make sure the red pin (pin 14) is not shorted to ground. Put a probe on the PU-18 test point and a probe on pin 1 of the ribbon connector. In my case excess solder bridged to the yellow highlighted pad which grounded it and caused this issue for me.

ryanm101, 2021-01-02:
This was just posted to a FB group I am in where someone installed MODE on a PS1. 
The QSB looks (as one would expect) very similar to the one for xstation.

```
It's actually not that bad of a install. The DSP chip pins are quite easy to lift (i used a x-acto knife)
However, the QSB can be tricky imo, due to the two small vias that need their solder mask removed, since i had to reflow it twice, as it kept going to the MC bios screen, even though they appeared fine with continuity.
It's caught a few people out.
```

DrDorito, 2021-01-11:
Good news everyone,

After checking and rechecking the QSB and everything else, I decided to start examining the main Xstation board a little bit more. So I booted the PSX with and without an SD card and was getting the same response and I focused my attention to the SD card slot. Turns out, if I hold the SD card pushed in, it works! I'm thinking that maybe there is a slight disconnect within the SD card slot. I will be testing it out a little bit more but I have a replacement board already coming in from Castlemania, which they decided to send me without me asking, but that's fine. Thinking this might have just been a fluke-bad sd card slot, as I was always meticulous when messing with it.

TL;DR
Issue with the SD card slot

ramapcsx2, 2021-01-11:
Okay, it seems a few of these slots are fine on the first few insertion / removal cycles, but then at some point they become bad.
The number of confirmed bads is still low, but we'll look into other slots for the future.

closed 2021-01-14 by ramapcsx2