x-station/xstation-issues #43: PU-18 boots to BIOS/CD Player, Thoroughly Stumped

Guhrant, 2020-12-23:
Hi there,

I received my XStation from Castlemania yesterday and went to install it into my SCPH-5501 right away.  However, I've had some trouble, and no matter what I do it always boots to the BIOS/CD player instead of launching the loader.  I'll try to thoroughly describe my situation.  This unit would not load a game before modding, but I assumed it was a laser issue and so figured this unit was just right.

The install seemingly went fine (I am very comfortable with soldering and regularly hand-change 0603s & other small components at work), and I put things back together and plugged it in.  The system booted to BIOS, and the loader would not appear.  Opening the lid and holding down the lid switch, I observed that the power light (blue) is steadily on, and the green access light does a single blink once or twice before the boot times out.  I am seeing probably the same behavior as https://github.com/x-station/xstation-issues/issues/13#issuecomment-728179541 , because a file "config.txt" is created in the 00xstation folder, which contains a single "square" character.

Between yesterday and today, I probably spent ~10 hours checking and re-checking (and re-checking, and re-checking, etc...) my work, and everything seems flawless.  I have tested for continuity of the flex cable connectors (on both ends), I have tested from each QSB solder point to the test points on the board, I have tested from each pin of the QSB's FFC connector to all test points on the board.  I'll attach photos of my work:

QSB Photos

[image]

[image]

[image]

Lifted Pins

[image]

Lifted Pins again, but with light shown through PCB to show cleanness of pads and pins:

[image]

[image]

Furthermore, I've gone through and checked every relevant (i.e. not PU-8) issue, both closed and open.  I tried a number of things suggested by others and by you, and I'll describe those and the results of each now.

As suggested in https://github.com/x-station/xstation-issues/issues/13#issuecomment-731862964 , I tried reformatting my card.  I thought this would be it, because I am using a similar Sandisk card to the one ryanm101 said didn't work for him, and another person in that issue thread mentioned how the created "config.txt" contains just a square, same as me.  However, I am confident now that this is not an issue with my SD card, because this behavior persists regardless of how I format the card, and does the same thing on this 64GB Sandisk U3 card, a 64GB Netac U3 card (both formatted to exFat with the official tool), as well as on a very old 8GB, probably first-generation Micro SD card.

As suggested here https://github.com/x-station/xstation-issues/issues/30#issuecomment-739451975 , I re-flowed every single solder point with added flux, and re-tested continuity everywhere.  No change.

You mentioned here https://github.com/x-station/xstation-issues/issues/39#issuecomment-747400309 that it's possible for a via to evaporate and keep a signal from getting where it needs to.  I used my multimeter and ensured that the trace signal you described reached the pin on the square IC.  In order to try and eliminate the possibility of a failed via, I followed every signal that I could to its destination, although some go underneath the shield on the top side of the main board, so I couldn't follow those.  Anyway, I found no abnormalities.

brilliancenp mentioned here https://github.com/x-station/xstation-issues/issues/39#issuecomment-747722155 that cleaning up the pads and pins of IC701 with flux fixed his problem.  I did this, but it had no effect.  I also checked for bridges with my multimeter at least five times both before and after re-heating them with flux.  I checked for bridges between every pin and pad on the two modified sides of the IC multiple times and found no problems.  

As you suggested here https://github.com/x-station/xstation-issues/issues/11#issuecomment-723510313 , I opened the console again and connected the original drive's 4-pin connector.  It behaves as you said it should -- drive spins up and keeps spinning when the lid switch is depressed, and stops spinning once the lid switch is released.

 -----------------------

PHEW, I think that's everything.  Do you have any ideas as to what my problem could be?  I am beginning to suspect something wrong with this PSX itself...  But don't know what it could be.  I'd like to make it work with this console so that I don't have to wait for the Christmas-delayed shipping times.  Thank you very much for reading, any assistance would be greatly appreciated!

ramapcsx2, 2020-12-24:
Hey, loader.bin is in /00xstation, right?

Guhrant, 2020-12-24:
Yes, it is (along with update.bin).  Not sure if it makes a difference, but: I'm currently using 1.05, but also tried releases 1.02 through 1.04.

jsnshrmn, 2020-12-26:
I am also running into a similar issue. Got my x-station from castlemania earlier this month and installed in a 5501. Did the install and I just get the cd player / memory card menu. After several checks on the QSB itself, I checked the continuity from the ffc pads of the x-station to the designated through-holes, chip pins, and ground areas on the main board from the issues page and it all checks out. The access light on the x-station just steadily flashes on powerup and I get dumped into the memory card / cd player screen.
differences from this reported issue:

- My psx was loading discs just fine.
- My cd drive doesn't stay on constantly through boot with the door sensor closed. It powers up, pauses briefly, and then powers up again. It does cut when the lid sensor is open.
- I'm not getting a config file written to my micro sd cards.

I've tried two different cards, a samsung 256GB, and a sandisk 128GB, both u3 rated.

Guhrant, 2020-12-26:
Hi jsnshrmn,

One more thing that differs between our cases is the behavior of the access light -- you said it flashes steadily on yours, but on mine it cannot be called "steady." One or two quick flickers immediately upon power up, and two very brief flashes (separated by several seconds) before the system goes to the memory card / cd player screen.

Between the light behavior and the lack of config file, I think that yours is a different problem from mine.

Is it a slow, on-off-on-off cycle?  If so, that sounds like how the X-Station behaves when there is no SD card inserted.  Are you sure you're pushing the card in all the way?  It should click into place, and then you must push on it again to release it (it's a spring-loaded mechanism).  If you're sure that you're inserting the SD cards properly, and if you're able to do this, I would suggest verifying continuity from the pins of the SD card holder.

bahamutfan64, 2020-12-26:
What I think is _imperative_ is that all consoles have to be tested with a **working CD drive** before the Xstation is installed - although most of the time a bad CD drive is the cause of discs not being read, in fact many other things can go wrong with the PS1. After all, several unique ICs are involved in the CD block, and any of the associated circuitry could have failed/corroded. In fact, I had a recent issue where a tested, practically mint CD drive wasn't working in a 5501. Lo and behold, installing the Xstation was a failure as it resulted in a flashing green light. 

Guhrant, I agree that these two are completely separate issues as well. 

For what it's worth, I have noticed that the flex cable can be a bit finicky - it needs to be **fully** inserted into both the QSB and ODE connectors as otherwise you'll get a flashing green light even even in otherwise perfectly operating consoles/successful Xstation installs.

Guhrant, 2020-12-26:
Thanks for the reply, bahamutfan64.

I agree with your sentiment, and mentioned in my first post that I suspect something else wrong with the unit.
I want to have this project done before classes begin for the Spring, so I ordered another 5501 (a working one this time), and will try with that one if I don't have this figured out before it arrives sometime in the middle of next week.

As far as corrosion goes, this unit looks to have been very well taken-care of, and I see no signs of rust or ruptured capacitors.  The traces are all clean and look practically new.

jsnshrmn, 2020-12-26:
> Hi jsnshrmn,
>
> One more thing that differs between our cases is the behavior of the access light -- you said it flashes steadily on yours, but on mine it cannot be called "steady." One or two quick [...]
This was really thoughtful feedback. Thank you!
I verified that the flashing behavior is identical with or without a card inserted.
I have an full sd -> micro sd extender ribbon that I was able to use to check continuity from the pads near the slot. Every pad for the micro sd slot on the x-station did have continuity with a pad on the full size sd slot on the other end of the ribbon. I'm scratching my head.

jsnshrmn, 2020-12-27:
I'll go ahead and open another issue with pictures of my my work since there seems to be consensus that my issue is unique from @Guhrant's

bahamutfan64, 2020-12-27:
@Guhrant yeah I had absolutely no idea how to troubleshoot my malfunctioning 5501 - I tried reflowing all the ICs and recapped the entire board to no avail. Unfortunately I could not find a clear, well-scanned copy of the service manual so that's all that I could personally do. Of course I also checked the Xstation installation as per some of the other threads here but I was pretty confident in my soldering. 

I decided to buy another [supposedly tested] 5501 from eBay and to part off the shell, CD drive, PSU, etc. to essentially subsidize the purchase. I think it'll just make your life easier as well.

Guhrant, 2020-12-27:
@bahamutfan64 

> I tried reflowing all the ICs and recapped the entire board to no avail.

Hmmm, sounds like you covered just about everything that I was going to try.  It would be great to identify what's wrong with these, as it seems like at least a few people have had similar, or the same issue with broken PU-18 units.  If that could be done it would save a lot of consoles from the landfill, which is always a good thing.  However, I haven't had a PlayStation 1 since I was a kid (and I've been more involved with Sega Saturn groups in recent years), so I'm unfamiliar with the PSX hardware, and certainly don't know enough to expect to crack this on my own.

> I decided to buy another [supposedly tested] 5501 from eBay and to part off the shell, CD drive, PSU, etc. to essentially subsidize the purchase. I think it'll just make your life easier as well to focus on the new one once you receive it.

Yeah, I think I'll do that.  So it ended up working for you after you installed it in the new unit?  Oh and by the way, if you have any tips for desoldering the QSB to reuse it, I'll take 'em.

I'll leave this issue open for now, just in case I run into issues even with the new unit, and I'll close it when I've got the X-Station working.  Still open to trying something with the first console if anyone shares a bright idea between now and then.

ramapcsx2, 2020-12-27:
> I verified that the flashing behavior is identical with or without a card inserted.

Well, this should not happen. 
You will only get a constant LED blink when the SD card could not be mounted.
It is actually very simple to debug in that case: Either the electrical connection is bad, or the card has an issue / is malformatted.

ramapcsx2, 2020-12-27:
With regards to the thoroughly checked and debugged, yet still not booting consoles: I'm sorry guys, there is not much I can add that hasn't been said before.
The xStations leave the factory tested and working, the box and antistatic bags protect them from damage in shipping.
It is safe to assume that they are alright, and that the error is somewhere else. 
Keep debugging, is all I can recommend now ><

jsnshrmn, 2020-12-27:
> You will only get a constant LED blink when the SD card could not be mounted.
> It is actually very simple to debug in that case: Either the electrical connection is bad, or the card has an issue / is malformatted.
Thank you for helping me narrow this down. I've verified continuity between the mainboard and the x-station, and then between the micro sd card pads and an inserted ribbon adapter. I haven't checked voltages or anything, though, and I'm not sure what I should be looking for.

I'm now on my third micro sd card, this time a 64GB samsung card that I pulled from a working raspberry pi. I formatted with the recommended sd formatting tool in windows:
[image]
and dropped the 1.0.5 release under `00xstation`
[image]
I've looked over the mini faq several times now. Am I doing anything obviously wrong here?

ramapcsx2, 2020-12-28:
Could you check without the adapter?
If it's still the same, could you take a quick video of the LED blinking? I'd like to verify we're on the same page then :)

jsnshrmn, 2020-12-28:
I was only using the adapter to check that I was getting a good connection through the card slot, since that was the only way I could think to check it. I've been running with the cards directly inserted.
Flashing x-station attached:
https://user-images.githubusercontent.com/2986893/103188136-ea645f80-488c-11eb-9cb9-07cde307e79c.mp4

jsnshrmn, 2020-12-28:
Also, I reflowed ic701 just to see if maybe there was a loose joint from pulling pins, but it didn't make a difference. I rechecked to make sure I didn't create any bridges.

ramapcsx2, 2020-12-28:
Okay yea, the SD card is not getting mounted.
You formatted it right, so there's no reason it should not mount correctly.
Maybe the card slot is busted..

bahamutfan64, 2020-12-28:
Rama, these seem most likely to be PS1 hardware issues and not card slot issues. Would you be able to touch a little bit on how the Xstation interacts with the CD block? That might help with everyone’s troubleshooting.

jsnshrmn, 2020-12-28:
Yeah, I have good contact with the sd card slot, which I verified with a continuity check. I'm fairly confident that I haven't done anything silly with the installation. Although the system was loading discs before install, I suspect the 25 year old board over the new x-station. If there are any other points I can check for continuity or other electrical characteristics than I'm happy to check.

birdybro, 2020-12-28:
The board has a few potentiometers on it, are these related to the xstation, and if so would them getting bumped and slightly modified effect this functionality?

ramapcsx2, 2020-12-28:
No no guys, there is a clear path to get a blinking access light like that, and it has nothing to do with the PSX.
You could just power the xStation board alone and it would do the same thing.
The SD card (or file system) is not getting recognized.

jsnshrmn, 2020-12-28:
Should I be checking voltages coming through the slot then? Not sure what to look for there.

ramapcsx2, 2020-12-28:
I recommend you return the unit for a replacement. If 3 cards are not even getting mounted, there is something wrong with the hardware. My guess is the slot.

DrDorito, 2020-12-28:
This sounds exactly like my issue. I wonder if there's any way to test the connections of the sd card slot effectively. I emailed castlemania, but I'm not sure that's who i should reach out to about an RMA or something.

jsnshrmn, 2020-12-28:
> 
> 
> I recommend you return the unit for a replacement. If 3 cards are not even getting mounted, there is something wrong with the hardware. My guess is the slot.

I certainly seem to be running out of things to debug. I did a continuity check from my ribbon inserted into the sd card slot and was able to trace every pin to either the esp32, the little dip between the card slot and the esp32, or ground.

bahamutfan64, 2020-12-28:
@jsnshrmn Yeah exactly, which is why I am very curious as to the principle of how the SD card is being mounted (i.e being recognized by the PS1 as a disc), since that's the most obvious point of failure here. 

Out of curiosity Rama, if the lifted legs of IC701 are connected back to their respective pads and the QSB remains, would putting back a CD drive restore disc reading functionality?

jsnshrmn, 2020-12-28:
I did notice that 3 of the pads on that dip are about 0.5mm shorter than the rest. Is that expected?
[image]
Also, I found more than 1 pad for the card slot seems tied to ground. Is that expected? I'd only expect 1 pin to be ground, but I don't know too much about it.

jsnshrmn, 2020-12-28:
ah, that first pad (if I'm counting my pins correctly) is only grounded when a card is inserted. That must be for no-connect, meaning this is running in SPI mode? In that case, The other pad that's grounded is number 4. I did these tests on the pads, not the ribbon, so those could be in different order in the slot.

Guhrant, 2020-12-28:
Hello all, 

Good news!  The additional PS1 that I purchased arrived quite early, and I received it midday today.  After a careful installation (which went quite quickly, as I've become VERY familiar with the PU-18 motherboard through my first attempt and subsequent debugging), the X-Station booted right up and seems to work flawlessly! :smile: :smile: 

[image]

So, as @bahamutfan64 suggested, this was indeed a problem with my PSX.  The **cause** of that problem remains a mystery, but oh well.

Thank you all very much for your comments and assistance.

-----------------------------------

As promised, I'll close this issue as my unit now works.  Perhaps further discussion of @jsnshrmn's issue should be taken to #44, for posterity and for the sake of those that may read these threads in search of answers in the future.  Perhaps a post could be made there, informing people that the start of the conversation is in this issue.  I'll leave that up to you all.

closed 2020-12-28 by Guhrant

ramapcsx2, 2020-12-29:
Just to reply to the open questions here:
The SD card slot should have one grounded pin if no card inserted, and 2 grounded pins if a card is inserted.

The resistor array looks fine to me. The rightmost pins on it should tie to 3.3V from the regulator.