x-station/xstation-issues #44: Another PU-18 boots to BIOS/CD Player, Also Thoroughly Stumped

jsnshrmn, 2020-12-27:
I have an SCPH-5501 whose x-station has a fast blinking access light upon boot, which then dumps me into into the cd player/ memory card screen. I get the same behavior with or without an sd card. After going through the [issues mini faq](https://github.com/x-station/xstation-issues/blob/master/README.md), I also looked at #43, and its references to:

[#11 (comment)](https://github.com/x-station/xstation-issues/issues/11#issuecomment-723510313)
When connected to power, my cd drive doesn't stay on constantly through boot with the door sensor closed. It powers up, pauses briefly, and then powers up again. It does cut when the lid sensor is open.

[#13 (comment)](https://github.com/x-station/xstation-issues/issues/13#issuecomment-731862964)
I've tried two different cards, a samsung 256GB, and a sandisk 128GB, both u3 rated. I've tried both exfat and fat32 format with no success. Additionally, I checked the contacts on the microsd slot by plugging in an sd card -> micro sd extender and verified that I was getting good contact.

[#30 (comment)](https://github.com/x-station/xstation-issues/issues/30#issuecomment-739451975)
I've gone through and tested continuity as far out on the board as the indicated markings go (including chip pins and via holes) back to the pads on the x-station with the fcc connected. That all checked out. I think my qsb and conductor wire look pretty good too.
[image]
[image]

[#39 (comment)](https://github.com/x-station/xstation-issues/issues/39#issuecomment-747722155)
Pulling up the pins on ic701 went smoothly. I checked the pads for continuity with neighbors, but didn't find any bridging.
[image]

devsfan1830, 2020-12-29:
After troubleshooting THIS issue (https://github.com/x-station/xstation-issues/issues/47) I am now faced with this same issue. Tried two memory cards. A cheap 16GB Micro Center one then changed up to a 128GB Samsung EVO Select U1. Each time the white boot logo appears but runs for quite a while and then times out and goes to the cd player/memory card screen. I popped open the lid and held the lid close switch down and booted. The blue light is solid lit, the green led eventually flickers rapidly which appears to be it accessing the card, but then it stops as the boot times out. I have triple checked the lifted pins, continuity tested every solder point on the QSB to both the ps1 board test points as well as between each QSB pad and the ribbon cable connector. Ribbon cable itself has continuity on every trace. My closed issue linked above has photos of my install. I've ordered two spare QSB and ribbon kits to see if maybe the ribbon is somehow still damaged OR the QSB is somehow. The second spare being if I need to ebay another 5501. It worked fine before with the drive. I even tried reflowing the un-lifted pins to see if maybe I knocked one loose. Dunno what else to try at this point.

jsnshrmn, 2020-12-29:
I have another x-station on the way. I'm also getting my hands on another working 5501, so, I'm optimistic that we'll get to the bottom of my issue.

ramapcsx2, 2020-12-29:
@devsfan1830 
My guess is a cold solder joint on your QSB install.

@jsnshrmn
"fast blinking access light upon boot ... the same behavior with or without an sd card"
If the blink rate is the same with and without SD card, then the card / filesystem is not getting mounted.
In that case, you can narrow down your debugging to just the SD card and the xStation cart slot.

devsfan1830, 2020-12-29:
I found the culprit on at least MY install and it may be worth others checking. It has to do with the cluster of 3 solder points for pins 4,  14 and ?. In my case, excess solder caused pin 14 to short to ground. I used solder wick to remove much of it and left just enough to connect these pads. My xstation now properly boots to the menu and I can boot games! I determined this by probing between each of the test points identified on the board and the ribbon connector pins. I discovered continuity between pins 1 and 14 on the test point for 14. I suspect it was due to the yellow grounded connection as the trace for 14 runs JUST below it. I had a solder blob that extended over it and i suspect made a connection. That or a the blobs on the QSB were touching and it I missed it when inspecting it with my magnifier.

[68747470733a2f2f6d6d6d6f6e6b65792e636f2e756b2f77702d636f6e74656e742f75706c6f6164732f323032302f31322f7853746174696f6e2d353530582d5153422d6d61702e706e67](https://user-images.githubusercontent.com/27990284/103312798-1e09cb80-49ec-11eb-8460-9451c0b4735f.png)

bahamutfan64, 2020-12-29:
@devsfan1830, that's great! I'm glad you figured it out. 

@jsnshrmn I've some news: I received a **new** Xstation with a **new** QSB and installed it into a tested and working PS1 I recently got from eBay - 100% success. I then switched in the **old** Xstation and it I got the flashing green light.

Having isolated the issue to the **old** Xstation, I reflowed absolutely everything on the PCB, cleaned out the micro SD card slot with 99% IPA, and verified there's continuity everywhere. I inserted the micro SD card, and green light again. Feeling defeated, I tried out a micro SD to SD card extender/adapter instead of a direct connection and lo and behold, the green light stopped and I 
was able to finally boot into the Xstation menu! 

I tried out the SD card by itself again, and got the flashing green light once more. Switching back to the extender = Xstation menu. 

So it seems like Rama was 100% right that (at least in a case such as this) that there's something odd about the card slot. I don't know if there was a bad batch of slots or what, but it appears to have solved my issue with the old Xstation.

ramapcsx2, 2020-12-30:
Nice, so 2 issues figured out :)

Solder blobs get pretty large if the QSB isn't pushed flat onto the psx board. This may have been the problem :)

The cart slot thing, well, the adapter could simply exert more pressure and thus make contact with the pins inside.
These slots are pretty good, and it was quite hard to find ones that have consistent good quality.
Nevertheless, they can still break or develop some issues after little use. If you can't bend it right again, it'd be best if you RMA the xStation. It's no fun with an unreliable SD card.

jsnshrmn, 2020-12-30:
Yes, I was able to verify this was an issue with the card slot; inserting my ribbon adapter past the point of locking and holding it allows it to work as expected. I'm going to see if I can either correct the card slot, replace it, or just solder in an adapter.

ramapcsx2, 2020-12-31:
Alright, Ryan knows, in case you want to return it.

closed 2020-12-31 by ramapcsx2

jsnshrmn, 2021-01-05:
I just wanted to drop a note to say that I soldered in a new card slot last night and the unit is working just fine. I'm going to be using this with an extender + 3d printed mount, so I think it should stay pretty reliable even if the new slot isn't of great quality.