x-station/xstation-issues #45: Audio skipping on Torneko: The Last Hope

birdybro, 2020-12-27:
I have not tested this on original hardware yet but it's intermittent and kinda bizarre. Audio skips, almost like when you would have a bad audio cd, garbled audio skipping around. Happens rarely. Maybe once every 10-30 minutes of gameplay in dungeons. I'm going to burn a cd and try and recreate it on my psOne as I don't have the original disc atm (at my grandmother's). 

http://redump.org/disc/18414/

My image used matches this redump.

ramapcsx2, 2020-12-27:
Hey, thanks for going through the trouble for the demo! :)

Think I can just let the game run for a while and run into the bug?
And will it recover by itself when it happens, or keep on glitching?

birdybro, 2020-12-27:
It recovers by itself. It usually happens when the song is about to end, and it almost sounds like it's rapidly skipping through the song again, and then it goes back to normal.

ramapcsx2, 2020-12-27:
Alright, this sounds a bit like an issue I'm addressing for the next firmware update. I hope it is :)

ramapcsx2, 2021-01-14:
Quick update: 
Thanks to the nice music in this game, it wasn't too hard to do a stress test on the new firmware :p 

It seems to be fine for the tested (so far) 2-3 hours when standing around in the Mansion Dungeon LV1.

birdybro, 2021-01-15:
Ohhh very good to hear. Can't wait to try it!

ramapcsx2, 2021-02-01:
Tested with the beta yet? :)

birdybro, 2021-02-01:
Where do I get it? :)

birdybro, 2021-02-02:
Seems like it's all good. I'll reopen if I see the problem come back. I tested for 3 loops of the song (loops are indicated by the song fading out since it's the end of the cd track), which it would always happen by that point.

closed 2021-02-02 by birdybro