x-station/xstation-issues #48: Issue after install on PS1 SCPH-1001 (Late PU-8): PS1 boots up once, then black screen

victorhernandezm, 2020-12-30:
Hi everyone,

I'm installing the Xstation on a PS1 SCPH-1001 with a Late PU-8 Motherboard.

In my first try, it didn't worked. The PS1 booted up (closed lid) and it took me to the BIOS screen. After some tries, the boot up screen didn't came up anymore, leaving just a black screen (but giving a NTSC signal on the PVM).

So, i've tested everypoint with the schema published here: https://github.com/x-station/xstation-issues. The points 7 and 9 were joined by mistake, so i've solved that issue fixing the soldering. Everything seems to be OK now. I've tested every point again.

Well, here comes the weird thing.

I've turned on the PS1 with closed lid, it booted up with the Sony logo, it stuck there a little while, and then it took me to the BIOS screen. Here, the screen started to glitch. The menu options started to blink, then the screen started to change it patterns, some crazy glitches like a cartridge-based retro game, and i turned off the console. Turning on again the console only gives me a black screen.

Now i'm stuck. I don't know what to do, considering that i've tested the soldering previously. Now i'm formatting the micro SD and i will try again with it.

Any ideas? Thanks in advance!

bahamutfan64, 2020-12-30:
Visual glitches in the PS1 main menu are typically indicative of issues with the CPU, GPU, or BIOS, and not the Xstation.

I would reflow these three chips, as it's possible the pins are no longer making good contact with the traces.

victorhernandezm, 2020-12-30:
> Visual glitches in the PS1 main menu are typically indicative of issues with the CPU, GPU, or BIOS, and not the Xstation.
> 
> I would reflow these three chips, as it's possible the pins are no longer making good contact with the traces.

Maybe, but remember that it didn't happen in my first try. That behaviour changed suddendly.

ramapcsx2, 2020-12-30:
Did you remember to disconnect the PSX CPU oscillator?
https://i.imgur.com/5iGkVr9.png

birdybro, 2020-12-31:
Can you provide some decent pictures of the front and backside of the board with your cellphone camera or something like that?

bahamutfan64, 2020-12-31:
> Maybe, but remember that it didn't happen in my first try. That behaviour changed suddendly.

I've had a case where while screwing down, the motherboard would flex a little and would cause the RAM chips to no longer make contact, leading to a black screen. Reflowing everything solved it.  Things work until they don't, and a lot of times it does happen suddenly with no prior warning and very little indication of what actually is the issue, sadly.

ramapcsx2, 2021-01-12:
Any updates here?

ramapcsx2, 2021-01-14:
Reply if you need this reopened.

closed 2021-01-14 by ramapcsx2