x-station/xstation-issues #49: SCPH-1001 Not booting into loader

Barawer, 2020-12-31:
Hey,
After installing everything i cant get the Xstation to boot. The green access light blinks at a steady pace, but nothing more.
I've tried making sure the lid is closed with something that holds it, not change.
I made sure every pin is connected based on the diagram in this page. I checked for shorts and seems fine.
I tried 2 SD cards, on exFAT and FAT32, but no change. Haven't tried older versions though.
Any help would be appreciated! 

[image]
[image]
[image]
[image]

birdybro, 2020-12-31:
Can you post a picture of the whole qsb and anything you had to solder? Your pics only show certain parts.

Barawer, 2020-12-31:
Sorry, thought I posted the whole QSB. Hope its clear enough, if not let me know
[image]

Barawer, 2020-12-31:
Well! after reading some of the past issues here, I tried connecting the SD with an extension cable, and it worked! guess i got a bad slot?

birdybro, 2020-12-31:
I wouldn't jump to conclusions totally yet, even though that is a decent guess. There is variation on microSD slots that are sold online, even the better ones, and that does happen time to time (even with higher profile devices like the DE10-Nano and the Raspberry Pi, I have received some with faulty MicroSD slots before). 2 things I noticed with your soldering (1 which probably is unrelated to your problem, the other which might be related).

1. Your IO16 is too long and is running over a metal pad that has pressure against it once the ps1 is sealed up, this could cause problems at some point and should be fixed. Nothing major though.
2. The solder from the QSB to the capacitor in the bottom left seems to be completely disconnected from what I can see. Use some flux, tin your tip, gently press the board down and get a little solder in there, and hold the board till it's cool.

[image]

Barawer, 2020-12-31:
> 
> 
> I wouldn't jump to conclusions totally yet, even though that is a decent guess. There is variation on microSD slots that are sold online, even the better ones, and that does happen time to time (even with higher profile devices like the DE10-Nano and the Raspberry Pi, I have received some with faulty MicroSD slots before). 2 things I noticed with your soldering (1 which probably is unrelated to your problem, the other which might be related).
> 
>     1. Your IO16 cable is too long and is running over a metal pad that has pressure against it once the ps1 is sealed up, this could cause problems at some point and should be fixed. Nothing major though.
> 
>     2. The solder from the QSB to the capacitor in the bottom left seems to be completely disconnected from what I can see. Use some flux, tin your tip, gently press the board down and get a little solder in there, and hold the board till it's cool.
> 
> 
> [image]

Thanks! i did notice the gap in the bottom a few minutes after posting the picture, fixed that, sadly that didn't make a difference.
Regarding the cable, would you suggest I solder a new shorter one? any other places I should avoid going over?

Even though I did try running a few games with the extender and it seems everything works perfect, I would rather have a better install overall.

birdybro, 2020-12-31:
The main thing that too large of a gob of solder can do is it can cause a cold solder joint internally that you can't see on the outside. But given that the extender fixes your problem, it's possible that the extender's connection is more solid than the average MicroSD card. Possibly it's just marginally thicker and has a better contact. One thing you can test is when you press in the MicroSD card, just slightly push up on it to make it stick to the contacts just a bit more, and see if that clears the issue. If that does, then you know that the microSD slot is not pushing the contacts flush currently, and you might be able to fix that with just barely pressing down the top of the slot to just barely bend it to an almost undetectable degree :P

ramapcsx2, 2020-12-31:
Hey,
please also try repeated insertions / removals of a preferably not very valuable SD card.
This will exercise all the slot mechanics, scrub off any oxide from the pins, etc.
It might just work after that.

Barawer, 2020-12-31:
> 
> 
> Hey,
> please also try repeated insertions / removals of a preferably not very valuable SD card.
> This will exercise all the slot mechanics, scrub off any oxide from the pins, etc.
> It might just work after that.
> 
> Edit: Oh, and can you tell me whether the slot behaves as expected mechanically? Does the card snap into the locked position easily, or is there an issue with it at all?

I have tried insertions, no real difference. Maybe it needs more. 
The slot works fine mechanically, no real issue there. I did try what @birdybro suggeted and it did work when i pushed the micro SD in a little bit more (before it locks in or comes out), so it does seems like not enough contact in the slot

ramapcsx2, 2021-01-01:
Ok. You should RMA the xStation then.

Barawer, 2021-01-01:
> 
> 
> Ok. You should RMA the xStation then.

Thanks. I will talk to them about RMA.

closed 2021-01-01 by Barawer