x-station/xstation-issues #50: Feature Request: CHD support

birdybro, 2020-12-31:
Even though I know you are very busy, supporting CHD's would probably be of decent benefit, however I'm not sure if it is achievable since it's a compressed format.

CHD is a compressed format developed by the MAME team for disc images which combines the CUE and the image file (bin/etc...). The compression is superior to doing a regular zip of the bins, and is designed to be used with emulators and even works in some FPGA cores on the MiSTer (work in progress, recent pull requests).

[image]

Example photo of the kind of difference the compression creates.

https://github.com/mamedev/mame/blob/master/src/lib/util/chd.h - this has all the information for the file format, I believe.

ramapcsx2, 2021-01-01:
Hey,
it's a neat concept but I'm afraid the extra load of uncompressing data on the fly is a bit too much for the processor.
I think it may barely be able to do it, but then there is no more headroom for any other variables.

But good on the MAME team to combine image and cue file. That's definitely forward thinking!

birdybro, 2021-01-02:
That's what I figured, the esp32 is pretty capable, but decompression on the fly is probably a bit much ;)

closed 2021-01-02 by birdybro

ramapcsx2, 2021-01-03:
Yeah, doing it on top of the other real time work is.

Firehawke, 2021-01-31:
Well, if you ever do a second revision hardware it might be worth looking into the idea of picking one that can handle CSO, CHD, and EBOOT compressed images.

ramapcsx2, 2021-02-01:
But it would bite with the mission of having an affordable ODE. We'll see :)