x-station/xstation-issues #51: FMV skipping most - not all - of the time (also audio)

ianinthefuture, 2021-01-02:
I've just received back my PS1 (PAL 5502) with both a PS1Digital and XStation installed - both do work, have been updated by me to latest firmware, and both have been well installed as far as I can tell, though I'm no expert.

But I've been seeing FMV intros skipping and glitching - the thing is, **it doesn't happen every time** and, **sometimes, is fixed by a soft reset**. But it isn't always fixed that way - **it might take one reset, it might take 10**, it's completely inconsistent.

- I've mainly been testing using Tekken 3, it has three intro vids. Sometimes the main one skipped, the second one didn't and the third one did. Sometimes they all did. Sometimes none did. As I say, it's inconsistent.

**I've made some vids of Tekken 3, here:** https://drive.google.com/drive/folders/1-j5Y-L_8b7UH5fKPrcugdy-lrJLWf-U_?usp=sharing 

- Trying it with Command and Conquer Red Alert Retaliation either results in both the FMV intro and in-game music skipping, or neither skipping.

- Trying it with G-Police is an interesting one, as it has about six separate videos to open the game (three logo vids, three story vids) - sometimes the first and second skip, the third, fourth and fifth don't, and the sixth does. Sometimes they all skip. Sometimes the first, second, and sixth don't, while the others do. I've no idea of the pattern.

- I've tried with five different SD cards, four micro, one full size - Sandisk Extreme 64GB U1 / Sandisk Ultra 64GB U1 / Sandisk Extreme Pro 1TB U3 / Lexar 1667x 128GB U3 V60 (full size SD) / unbranded U3 256GB U3, makes no difference.

- I'm using the Laserbear 3D printed SD card tray, but have tried without it and it's the same problems.

- I've tried multiple versions of Tekken 3, mostly Redump.

- I've even tried a different PSU, as I had a PicoPS1PSU available - I think that actually made it worse, but either way neither the Pico nor the original PSU stopped the skipping.

Trying to think of anything else - basically I've tried everything I can reasonably think of, and nothing's helping. Games work, the HDMI output works, it's just this skipping problem cropping up.

I've also seen some green artifacts/glitches on screen sometimes (they're in the videos linked above), though only when using 960p mode OR when using passthru on the PS1Digital. Aware that makes that not an XStation issue, just something to mention here for the sake of it.

Test045, 2021-01-02:
I've also noticed some random skipping on the PAL version Crash Team Racing introduction (PAL 5502 as well), especially when N. Oxide leaves his spaceship to join earth, as well as when characters are presented following that scene. 
It is better if not perfect from official CD rom. But since it so random, I gave up trying to find a way to reproduce and thus didn't report anything yet.
Also I didn't install the PS1Digital, I only installed the Xstation myself, so there are no other variables here.

ramapcsx2, 2021-01-03:
Ok, I've tested Tekken a few times due to similar reports in the beta phase.
The results were inconclusive as well: Sometimes it happened for a tester, sometimes not. 
It never happened here, no matter the card used (except the outright high latency bad ones).

My guess is that the games (and others like it) produce an access pattern that the SD card doesn't like.
So some SD cards might do this better than others, but it's not the cards fault per se.

I'm working on the next firmware update, and part of it is SD access pattern optimizations.
Hopefully this can fix the issue for you :) 

(Though I must say that it should not happen on 5 different cards. Something else might be >also< amiss there.)

ianinthefuture, 2021-01-04:
Thanks for replying, appreciate it. Do you have a rough idea when the next firmware might be out?

I'm just thinking it'll be quicker/cheaper/easier to just update fw and see if that fixes it before sending it back to the installer - but if it's still a ways off then no worries, I'll just send it back to him this week!

ramapcsx2, 2021-01-04:
It is unlikely to be an installation issue, but your installer may have an extra xStation or QSB-installed test mainboard, so that he may be able to close in on the problem.
No matter what the next firmware does, you should not have problems with so many (known good) cards.
So if you could ask him to help debug the problem, that's what I would recommend doing.

I've been working on the next firmware daily for 2 weeks now. It is a tough challenge, so I can't tell yet when it'll be done.

ramapcsx2, 2021-01-08:
As an update, I'm still on it and SD performance looks much improved. Once I finish mapping this to the PSX expected timings, this may fix your issue :)

ianinthefuture, 2021-01-08:
Thanks for the update - I've sent it back to the installer so he can have a look over things, in case there are any obvious issues. I agree with what you said previously, that it's unlikely to be an issue with installation (given there are 0 faults beyond the skipping), but worth a look anyway.

Hopefully the firmware does improve things!

jamesrpatterson, 2021-01-30:
Hi, I'm the installer!

I've checked the install using the images to test continuity and double checked the lifted pins etc. No issue there.

As advised by @ianinthefuture, I've updated the board to use the 1.1 beta firmware and it has definitely improved things.

If I run Tekken 3 now it will play the intro FMV without any noticeable skipping/jitter.

However, if I reset the console and let the game start up again it almost always starts having the problem again.

Switching the console off, on, selecting the game, etc. and the game loads fine again. I can switch the console off and on and load the game over and over and the skipping doesn't happen. It only comes back when resetting the console and the game loads automatically.

Hopefully, @ramapcsx2, this is helpful to you when debugging the issue. If you need any more information please let me know.

ramapcsx2, 2021-01-31:
@jamesrpatterson and it does this with every micro SD card you can test, with or without extender (if you have one)?

ramapcsx2, 2021-02-01:
From the pure randomness of things, it could be something like a slightly corroded lid or reset switch.
@jamesrpatterson Could you try a drop of Deoxit on both switches, then pushing them a few times?

It's possible that one of the switches is permanently sending a jittery signal to the xStation, which would stall it, but not cause a full reset or lid open action.

jamesrpatterson, 2021-02-01:
Thanks @ramapcsx2. I'm not going to be able to test further until Wednesday but I'll update this thread as soon as I do.

jamesrpatterson, 2021-02-03:
Very disappointing. Today I've been completely unable to replicate the pattern I was seeing a few days ago. 

Turning the console off and on, 4 out of 5 times it was fine. Resetting the console, it always seems to skip.

So… it's essentially the same as before the 1.1 beta update.

ramapcsx2, 2021-02-03:
Try the xStation on a different console. It might be some obscure chip damage..

ramapcsx2, 2021-02-21:
Would be nice to hear back on what happened on a different console, but it's okay if you don't have the time / resources :p

closed 2021-03-08 by ramapcsx2