x-station/xstation-issues #52: PU-8 (Early)

ryanm101, 2021-01-05:
@ramapcsx2 

Based on your images below I've traced the following connections out, I just want to confirm they look correct to you before I get to soldering.

Also I assume other than this wiring all other parts of the PU-8 (late) guide are applicable. 

|FFC_Pin|ESP_Pin|ESP_GPIO|BoardConnection|Notes|
|-|-|-|-|-|
|1|GND|-|-|-|
|2|-|-|5v?|LDO IN|-|
|3|16|27|PSNee|-|
|4|15|26|26|-|
|5|14|25|Subq|-|
|6|13|33|CLOK|-|
|7|12|32|data|-|
|8|11|35||NC????|
|9|10|34|xlat|-|
|10|8|39||NC???? |
|11|GND|-|-|-|
|12|25|16||NC????|
|13|5|36|SQCK|-|
|14|24|4|4|-|
|15|34|5|5|-|
|16|38|19|SENS|-|
|17|42|21|RST|-|
|18|GND|-|-|-|
|19|39|22|SC|-|
|20|36|23|Limit Switch|-|

- [ESP Pinout](https://camo.githubusercontent.com/3dd25d29ae9d93139e7f2fbca59c424117ff10d8115ec3a6897e0c309397cce3/68747470733a2f2f322e62702e626c6f6773706f742e636f6d2f2d484e794a55744334684c552f574d775a7a44427a5759492f4141414141414141414c492f56734953555850427645595069354a4a4d4e475759773432705432683771504b77434c63422f73313630302f70696e6f75745f77726f6f6d5f70696e6f75742e706e67)
- [PU-8 early connections](https://user-images.githubusercontent.com/6278722/103180809-9e6be780-4899-11eb-8ef3-3578e8d3606b.jpg)

ramapcsx2, 2021-01-08:
Oh gee, a large table :/
So the "5v?" is actually the 8V supply, so that's correct, just a different voltage.
I'd get rid of the "ESP Pin" vs "ESP GPIO" moniker, as that is too confusing.
Just use the ESP GPIO as they're named GPION in the pinout.

Once the table is updated, we can properly place missing signals like the APLL (ESP GPIO16).

ryanm101, 2021-01-09:
@ramapcsx2 Ok updated

ramapcsx2, 2021-01-14:
Oh, I forgot to update you. Sorry ><

So here are some of your missing signals:
- GPIO35: LID_SWITCH_IN
- GPIO39: not used
- GPIO16: APLL_CLK_OUT (goes to the CPU oscillator)

If you need the unlabeled ones as well:
- GPIO4: BCK_OUT_CDCONTROLLER
- GPIO5: WS_OUT_CDCONTROLLER
- GPIO26: DATA_OUT_CDCONTROLLER

Please tell me how it went ;)

ryanm101, 2021-01-14:
Do you have pics of where those points are? Or is there a schematic I can reference?

ramapcsx2, 2021-01-14:
This is what I made so far. It's all I have ;p
[image]

ryanm101, 2021-01-15:
No worries, i misread 'unlabeled' as they werent on the diagram instead of not labeled in my table. Doh!

Will do, that's the table updated now

ryanm101, 2021-01-15:
[image]

ramapcsx2, 2021-01-15:
It's the same as usual, you remove the resistor ("151" label) and solder APLL to the usual spot on the mainboard underside.

ryanm101, 2021-01-15:
Do i need to remove the filter cap per the PU-8 Late instructions?
(Sorry for all teh questions just dont want to blow it up :) )

ramapcsx2, 2021-01-16:
You might as well, though it's not required. With the filter removed, the oscillator is not powered, and is not needlessly generating a clock into nothing.
It's preference :p

ryanm101, 2021-01-16:
Ah ok thanks

ryanm101, 2021-01-17:
https://user-images.githubusercontent.com/5185337/104845509-c4e7d600-58cd-11eb-9019-9682049be7a6.mp4

[image]
[image]
[image]
[image]

birdybro, 2021-01-18:
Probably a dumb question, but is that wire shielded or is that bare copper? If it's not shielded, that might be the source of your problems entirely.

ramapcsx2, 2021-01-19:
Well, I can't really help much. I understand what you did there, it should probably work, but definitely use better wires for power (8v) and GND.
For what it's worth, when I built an EARLY PU-8 setup for testing, it worked right away. The code supports that version fine.

ryanm101, 2021-01-19:
Ha nothing ever works 1st time for me (well the last 5502 i did did :) ) i'm not that good or lucky. 
Do you have any examples of oscilloscope / logic analyser output i can 'look' for to detect a problem. I can beef up the power and gnd wires (I only connected 1 of the GND points as it looked like it was a common ground that was being used). 

I was mainly hoping that a second pair of eyes could check i'd not made any obvious wiring mistakes. Continutity on all points was tested ( i just tested for a beep not resistance.) 28awg enameled wire was used, i figured it would be ok for those connections & distances.

@birdybro No fair call, i'll verify i've not nicked the enamel / shorted something, It's not impossible (or even unlikely).

ramapcsx2, 2021-01-19:
Ah, in your installation, there is no ground link for the PsNee gate.
In a LATE PU-8 install, the QSB grounds this point, but you'll have to do it manually.
https://i.imgur.com/j2908Ys.png

ryanm101, 2021-01-19:
Just so I am clear just drop a wire between those two points (i.e. replace the pink lines with real wire?)

Quantum-cross, 2021-05-31:
Any updates on this? Any boards I can be involved in testing? I have this revision PU-8  as well.

Quantum-cross, 2021-06-01:
Also, I'm fine with doing a non-qsb wired install like @ryanm101, but did you ever get your install working after fixing the ground links?

ryanm101, 2021-06-04:
No, it's on the back burner until i get time to look at it again

Quantum-cross, 2021-06-04:
@ramapcsx2 would you release the schematics and pcb files for the 100x qsb? I wouldn't mind giving it a shot myself, it doesn't look like too many modifications would be needed. I'll release back any modifications if it works.

ramapcsx2, 2021-06-04:
Hey,
I don't have much to do with the QSBs, but I can work with you on this.
Let's meet on Discord :)
https://discord.gg/aKgCvUtQ

scp-electrocution, 2021-09-25:
Hi guys, I can confirm that the information you collected to install xstation on pu-8 early board is correct, I followed the table in the first post and everything works, thanks
(temporary installation for testing only)

 @ryanm101 @ramapcsx2 
[image]
[image]

ramapcsx2, 2021-09-25:
Great!
Also, the new QSBs should be ready any day now :)

ryanm101, 2021-09-26:
@ramapcsx2 please let me know when i'd love one or two..

ramapcsx2, 2021-09-26:
I'll probably write something on Twitter and the Discord :)

L10N37, 2022-06-09:
happy to make a QSB for this if you can send complete installation points and their corresponding pins to the ribbon cable header. I have quite a few of these boards going to waste... not that anyone ever gets PU8 systems off me. They're always chasing PU18! none the less.. a project for when I get some time. It's quite relaxing for me to use kicad, and I'll share the design openly when finished. Just need complete info on what to remove etc.

ramapcsx2, 2022-06-09:
Let's chat on Discord :)
I have some images with a couple points, but mostly you'd have to ring the signals out.
Their end points are known and the same as on EARLY PU-8, but where the test points / access are is different.

scp-electrocution, 2022-06-12:
@ramapcsx2 Can you add discord server link?

pdiggitydogg, 2022-06-27:
[image]
I can confirm the points provided work while using a trimmed 100x original QSB.

RickTrajano, 2024-06-28:
Hello,

I followed this discussion to connect successfully my early PU-8 by cutting the Late PU-8 QSB and was successful so thank you for this post. I went ahead and created a diagram to help on my following installations and wanted to leave it here so it helps others that might be struggling with this installation. 
Hope it helps

[image]

[image]

ramapcsx2, 2024-06-29:
Thanks a bunch! One of these days, the actual production QSBs will be available :)

RickTrajano, 2024-06-29:
@ramapcsx2 Do you know why has no one made a QSB for this model? I don't have much experience in designing PCBs but I can't really see a reason why it would more complex than the other ones. But then I've never design something with this many connection and as precise as this needs to be.

Thanks

L10N37, 2024-06-29:
V1 was sent to me for early PU8 and installed last July, he ended up taking 2 late PU8 off me and I sold my system with that customers early PU8 QSB to someone in Cuba. It already exists. I didn't realise it wasn't easily available but there are photos of my installation... I definitely wasn't imagining things :P

L10N37, 2024-06-29:
[image]

L10N37, 2024-06-29:
[image]

RickTrajano, 2024-06-29:
@L10N37  the question is where to get them 😂
Yeah I remember getting some from castlemania at a certain point but they don’t sell the xstation anymore. Thanks for the pic. I wouldn’t mind making some if I had the gerber files

L10N37, 2024-06-29:
You need a ruler and kicad. You should be able to print the attempts out on paper and cut them out and lay them over the board In position with some trial and error until it's perfect. The people who do this stuff all the time have special machines that measure everything precisely for them but you can do DIY methods.

Spacefreak18, 2024-10-01:
I have also managed to get an early PU-8 working using a trimmed late PU-8 QSB.

CactusJonPS, 2024-11-05:
Thanks to this thread I managed to get my x-station working. The wiring below is frankly embarrassingly bad, but hey, it works! Thanks to everyone here for their contributions!

[image]

Spacefreak18, 2024-11-05:
> Thanks to this thread I managed to get my x-station working. The wiring below is frankly embarrassingly bad, but hey, it works! Thanks to everyone here for their contributions!
> 
> ![tempImagecyY0OY](https://private-user-images.githubusercontent.com/187448184/383295674-a13c29b6-9ccd-4a9d-b3fb-51b92f7e7587.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4NDM4NjYsIm5iZiI6MTczMDg0MzU2NiwicGF0aCI6Ii8xODc0NDgxODQvMzgzMjk1Njc0LWExM2MyOWI2LTljY2QtNGE5ZC1iM2ZiLTUxYjkyZjdlNzU4Ny5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQyMTUyNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZjg1OWM3NmNkYmQ5OWIxMTMwZGZlNjdhYWVhODRlYzhjOGQ3MzA1MjI3OGVjZTA3ODcyMjc1NjEyMGM1NWZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jT9l12jcIty8cfHwYdr8QNbk0Lo0j5Nc7JHwDBG8vCU) [ ![tempImagecyY0OY](https://private-user-images.githubusercontent.com/187448184/383295674-a13c29b6-9ccd-4a9d-b3fb-51b92f7e7587.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4NDM4NjYsIm5iZiI6MTczMDg0MzU2NiwicGF0aCI6Ii8xODc0NDgxODQvMzgzMjk1Njc0LWExM2MyOWI2LTljY2QtNGE5ZC1iM2ZiLTUxYjkyZjdlNzU4Ny5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQyMTUyNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZjg1OWM3NmNkYmQ5OWIxMTMwZGZlNjdhYWVhODRlYzhjOGQ3MzA1MjI3OGVjZTA3ODcyMjc1NjEyMGM1NWZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jT9l12jcIty8cfHwYdr8QNbk0Lo0j5Nc7JHwDBG8vCU) ](https://private-user-images.githubusercontent.com/187448184/383295674-a13c29b6-9ccd-4a9d-b3fb-51b92f7e7587.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4NDM4NjYsIm5iZiI6MTczMDg0MzU2NiwicGF0aCI6Ii8xODc0NDgxODQvMzgzMjk1Njc0LWExM2MyOWI2LTljY2QtNGE5ZC1iM2ZiLTUxYjkyZjdlNzU4Ny5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQyMTUyNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZjg1OWM3NmNkYmQ5OWIxMTMwZGZlNjdhYWVhODRlYzhjOGQ3MzA1MjI3OGVjZTA3ODcyMjc1NjEyMGM1NWZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jT9l12jcIty8cfHwYdr8QNbk0Lo0j5Nc7JHwDBG8vCU) [ ](https://private-user-images.githubusercontent.com/187448184/383295674-a13c29b6-9ccd-4a9d-b3fb-51b92f7e7587.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4NDM4NjYsIm5iZiI6MTczMDg0MzU2NiwicGF0aCI6Ii8xODc0NDgxODQvMzgzMjk1Njc0LWExM2MyOWI2LTljY2QtNGE5ZC1iM2ZiLTUxYjkyZjdlNzU4Ny5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQyMTUyNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZjg1OWM3NmNkYmQ5OWIxMTMwZGZlNjdhYWVhODRlYzhjOGQ3MzA1MjI3OGVjZTA3ODcyMjc1NjEyMGM1NWZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jT9l12jcIty8cfHwYdr8QNbk0Lo0j5Nc7JHwDBG8vCU)

doesn't look too bad honestly, just make sure the wires don't get caught during assembly.

what i like about the semi-qsb-less install is that you don't have to worry about the qsb flex putting pressure on the solder points

ramapcsx2, 2024-11-06:
Well, I do have to critique, eventhough I'm happy for you that it's working :)

Wire length: Every wire should be of a fitting length. It's easy to shorten the extensively long APLL wire, for example. 
Make sure to route it away from ground contact points to the shield though.

Using hotglue for anything:
A soldered wire does not easily "come off". There is never a need to glue the work area shut, where you can now not work on this again, should you need to. It's also not doing anything to prevent shorts, if the original solder job was decent enough.

Wire patched? : I think you ran out of wire there.. :p
I really recommend you get a spool of fine signal wire, very small diameter. The small diameter also helps getting more length on a spool for the same price.

ruppertf, 2026-02-27:
Hi everyone, first I like to say a huge thank you to all who made this possible. I've been searching and reading but haven't found a definitive answer. Can I with the lifted pins and a selectable switch on a PU-18 rewire to use both the disc reader and the ODE. I'd like to try to keep a 55xx with disc reader for VCD disc use (an add-on Gamers VCD parallel adapter). Could switching to disc or ODE mode be achieved. I've mocked up placement and all fits. Using a micro SD card reader extender to a slot created out the back.

ruppertf, 2026-02-27:
Thanks @ramapcsx2  most appreciated 👍