x-station/xstation-issues #53: Xstation on SCPH-5500, black out but working

takataka2021, 2021-01-11:
Hello
I installed xstation on my SCPH-5500.
It can boot ans read some games.
After few minutes, it will black out.  I though troubles of PSU or motherboard.
But after few minutes, it is working , but it will be black outed.
I tried two sd card (sandisk and toshiba).
are there any suggestions?

ramapcsx2, 2021-01-11:
Hi, please check your soldering first to make sure every contact is good.
If the problem persists, find a way to run a game with the lid opened (example here: https://gbatemp.net/uploads/gallery/album_245/gallery_746_245_84369.jpg).
Check that when the blackout happens, the blue LED does not go off.
If the blue LED goes off, then there is a problem with power to the xStation that you need to solve. It could be a bad power supply.

takataka2021, 2021-01-15:
Hi,  ramapcsx2.
Thank you,
I checked soldering. and the problem persists.
When black out, blue LED is on. and the other LED is blinking and off.
After some minutes, game will be running.
The images are running on PSIO. I will re-dump them.

takataka2021, 2021-01-15:
I checked solders again.
And I checked bin+cue, img and iso files.
And I swapped the xstation main board.
That was same.
I will try station on an another SCPH-5500.

takataka2021, 2021-01-17:
While I am installing another PU-8 for SCPH-3500, my scph-5500 is running without blackout.
I changed hd link cable of POUND that could not change aspect rario, old version to PS2 HD cable of LEVELHIKE.

This could be a trouble of HD cable??

ramapcsx2, 2021-01-17:
Well, yea, it could.

closed 2021-02-13 by ramapcsx2