x-station/xstation-issues #54: No Games Found message screen frozen

mattd121, 2021-01-11:
I just got my XStation (plus PS1Digital) installed my a recommended modder. I followed all directions for formatting my card, creating the 00xstation folder and copying the firmware files to it. Created some folders with the .bin & .cue files in them. When I boot up I get the Sony splash screens (white one then black Sony logo) and then it goes to a blue screen that says "NO GAMES FOUND, PRESS (TRIANGLE) TO REFRESH LIST!"

At this point no inputs work (I tested various PS1 Dualshock controllers). I've used multiple SD cards as well. It's basically a brick at this point. The modder tested this before mailing it to me and it worked there. Any ideas as to what is happening?

mattd121, 2021-01-11:
Looks like booting to the Memory Card/CD screen with the lid open has revealed that the controller port may be not be working. I've tried 3 genuine controllers that work on my other PS1 and none of them work on that screen either. Is there anything about the XStation installation process that could cause the controller port to stop working?

mattd121, 2021-01-12:
OK the first issue was the controller port not being connected. I pushed in the connector, re-assembled, and now I get a steady blue light and a blinking green light on the XStation. No card reader is installed, just the microSD directly in the input. What does a blinking green light mean? Also, the lid is closed.

ramapcsx2, 2021-01-12:
This sounds like a bad installation overall. How come the controller port was not plugged in? That is a bad sign.

The first time you ran this, the xStation worked. You got to the loader fine, which means every part of it worked as it should.
Since now you're getting a steady blinking green light, your SD card is not getting recognized anymore.
Clean your SD card contacts, make sure you have the same setup of that SD card that you had before.
Try reseating the SD card a few times. It should come up again.

mattd121, 2021-01-12:
I've probably reseated 2 brand new SD cards (Sandisk) about 30 times yesterday so it's definitely not that. Also tried the LaserBear mount again many times. Blew out where the card goes with compressed air. The SD cards are spotless since I took them out of the packaging yesterday. Here are some pics of the installation that maybe you could look at?

[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2021-01-12:
The issue you have now has nothing to do with the rest of the install.
At some point, the xStation lost the ability to mount the SD card.

mattd121, 2021-01-12:
Well I guess at this point I may have a bad SD card slot? Including the LaserBear mount I've now tried 3 different SD cards...

ramapcsx2, 2021-01-12:
If you can rule out any formatting errors, then yea, sounds like a bad slot.
It's just weird that it first worked fine.

mattd121, 2021-01-12:
Thanks. Agreed that it's weird that it worked before. So what's the protocol? Contact CastleMania Games? Hoping I can just replace the board with the card slot since it's not soldered or anything.

ramapcsx2, 2021-01-13:
Yes, please check for RMA procedures with Castlemania Games.
You may want to link to this issue with your request.

bahamutfan64, 2021-01-17:
Hey @ramapcsx2, the following (crude) method seems to have worked for me in terms of improving the connection the SD card extender makes with the slot: cutting the adhesive portion of a Post-It to size and taping it to both sides. Seems to add just a bit of extra thickness to hold everything in place. 

[image]
[image]

mattd121, 2021-01-17:
@bahamutfan64 Wow thank you for posting this!! My XStation is reading SD cards from my LaserBear mount now! Those post-its on both sides are the perfect thickness to hold the insert without it backing out to "click" in place. This isn't how it's *supposed* to work and I still consider it defective but at least I know the problem now.

ramapcsx2, 2021-01-17:
Okay, so the retention clip thingy is the issue.

matjolic, 2021-01-24:
I just had a customer send in a problematic install done by someone else, and I can confirm the SD card access issue. I swapped his Xstation in my own working console to troubleshoot and got stuck on the memory/cd player screen with the access light blinking.  I found this thread and tried to wedge a piece of thick paper in the SD card holder, and now it boots reliably. The only difference I can see with mine is that this one has a gold copper layer (xstation logo) whereas mine has a silver, tin appearance. Both PCBs are rev1.2a

mattd121, 2021-01-25:
I have not heard back from Ryan on a replacement. I've contacted him various times in the last 2 weeks and no response. Is there a plan to replace these boards or any way to get any communication on this?

mattd121, 2021-02-02:
So for those that end up in this thread who have this exact issue, you will not get any support from Castlemania Games for this fix. Post-it notes are your solution.

I've gotten no replies in 3 weeks now from Castlemania Games so looks like there's no chance of getting a permanent replacement for a bad SD card slot. This is really bad customer support in my opinion.

marcopsf79, 2021-02-02:
I have the same issue, received the xstation from Castlemania Games. I cannot believe this is broken...This is sick!

MF

geekmiki, 2021-02-02:
Same here, just installed mine today and it won't read sd cards, boots straight into the memory card/cd player.
They must somehow solve the issue and Castlemania needs to address this!

bahamutfan64, 2021-02-03:
I’ve done about 12 installs so far and half of them have needed a post-it or Kapton tape as otherwise the SD card won’t be detected at all. 

We can’t know what percentage of XStations have these issues, but it certainly seems like the number of faulty card slots is adding up quickly.

ramapcsx2, 2021-02-03:
So there's 2 things going on:
First, the RMA requests are getting handled. 
I'm always getting asked to forward an issue to Castlemania Games, only to learn that the replacement is already in shipping, while the user is asking for updates on social media.
Please use the ticket system on the website and then wait for your replacement.

Second, I ordered a bunch of SD card extenders, and it looks like most of them have a male connector that is thicker and much longer than a regular micro SD card.
Pushing one of these into SD card slot, especially at an angle, can break pins in the fragile slot. These are not made for these stresses.
If you use an extender, please be very careful when you insert it into the slot. Make sure to not force it in, and to never insert it at an angle.
We will update the installation guide with this information soon.

mattd121, 2021-02-03:
Thank you for the update Robert. I'm not on Twitter so I really didn't have any other options to find out what is going on with my ticket other than email which wasn't getting responded to.

Looks like LaserBear has posted an updated installation video that should prevent this issue going forward. Very useful for installers:

https://www.youtube.com/watch?v=uNoj5w_RilE

ramapcsx2, 2021-02-21:
Newest findings on the extenders: The slot facing edge on the extender PCB is commonly not beveled like it should be. On insertion, this can easily just grab some pins and bend/drag them with it. See the pictures here: https://twitter.com/DragonsHoardMod/status/1362473768745058312

All that can be done at this point is a slot replacement / RMA.

At least we know about the problem now and can take steps to mitigate it.
Until then, as always, be very careful with the extenders. Just file down the PCB edge to fix the problem.

closed 2021-02-21 by ramapcsx2

mattd121, 2021-02-24:
@ramapcsx2 if you could contact Castlemania about this issue for an RMA I'd appreciate it (Order #C26137).  I opened the ticket 6 weeks ago, got an initial reply from Castlemania to check a few things that day, then complete silence for these 6 weeks. I don't know what to do at this point. This issue was closed by you but my problem remains without a resolution.

ramapcsx2, 2021-02-24:
I've contacted him about your case, yea.

mattd121, 2021-03-02:
Finally got my PCB replacement from Castlemania and I've gotta say I think I'll stay far away from any extenders from now on. Just looking at the differences here there's no room at the end of the extender to file down a bevel without running into the contacts! Unless there is a serious revision to that extender it's not going anywhere near that SD input.

[image]

JoeTheRetro, 2021-03-03:
^^^ Same here. I just looked at several of my extenders, as I have some generic ones I bought in the past and those two longer bronze pins sit right at the edge of all of my extenders. There's like, zero room to file down. I put my SD card straight into my XStation and just stuck the mount into my PS1 laser hole to make it look nice. I have 300 games on my card. On the rare occasion I want to add one, I'll pop the top of my PS1. Just not worth the risk of an extender to me.

ramapcsx2, 2021-03-03:
Being careful is fine, but I'd like to mention that it's no problem to hit the contacts or even take away a little.
As long as there's a defined bevel in the end :)