x-station/xstation-issues #55: Late PU-8 Xstation won't boot into Xstation menu, works perfectly after reset

KenToad, 2021-01-12:
I've been searching to see if anyone else had this issue. I installed everything myself and it seemed to go well, but the Xstation gives me the insert Playstation CD message after booting. I was relieved when I discovered that pressing the reset button makes it boot into the Xstation menu. From then on, the Xstation seems to perform perfectly, multi-disk iso's, reset button as expected. I do have the late PU-8 motherboard that kills any PAL signal, unfortunately, but otherwise everything is great. I just have to reset the console in order to access the Xstation menu and setting the console to boot into a game from power on doesn't work, since I just get the insert Playstation CD message.

I'm posting here to see if anyone has any tips for troubleshooting or a similar experience. I'm using a high quality Sandisk card formatted with exfat using the official tool, per the instructions. The console seems to work really well and with kids in the house not booting straight to the menu could almost be considered a feature, so I'm loathe to open it back up again, but if anyone thinks that this issue could lead to further problems down the line, I will.

Thanks in advance for any help or advice regarding this matter.

ramapcsx2, 2021-01-12:
That is some odd behaviour. Is there anything else you noticed?
I recommend you unlock the PAL signal in any case, just to avoid getting locked up for some games / homebrew.

KenToad, 2021-01-12:
I haven't noticed anything else.

ramapcsx2, 2021-01-14:
Is this an NTSC-U console? Anything special about it maybe?

sendep7, 2021-01-25:
I'm having a similar issue.  Mine usually goes to the CD player menu more often than not from cold boots.  Usually hitting reset in the menu boots it into xstation menu.  one xstation is loaded, everything works fine. on a pu-8 -22  Just did the install yesterday, opened it back up, resoldered everything, rechecked everything, Pins are lifted. Resisters/Filters are deleted.   *shrug*

bahamutfan64, 2021-01-25:
Maybe bad caps around the CD block area?

ramapcsx2, 2021-01-25:
Well, it can't be a logic (code on xStation) issue, because then I would at least sometimes get this on my PU-8-23 unit.
I wonder if the modchip unlock symbols fail to send on a cold boot for you..

sendep7, 2021-01-25:
Actually after working for a few hours, mine has just reverted to going straight to the cd player menu. I cant get it into the xstation menu.  I guess i'll crack it open and resolder everything again.  I did notice it is a SONY factory refurb unit.  Maybe that has somthing to do with it.  But the board looked pretty good. so *shrug*

bahamutfan64, 2021-01-25:
Is the green light blinking consistently, or?

sendep7, 2021-01-25:
no, it blinks quick like 2 or 3 times at power on, but not after reset it
stays off.  blue light is constant.

sendep7, 2021-01-26:
I resolderd the QSB and used more flux and solder than i normally do. I used capton tape to isolate the lifted pins as well as protecting some of the qsb solder points.  

so far so good, its cold booting into the X menu now, I guess i'll chalk it up to bad qsb install.  3rd times the charm

bahamutfan64, 2021-01-26:
Great, glad it works! I just typing this up before you published your most recent comment; it may not be relevant anymore, but  I think it may have some merit to someone:

I would consider reflowing the IC with the lifted pins, as well as the associated CD chips, since the console basically acts as if it can't even detect a disc at all. Somewhat analogously, I've had consoles which would power on into a black screen with no audio, but at reset would boot up normally. Reflowing the CPU fixed it, so it's very well possible you're dealing with weakened solder joints between the legs and the motherboard, and in your case they're no longer making a proper connection..

KenToad, 2021-01-27:
Thanks for the advice! I'll keep it in mind if I have trouble going forward, but so far so good. It's very consistently working perfectly upon reset. 

I also wanted to mention that my playstation motherboard seemed the same as the model pictured in the late PU-8, i.e. PU-8-23, install instructions. Chip locations and everything looked similar, if not identical, to the photos in the guide. I have the parallel port and composite connections but no S-video port on the back. Also, there is that weird PAL grounding issue, which I guess is somewhat unusual.

I guess that I might have a connection issue, but it's not worth the headache of fixing as long as it works.

closed 2021-02-10 by ramapcsx2

cristobalflorian, 2021-04-09:
I have a similar issue with a PU-8 late, 23, lifted the pins, installed qsb, checked 3 times the test points, format SD (8gb Kingston) with sd formatter and followed the user guide to finish the setup, but I just got a blinking blue light, with lid closed, If I open the lid it oges off, pressing the reset button does nothing, no boot image, no MC/CD menu, so it just get on and wont boot to anyhint, even without the xstation... someone else have the same problem?  _

ramapcsx2, 2021-04-09:
Hey, your issue is different. Is the blue LED actually blinking? Anything on the green LED?
Did you install the APLL wire and did you remove the resistor on the original oscillator as per the installation guide?
Please open a new issue if you need help here :)

cristobalflorian, 2021-04-09:
> Hey, your issue is different. Is the blue LED actually blinking? Anything on the green LED?
> Did you install the APLL wire and did you remove the resistor on the original oscillator as per the installation guide?
> Please open a new issue if you need help here :)

Hi, thans for replying, by APLL do you mean the IO16 from the QSB to the marked location on the board?

ramapcsx2, 2021-04-09:
It's point 12 here: https://camo.githubusercontent.com/1c8ab0e280c04d7aa869175ce70adb2e36547936168dd837fb953dab8d50664e/68747470733a2f2f692e696d6775722e636f6d2f675448673363612e6a706567

The parts to remove are this part in the guide:
https://i.imgur.com/YY7d0a0.png
Guide is here, in case you wonder: 
https://github.com/x-station/xstation-releases/blob/main/xStation_Installation_Guide_LATE_PU-8.pdf