x-station/xstation-issues #56: Weird AV  issue after install (PU-18)

barth0lin0, 2021-01-13:
Hey! 
i'm having an issue with a PU-18 install.  I'm not certain if it's a Xsation issue or simply a console issue. 

Installed the kit without any difficulty on a a customer's console ( the console came without the CD drive so  unfortunately I could not test it before the install...) 

after the install, Xstation booted a first time. Then the video signal dropped. Rebooted the console: black screen.  After i while, rebooted, got a video signal with a distorted playstation logo and startup sound. Then the black screen returned. I can sometime boot to Bios after the distortion happens. I had luck once or twice booting to the Xstation and playing a game for a few minutes before the distortion and black screen. 
Tried connecting on the console my OSSC: resolution changes constantly, rare sync for a second or two and then Sync drops : no signal.
Blue power light and green acces lights seems to work correctly.

What i did yet to troubleshoot the console/install:

-tried 2 different SD cards
-Redid all the QBS soldering
-checked continuity of each point on the MOBO and also the continuity of each pins of the ribbon cable connector on both QSB and Xstation PCB. 
-Ribbon cable is ok
-Checked for shorts between each points
-Swapped the power supply
-composite cable and RGB scart cable
-tried removing the botton metal plate in case of short
-Reflowed the pins next to the lifted one and tested continuity/shorts
-No lifted pins are touching each other or the chip's other pins. 
-Cleaned the QSB and IC701 with 99% IPA. dried with blower. 
-Tried booting the console while pressing on the different IC of the mobo in case of a bad contact. No luck

Only thing i didnt tried yet is removing the QSB and check for shorts under it.... i doubt it would be the case....

So ? Bad install ? Bad console? Bad luck ?  

any advice is welcome !!! Thx 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

barth0lin0, 2021-01-13:
If it can help : the PS1 never boots when the Xstation is disconected. No booting to bios. I don't know if it normal or not!

ramapcsx2, 2021-01-13:
Well, it sounds like either the APLL clock source for the CPU is failing (this is an xStation + APLL wire thing), or the GPU is failing (this is a chip damage / snapped solder joint on the GPU thing).

In any case, this doesn't sound like I could debug this remotely.
You'll have to try and figure out which part of the console is failing, narrow it in a little.

barth0lin0, 2021-01-13:
> Well, it sounds like either the APLL clock source for the CPU is failing (this is an xStation + APLL wire thing), or the GPU is failing (this is a chip damage / snapped solder joint on the GPU thing).
> 

Is there other solder point related to the APLL clock source ? I will resolder another wire to test it. i had continuity between both point...resistor is removed.

ramapcsx2, 2021-01-13:
No, it's a pretty simple thing and in your pics, I can see it's fine.

barth0lin0, 2021-01-14:
Removed completely the QSB, redid all the soldering, same issue
Reflowed IC701: same issue.
PS1 is dead! i wont invest more time with it. Let's hope the QSB will survive a second desoldering ...
thx

ramapcsx2, 2021-01-14:
IC202, IC203, that kind of area is what I would expect for the error description. But you're right, better replace it at this point.

L10N37, 2022-01-12:
If you're anything like me you get a bit cocky after doing so many of these, once that happens you make mistakes. Installed one yesterday as a customer came and picked up from my house. Did another tonight, issues like this. Bridge on IC310.

MegaDelta25, 2022-04-18:
Hi i need help. I accidentally depined the chip above the pin that was suppose to depin for the xstation mod. So i depinned the right pin. I then soldered the l (i dont know of solder would kill it.) I looked and the power was on when i did this. I quickly removed power connecter. After using flux to move the silder throuought the chip above i put jt together and now it doesnt display anthing but the chip is getting power.

MegaDelta25, 2022-04-18:
The chip im talking about is the cdx2925 chip that i was talking about

ramapcsx2, 2022-04-19:
Hey, first of all, don't panic! :)
You can restore any pin lift easily, as long as the pin hasn't broken off.
Best if you take some pictures of the problem. Please show your solder work.

MegaDelta25, 2022-04-19:
[image]

ramapcsx2, 2022-04-19:
... Don't put power on that. It needs a major major fixing.

ramapcsx2, 2022-04-19:
In fact, I think you should look for an installer right away :)

MegaDelta25, 2022-04-19:
1. I wasnt panicing so i dont know where you got that from.
2. Im trying to learn more about soldering and electronics. So i wanna fix this myself. So im looking for a solution or an explanation of the wrong.
3. Dont got time to look for an installer just to fix this and me not learn really anything.

ramapcsx2, 2022-04-20:
Oh, all right.
You need to restore every bridged pin.

MegaDelta25, 2022-04-20:
Thank you.  I will try this tommorow. My supplies come in today but i git work. Not bring rude, sorry if you think i am. But i should've said that i was trying to learn electronics form the beginning.