x-station/xstation-issues #57: Refresh Game List Freezing

lechuck333, 2021-01-15:
I have a 1TB SD card in my Xstation that's basically fully loaded. I had the XStation refresh the list so it can read the games. It went for about a half hour and then froze. I figured that it would take a long time, and eventually it would unfreeze. But that wasn't the case. I'm on the most recent firmware, and I have a PS1Digital installed too (also latest firmware). SD card was formatted correctly.

ramapcsx2, 2021-01-15:
Hey,
you have to optimize the file layout for large game sets.
The guide explains how to do it: https://github.com/x-station/xstation-releases/blob/main/xStation_User_Guide.pdf

As a note, the loader animation does stop after many minutes. The card is still getting scanned though. You can check that it does by having the lid open and watching the green access LED.
Another note: If the access LED seems to be constantly on, then the SD card is in slow crawl mode. Something needs to be changed with the folder / items layout then.

lechuck333, 2021-01-15:
So by optimizing the file layout, do you just mean the putting them into folders (like A-Z)? If so, I've done that.

I started the fresh again. When the animation freezes, I'll check the light. If the SD card is in slow crawl mode, how can I fix that?

birdybro, 2021-01-15:
If you are using Windows to setup the SD card, plug in the SD Card to your Windows PC, open command prompt and navigate to your SD card. Type this:

`dir /b /a /s > filelist.txt`

Now you have a list of all files on the drive with their full path. Paste that list to a pastebin or alternative (i use https://privatebin.net/) and share the link here. That way we could see if there is something wrong with how it's setup.

lechuck333, 2021-01-15:
https://privatebin.net/?d8fa5aa620cd7ad4#E3vFGPVdnLi6zTjw9Dd5AT2Zy6fgSP2E5yo3Yu9NgDvU

Also, the light was on solid green, so it's on slow crawl. So I'm guessing it's something I'm doing.

ramapcsx2, 2021-01-16:
Hm, the layout should work. I don't know where it's choking on.
Someone on Twitter mentioned they had issues with folders containing the # symbol, so try deleting that for a test?

lechuck333, 2021-01-16:
Still giving me problems. Crawling.

Is there a way to generate a list outside of the XStation? It's just a text document, so is there a way to take that filelist I made and edit it so it's useable?

ramapcsx2, 2021-01-16:
One could code a PC app that does this, but that's outside my expertise.
It should just work though, as it did for many others.
I wish I could replicate this. Must be some kind of bad path name or problematic SD card, or even formatting problem..

blzla, 2021-01-16:
You can use 'sed' to keep only the first bin file in a folder from your file list. I haven't actually tried this with a blank games list, so give this a try and report back if it works.
Save this file as games.txt in the 00xstation folder: https://pastebin.com/raw/quAtyUUf
Save this file as gamescache.txt in the 00xstation folder: https://pastebin.com/raw/b1mKNnNj

lechuck333, 2021-01-16:
I just reformatted the SD card. Now I have to put all the files back on. So that'll take all night. I put a few on, and the XStation read the list. If the game refresh doesn't work with all the games, I'll try these documents and send an update.

birdybro, 2021-01-16:
@ramapcsx2 Would the fact that exFAT as a standard has a 255 character limit on filename+directory have anything to do with this?

`F:\Ca-Cm\Castlevania - Rondo of Blood - Akumajou Dracula X - English (J) T+Eng v1 Gemini & Absolute Zero\Castlevania - Rondo of Blood - Akumajou Dracula X - English (J) T+Eng v1 Gemini & Absolute Zero\Akumajou Dracula X - Gekka no Yasoukyoku (Japan) [T-En v1.0].bin`

This is 265 characters technically, 10 over, and that could be slowing down the scan or stalling it.

`F:\I\Imadoki no Vampire - Bloody Bride - English (J) T+Eng v1.01 Bloody Bride Translation Project\Imadoki no Vampire - Bloody Bride - English (J) T+Eng v1.01 Bloody Bride Translation Project\Imadoki no Vampire - Bloody Bride (Japan) [T-En by Bloody Bride Translation Project v1.01] [i].cue`

This has 289 characters.

ramapcsx2, 2021-01-16:
Oh yes, that would definitely not work. But I coded (and tested) this, and the result should be that these files are skipped.
The limit is definitely 255-1 characters for a full path + file name.

@lechuck333 It may be too late now, but you could copy your files in batches, then see whether they still get scanned okay.
This would tell you where it chokes, and if we can narrow it down, I could replicate and fix it.

lechuck333, 2021-01-16:
I did originally have it formatted to exFAT, so that's probably what happened. I formatted it to Fat32 this time to see if that would make a difference. My games are 70% loaded right now, so I'm going to let it finish first. If it still gives me the issue, my next plan was to clear the SD card and put games on letter by letter rather than all at once.

I will keep you posted if this fixes the problems or if I have any errors.

lechuck333, 2021-01-16:
So my game list loaded this time, but now I'm having a different issue. I loaded Spyro, but the game I got certainly was not Spyro. I tried another game, and it loaded something else.

ramapcsx2, 2021-01-16:
Well, you probably had some games with a path name > 255 characters in there?

lechuck333, 2021-01-17:
It's totally possible. It was just three games that messed things up I think, since when I checked the last game in the list, it loaded the game that was three lower than it. But either way, it's all working so far. Haven't had an issues since.

ramapcsx2, 2021-01-17:
Good good. 
I'll still investigate what's going on with too long file names, as it should just skip over them fine.
Maybe even the folder was too long and that tripped it?

birdybro, 2021-01-18:
Just need to mention for the future, Fat32 and exFAT share the same path character limit as each other.

closed 2021-03-08 by ramapcsx2