x-station/xstation-issues #61: Ore No Ryouri (JP) wont boot native or with Import Player Lite

beyondthemat, 2021-01-29:
I used to boot the classic cooking import Ore No Ryouri with the Import Player Lite, but now that doesnt work, nor will it boot natively

I'm begging you, as perfect as the XStation is, there MUST be a way I can get this one to work its an all time favorite and hoping someone can help me fix/understand the issue

I have tried multiple images and boot loaders properly in XStation

thank you!

beyondthemat, 2021-01-30:
Im running SCPH-5501 with XStation

blzla, 2021-01-30:
Are you using the Redump version? Game loads fine on my SCPH-5500, both with fast boot and normal boot, on FW 1.0.5.

beyondthemat, 2021-01-30:
yes, redump, and no luck with either

ramapcsx2, 2021-02-01:
It's unlikely that this game (or any other) don't work on xStation. Maybe it's a problem with the SD card?

closed 2021-02-13 by ramapcsx2