x-station/xstation-issues #62: 4 Capcom Games goes to Black Screen after Camcom Logo FMV,  SCPH-1001

JerryMh1, 2021-02-01:
Have 5 games thas goes to black screen after Capcom logo, rimware 1.05
Board PU8, SCPH 1001, changed the SD exfat to a FAT32 one.
Tryed Full Boot and fast boot sane result.

--Marvel Vs Capcom USA: Goes to black screen after capcom logo FMV, rip from original disk using Isobuster pro
--Marve Super Heroes vs street FIgther USA Goes to black screen after capcom log FMV, rip from original disk using Isobuster pro
--Capcom vs SNK USA Screen Blinks, then go to  black screen, rip from original disk, using Isobuster pro
--X-men vs Street Fighter goes to Black Creen, 
--Street Figher EX plus USA woks fine but has  initial long loading issue almos 25seconds of black screen.

Also downloaded several images from emupadise to try but same result. 

All Other capcom games boots fine, .

ramapcsx2, 2021-02-01:
Hey,
I've tested several of the Capcom fighters and they worked fine then.
Please try the Redump versions. I assume your rips are bad.

ramapcsx2, 2021-02-01:
Marvel vs Capcom Clash of Super Heroes NTSC-U is working on my PU-8.

JerryMh1, 2021-02-01:
Already dowloaded a redump versión of these 4 games. But the same result.
 Aldo tryed the japanese and euro versión. But stuck, on the same part.

My rips works ok, ps1 burned disk, PS3 iso, PS2 opl
And pc emulator.   I think is not the rio.

Do You Also have a PU8 board scph 1001 Playstation?

ramapcsx2, 2021-02-01:
Mine is an NTSC-J PU-8, but they're practically identical. Have you tried any other games, see whether those work?

retrofank, 2021-02-01:
I have all those and they boot and play fine on two PU-18 JP consoles I have.  All Redump images with matching MD5.  

Stay clear away from iso sites, as that is were problems arise, using bad dumps.

JerryMh1, 2021-02-01:
Yes other games works fine.
Mine is a PU8. Ntsc usa. scph 1001

Only this 4 games has this error.
And tryed the dump from the original disks , and the redumped versión from
3 regions, usa, jap and PAL from archive.org and emuparadise.  Same error
on all versions.

Have another PU8 whit a laser un good condition whit a modchip and all this
games works fine on  burned cds

ramapcsx2, 2021-02-01:
"X-Men vs. Street Fighter (USA)" tested with an ECC checker and it's fine > the game boots right up, no matter fastboot or with reset first.
I'm out of ideas for reasons why.

If anyone else has an NTSC-U PU-8 machine, can you test please?

JerryMh1, 2021-02-02:
I would like to know some technical details, if someone can help me.
1.- Xstation is affected in its operation by the values of the potentiometer that powers the PS1 lens?
2.- Xstation is affected in its operation in the values of Gain, bias and the power value of the lens motor?

I want to know this to try to recalibrate it to the original values, this board was used with a cartridge called Goldfinger that allowed to read backup copies and as I remember the values were moved so that the lens could read CDR discs in good conditions.

The other thing that remains for me to do is to reposition the flex again, although I see it unfeasible since many of the games that I have tried work correctly.

JerryMh1, 2021-02-02:
No result, recalibrate the potentiometer, same black screen, I am done whit this thing, maybe is a lemon ja ja ja..
I am gona try to buy a 5501 board, and use the other QSB

ramapcsx2, 2021-02-02:
The potentiometers have no effect with an xStation installed.
If you can get a PU-18 board, that would be a good way forward, I agree.

ramapcsx2, 2021-02-13:
I'll close this issue (cleanup time), but you can comment on it in case you have an update :)

closed 2021-02-13 by ramapcsx2

JerryMh1, 2021-02-17:
Update:  Already Has a 5501 board,  intalled and works fine, all games loads fine.  Maybe it was my board or the model not fully compatible whit xstation.
Note1: Already has a random freezen on 5501 board and test everyting again, and discover that my FCC in in bad condition, there is something break on it, it needs to be changed by a new one.  Do not know if it comes damaged. (tested putting some doble sided hard tape and woks fine, so it is broken)

ramapcsx2, 2021-02-17:
Maybe you've pinched it when installing. On a 5xxx series, you need to remove one plastic post, as described in the installation guide.

JerryMh1, 2021-02-17:
Thanks for the tip!
I bought this one,  I hope that is compatible whit xstation

https://www.ebay.com/itm/20-Pins-AWM-20624-80C-60V-VW-1-FFC-Flexible-Flat-Cable-Wire-Ribbon-1mm-Pitch/293234321960?ssPageName=STRK%3AMEBIDX%3AIT&var=592359227024&_trksid=p2057872.m2749.l2649